libretiny-eu/libretiny #1: Bounty RTL8720DN

nx10, 2022-04-30:
Cool project. I hope it's ok for me to add this as an issue, feel free to close it.

In case you don't know there is a $ 100 bounty on platformio RTL8720DN support:

https://github.com/platformio/platformio-core/issues/769#issuecomment-903447789

kuba2k2, 2022-05-01:
Unfortunately, I do not own any RTL87xxD chips. As they're substantially different, I can't guarantee that my current code will work on them. Though if this project gets more interest, I will consider buying one of these boards in the future.

kuba2k2, 2022-05-31:
Right, I see. I'm not entirely sure if my project would suffice for their needs: because Realtek's SDK and Arduino cores are quite messy, I'm mostly rewriting everything from scratch to make it compatible with ESP32 cores. 

While this extends the possibilities of running existing ESP32 programs, it makes the code less compatible with non-PlatformIO Arduino cores - i.e. those from Realtek itself. That's why I'm not sure if this is what the community wants to have.

Even then, I'm still planning to do AmebaD support at some point - after all, the base code is not that much different from AmebaZ which I have already ported. Currently, this allowed me to run ESPHome with minimal modifications on RTL8710BN.

If you're interested in using my project with RTL8720D, and would like to help - let me know, as any contribution will be much appreciated.

BinaryBrain86, 2022-11-15:
Hi, I appreciate you work and effort in this project. I followed the conversation on [elektroda](https://www.elektroda.com/rtvforum/viewtopic.php?p=19984465#19984465) and contributed to his project another device I own. That's how I found and saw this project.

I own now two other Tuya devices (power switch) with the [WR2E Module](https://developer.tuya.com/en/docs/iot/wr2e?id=K97scnsjhue4h), which have the Realtek RTL8710BN MCU. 

> Currently, this allowed me to run ESPHome with minimal modifications on RTL8710BN.

I'm wondering If you have kind of a tutorial or some small steps how to do this.
Thanks in advanced allot for reading and writing back.

kuba2k2, 2022-11-15:
Yes - we have an entire small community :slightly_smiling_face: 
We usually discuss everything on Discord and GitHub.

There's a getting-started guide: https://docs.libretuya.ml/docs/getting-started/

Also a guide for Realtek dumping/flashing: https://docs.libretuya.ml/docs/platform/realtek-ambz/dumping/

BinaryBrain86, 2022-11-15:
> Yes - we have an entire small community 🙂 We usually discuss everything on Discord and GitHub.
> 
> There's a getting-started guide: https://docs.libretuya.ml/docs/getting-started/
> 
> Also a guide for Realtek dumping/flashing: https://docs.libretuya.ml/docs/platform/realtek-ambz/dumping/

Thank you, I'll check it out.

MrAppAndCrap, 2023-02-24:
> Yes - we have an entire small community 🙂 We usually discuss everything on Discord and GitHub.
> 
> There's a getting-started guide: https://docs.libretuya.ml/docs/getting-started/
> 
> Also a guide for Realtek dumping/flashing: https://docs.libretuya.ml/docs/platform/realtek-ambz/dumping/

Hey, what is the discord link / name? Thanks

kuba2k2, 2023-02-24:
Hi, it's at the top of the README.

wrobelda, 2026-02-09:
Actually, am I wrong or is this the SDK? https://github.com/Ameba-AIoT/ameba-rtos-d