libretiny-eu/libretiny #4: Could tuya-convert flash libretuya-based firmwares?

felipecrs, 2022-07-13:
Your initiative with this project is simply amazing.

I'm not so experienced in it, so I wonder if the same approach used by tuya-convert could also be used to flash libretuya-based firmwares.

kuba2k2, 2022-07-13:
Not really. Tuya-convert works only for ESP8266 and ESP32 chips AFAIK.

There is a great project that works for BK7231 chips, which should make it usable with LT firmwares:
https://github.com/tuya-cloudcutter/tuya-cloudcutter

Currently, LT does not export an image for "stock" OTA flashing, so you'd have to fiddle a bit with Beken OTA packaging tools (available in Tuya SDK packages). I'm planning to provide a universal firmware image (board-independent), which would be available as an OTA file, and provide a WiFi hotspot to flash whichever version of LT you want.

nx10, 2022-07-13:
Very cool. I hope the vulnerability won't get patched before we can flash LT. The article mentions its been fixed in new stock since their disclosure Feb 12th 2022, but I don't if it has been patched in running (old) devices.

felipecrs, 2022-07-13:
That's very cool indeed!

felipecrs, 2022-07-13:
Feel free to close the issue if you would like.

kuba2k2, 2022-09-03:
Hi all, I forgot to mention: 
LibreTuya now exports an "UG" image, which is suitable for Cloudcutter flashing. I tested it once with a Cloudcutter-enabled device and ESPHome firmware, and it worked flawlessly :tada:

felipecrs, 2022-09-03:
This is awesome! :D

felipecrs, 2022-09-05:
I think this means this was resolved so. Thanks a lot!

closed 2022-09-05 by felipecrs