libretiny-eu/libretiny #6: Problem with esphome-libretuya

bonuzzz, 2022-09-02:
I didn't find where I could write about esphome, so decide to do it here.
When trying to compile firmware for wr3e, I got error
`Compiling .pioenvs/cur/libretuya_api/libraries/Update/uf2ota/uf2binpatch.c.o
In file included from /home/bonuzzz/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/time.h:42:0,
                 from /home/bonuzzz/.platformio/platforms/libretuya/arduino/libretuya/libraries/NetUtils/lwip/LwIPClient.cpp:24:
/home/bonuzzz/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/_timeval.h:52:8: error: redefinition of 'struct timeval'
 struct timeval {
        ^
In file included from /home/bonuzzz/.platformio/platforms/libretuya/arduino/libretuya/libraries/NetUtils/lwip/LwIPClient.cpp:23:0:
/home/bonuzzz/.platformio/packages/library-lwip@2.0.0-amb1/src/include/lwip/sockets.h:442:8: error: previous definition of 'struct timeval'
 struct timeval {
        ^
Compiling .pioenvs/cur/libretuya_api/libraries/Update/uf2ota/uf2ota.c.o
In file included from /home/bonuzzz/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/time.h:42:0,
                 from /home/bonuzzz/.platformio/platforms/libretuya/arduino/libretuya/libraries/NetUtils/lwip/LwIPServer.cpp:16:
/home/bonuzzz/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/_timeval.h:52:8: error: redefinition of 'struct timeval'
 struct timeval {
        ^
In file included from /home/bonuzzz/.platformio/platforms/libretuya/arduino/libretuya/libraries/NetUtils/lwip/LwIPServer.cpp:15:0:
/home/bonuzzz/.platformio/packages/library-lwip@2.0.0-amb1/src/include/lwip/sockets.h:442:8: error: previous definition of 'struct timeval'
 struct timeval {
        ^
Compiling .pioenvs/cur/libretuya_api/libraries/Update/uf2ota/uf2priv.c.o
*** [.pioenvs/cur/libretuya_api/libraries/NetUtils/lwip/LwIPServer.cpp.o] Error 1
*** [.pioenvs/cur/libretuya_api/libraries/NetUtils/lwip/LwIPClient.cpp.o] Error 1
`

I use ubuntu 22.04

kuba2k2, 2022-09-02:
Hmm, weird, that works for me. Can you post the `idedata.json` file? You can find it in `.pioenvs/cur/`.

bonuzzz, 2022-09-02:
that folder doesn't created. I run command python -m esphome compile 1.yaml and see that error just after start.

kuba2k2, 2022-09-02:
Hm, I think I meant `.esphome/build/something/.pioenvs/cur/`.

bonuzzz, 2022-09-02:
I have only these files and folders in that path:
`ambz_arduino_core  ambz_arduino_port  board_wr3e      libambz_arduino_port.a  libboard_wr3e.a  libretuya_api.a
ambz_arduino_libs  arduino_api        flashdb03500fa  libarduino_api.a        libretuya_api    src`

Anyway I redownload libretuya-esphome and reinstalled it.  I'm trying to build minimal config and get new error:
`Compiling .pioenvs/cur/ambz_arduino_core/cores/arduino/wiring_digital.c.o
Compiling .pioenvs/cur/ambz_arduino_libs/libraries/WiFi/WiFiAP.cpp.o
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/cores/arduino/wiring_digital.c: In function 'pinMode':
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/cores/arduino/wiring_digital.c:89:11: error: 'OpenDrain' undeclared (first use in this function)
    mode = OpenDrain;
           ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/cores/arduino/wiring_digital.c:89:11: note: each undeclared identifier is reported only once for each function it appears in
Compiling .pioenvs/cur/ambz_arduino_libs/libraries/WiFi/WiFiSTA.cpp.o
Compiling .pioenvs/cur/flashdb03500fa/port/fal/src/fal_rtt.c.o
Compiling .pioenvs/cur/flashdb03500fa/src/fdb.c.o
Compiling .pioenvs/cur/flashdb03500fa/src/fdb_file.c.o
Compiling .pioenvs/cur/flashdb03500fa/src/fdb_kvdb.c.o
*** [.pioenvs/cur/ambz_arduino_core/cores/arduino/wiring_digital.c.o] Error 1
Compiling .pioenvs/cur/flashdb03500fa/src/fdb_tsdb.c.o
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiAP.cpp: In member function 'bool WiFiClass::softAPConfig(arduino::IPAddress, arduino::IPAddress, arduino::IPAddress)':
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiAP.cpp:94:17: error: aggregate 'ip_addr ipaddr' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                 ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiAP.cpp:94:25: error: aggregate 'ip_addr netmask' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                         ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiAP.cpp:94:34: error: aggregate 'ip_addr gw' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                                  ^
*** [.pioenvs/cur/ambz_arduino_libs/libraries/WiFi/WiFiAP.cpp.o] Error 1
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp: In member function 'bool WiFiClass::config(arduino::IPAddress, arduino::IPAddress, arduino::IPAddress, arduino::IPAddress, arduino::IPAddress)':
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:40:17: error: aggregate 'ip_addr d1' has incomplete type and cannot be defined
  struct ip_addr d1, d2;
                 ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:40:21: error: aggregate 'ip_addr d2' has incomplete type and cannot be defined
  struct ip_addr d1, d2;
                     ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:53:17: error: aggregate 'ip_addr ipaddr' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                 ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:53:25: error: aggregate 'ip_addr netmask' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                         ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:53:34: error: aggregate 'ip_addr gw' has incomplete type and cannot be defined
  struct ip_addr ipaddr, netmask, gw;
                                  ^
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp: In member function 'arduino::IPAddress WiFiClass::dnsIP(uint8_t)':
/home/bonuzzz/.platformio/platforms/libretuya/arduino/realtek-ambz/libraries/WiFi/WiFiSTA.cpp:168:17: error: aggregate 'ip_addr dns' has incomplete type and cannot be defined
  struct ip_addr dns;
                 ^
*** [.pioenvs/cur/ambz_arduino_libs/libraries/WiFi/WiFiSTA.cpp.o] Error 1`

My config is:
`esphome:
  name: curtain

libretuya:
  board: wr3e
  framework:
    version: latest

# Enable logging
logger:

wifi:
  ssid: 'ssid'
  password: 'pass'`

kuba2k2, 2022-09-02:
Check that you have the latest version of LibreTuya itself (this is something separate from libretuya-esphome).

`platformio platform install https://github.com/kuba2k2/libretuya`

You might need to uninstall the platform first (probably something like `platformio platform uninstall libretuya`).

bonuzzz, 2022-09-02:
`platformio platform install https://github.com/kuba2k2/libretuya`
this command fixed all issues. firmware compiled successfully after that

closed 2022-09-02 by bonuzzz