libretiny-eu/libretiny #7: problem with flashing wr3e

bonuzzz, 2022-09-03:
I have curtains motor switch with wr3e and trying to flash it.
When I pull tx2 pin on ground when starting, I see diode on the board doesn't light. I think I switch chip to download board. But I can't connect to tx0 rx0 to flash firmware. I'm sure about hardware wiring. Maybe does it have something special to open port? I tried to pull tx2 permanently and pull about 1 sec at start.
In normal mode I see in logs on rx2 tx2 strings but in incorrect codepage.

Do I need to unsolder chip before flashing it?

kuba2k2, 2022-09-03:
Hi
What does it mean you "can't connect to tx0 rx0"?
UART2 is used for flashing, read here: https://docs.libretuya.ml/docs/platform/realtek-ambz/flashing/
You don't need UART0 at all, it's not used for flashing, nor for log outputs.

Pull TX2 pin to GND, meanwhile power up the board. The chip then enters "download mode" and waits for flashing commands. You have to disconnect TX2 from GND before the actual flashing process, but without powering the board off.

You can check if it is in download mode by opening up a terminal (PuTTY, or other) and setting any baud rate. It should print something (not readable) few times a second.

bonuzzz, 2022-09-03:
Thanks!
It was very simple. I read that manual many times:) I don't know why I wanted to flash with tx0 rx0. I thought to see here the same logic as bk7231t
ps: I flashed over the air esphome with web, but it doesn't boot. I will check later it. If I don't mistake somewhere again, I will write you
Big thanks again!:)

bonuzzz, 2022-09-03:
I noticed unstable ota update from console (esphome upload). very frequently firmware doesn't boot after upload ota and reboot.

kuba2k2, 2022-09-04:
Try using OTA through web server, it works much better. The changelog mostly applied to BK7231, since I didn't notice such problems on Realtek, but it might fix some stuff on RTL as well.

You can update LT, compile ESPHome and try uploading it in any way.

closed 2022-09-04 by bonuzzz