libretiny-eu/libretiny #9: esphome ota

bonuzzz, 2022-09-04:
this is what I see when updating esphome in console (esphome upload) through wifi. (wr3e board)
there was 0.9 libretuya version. Will update to 0.10
```
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:0, flash_speed idx:0
calibration_result:[1:9:9][5:d] 
calibration_result:[2:13:7][1:d] 
calibration_result:[3:3:3][1:3] 
calibration_ok:[2:13:7] 
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[ffffffe0]
OTA2 USE
OTA2 SIGN[35393138:31313738]
IMG2 DATA[0x813c430:3996:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x80f6061]
System_Init1
RTL8195A[HAL]: Hard Fault Error!!!!
RTL8195A[HAL]: R0 = 0x1003ef5c
RTL8195A[HAL]: R1 = 0x40003400
RTL8195A[HAL]: R2 = 0xff
RTL8195A[HAL]: R3 = 0xca205
RTL8195A[HAL]: R12 = 0x0
RTL8195A[HAL]: LR = 0x80f54f5
RTL8195A[HAL]: PC = 0xca204
RTL8195A[HAL]: PSR = 0x1000200
RTL8195A[HAL]: BFAR = 0xe000ed38
RTL8195A[HAL]: CFSR = 0x0
RTL8195A[HAL]: HFSR = 0x80000000
RTL8195A[HAL]: DFSR = 0x2
RTL8195A[HAL]: AFSR = 0x0
RTL8195A[HAL]: PriMask 0x0
RTL8195A[HAL]: BasePri 0x0
RTL8195A[HAL]: SVC priority: 0x00
RTL8195A[HAL]: PendSVC priority: 0x00
RTL8195A[HAL]: Systick priority: 0x00
```

bonuzzz, 2022-09-04:
on 0.10 update in console over wifi doesn't work. I get the same log as above. Update in web server works.
Is it possible to reboot automatically the board after flashing over usb?

kuba2k2, 2022-09-05:
I didn't have OTA issues on Realtek, but I'll check this when I get back to working on RTL devices (as I only have one or two of them).

It's possible to modify the flasher to reboot automatically, but I decided to disable that, since it takes a few seconds to open the serial terminal right after uploading, and I didn't want to miss any output.

kuba2k2, 2022-09-30:
Hi @bonuzzz 
I've looked into this deeper since I got another issue with similar problems (#24), and I think I was able to resolve it.

You can update LT (`platformio platform install -f https://github.com/kuba2k2/libretuya`), then upload firmware via UART. After that, OTA should work.

kuba2k2, 2022-11-26:
Closing this since it's stale, and the bug fix was tested already.

closed 2022-11-26 by kuba2k2