libretiny-eu/libretiny #11: Serial component broken on bk7231n

dbuezas, 2022-09-08:
In bk7231n, any usage of the uart component makes compilation fail due to missing SoftwareSerial.h file.

### Workaround: 
```diff
diff --git a/esphome/components/uart/uart_component_libretuya.cpp b/esphome/components/uart/uart_component_libretuya.cpp
index 21fc3e182a534339fd8efe432172f31f0bcbf790..4f4e3993648c20328deced4e85aed1a7f1482619 100644
--- a/esphome/components/uart/uart_component_libretuya.cpp
+++ b/esphome/components/uart/uart_component_libretuya.cpp
@@ -10,7 +10,7 @@
 #include "esphome/components/logger/logger.h"
 #endif
 
-#include <SoftwareSerial.h>
+//#include <SoftwareSerial.h>
 
 namespace esphome {
 namespace uart {
@@ -73,7 +73,7 @@ void LibreTuyaUARTComponent::setup() {
   }
 #endif
   else {
-    this->serial_ = new SoftwareSerial(rx_pin, tx_pin, rx_inverted || tx_inverted);
+    // this->serial_ = new SoftwareSerial(rx_pin, tx_pin, rx_inverted || tx_inverted);
   }
 
   this->serial_->begin(this->baud_rate_, get_config());
@@ -83,9 +83,7 @@ void LibreTuyaUARTComponent::dump_config() {
   bool is_hardware = this->hardware_idx_ == -1;
   ESP_LOGCONFIG(TAG, "UART Bus:");
   ESP_LOGCONFIG(TAG, "  Type: %s", UART_TYPE[is_hardware]);
-  if (is_hardware) {
-    ESP_LOGCONFIG(TAG, "  Port number: %d", this->hardware_idx_);
-  }
+  ESP_LOGCONFIG(TAG, "  Port number: %d", this->hardware_idx_);
   LOG_PIN("  TX Pin: ", tx_pin_);
   LOG_PIN("  RX Pin: ", rx_pin_);
   if (this->rx_pin_ != nullptr) {

```

kuba2k2, 2022-09-15:
Related to #15.

kuba2k2, 2022-09-21:
Fixed in 30f08a145ec38fd5b8f5106fd1ae9cb1768ba1bd (related ESPHome commit - esphome/esphome@e76fb0850016fdb741dd8b0684611df1e82d1b40).

closed 2022-09-21 by kuba2k2