libretiny-eu/libretiny #13: WiFi issues

kuba2k2, 2022-09-08:
- [ ] fix AP mode (DHCP issues, wrong IP setting/reading) on ESPHome
- [ ] enable fast connect (BSSID connect)
- [ ] fix STA connecting if multiple APs with same name are available
- [ ] fix rare STA connection problems (esphome) - "association leave" before fully connecting, and looping few times with these kind of errors
- [ ] fix AP mode on ESPHome (Realtek)

Elkropac, 2022-12-10:
Hi, i have flashed my lvl300 device (the one i mentioned in [tuyacutter](https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/174)), it worked well, but now it does not connect to wifi properly. 

I can see it associate in unifi management, but it does not get any ip. 
I did not compile AP to it, so i don't know if i can get to it somehow without disassembly?

kuba2k2, 2022-12-10:
Have you tried power-cycling that device?

Elkropac, 2022-12-10:
I have turned it off multiple times. It always associates to wifi, but it gets no ip from dhcp (i will check if it asks for ip, i don't know)

I have multiple APs in my house, could it be a problem?

kuba2k2, 2022-12-10:
It might be a problem, but then it would likely not associate at all. Did you change anything in your network configuration? Stuff usually doesn't "just" stop working; if it worked fine for weeks, I don't see any reason why it would break now.

If you won't be able to make it connect, then without AP there's not much you can do. In theory, you could take it far from your home network, and set-up another AP with same SSID and password as your home network.
And.. remember to always add the AP component, in case of failures like this one.

Elkropac, 2022-12-10:
I have disabled 2.4GHz radio on my wifi network for laptops/mobile devices, but not on network for IoT devices.
I will try to enable it again, but i don't thing it's related.

And i will add AP component to my common configuration, seems like good idea :)

Elkropac, 2022-12-10:
I can see dhcp packets on my AP
```
17:50:48.965920 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 10:5a:17:bd:de:cb (oui Unknown), length 308
        0x0000:  4500 0150 005d 0000 ff11 ba40 0000 0000
        0x0010:  ffff ffff 0044 0043 013c d268 0101 0600
        0x0020:  4748 fbe1 0000 0000 0000 0000 0000 0000
        0x0030:  0000 0000 0000 0000 105a 17bd decb 0000
        0x0040:  0000 0000 0000 0000 0000 0000 0000 0000
        0x0050:  0000
```
and on my firewall, it is dropped
```
Ethernet Header
 Ether Type: VLAN ID = 23, Priority = 0
 Ether Type: 0x6(0x6), Src=[10:5a:17:bd:de:cb], Dst=[ff:ff:ff:ff:ff:ff]
Ethernet Type: Unknown
Value:[0]
DROPPED, Drop Code: 17(Unknown Ether type ingress.), Module Id: 16(fwCore), (Ref.Id: _3057_joqvuIppl) 2:2)
```
i have to ask my collegue what does it mean

Elkropac, 2022-12-10:
Hm, when i deleted dhcp reservation for this mac address, it got different IP from pool for new devices and i was able to connect to it. Strange

mihsu81, 2022-12-11:
> [ ] fix STA connecting if multiple APs with same name are available

On a EZVIZ T31 (generic-rtl8710bx-4mb-980k) almost every reboot causes the device to go in AP mode. Is it caused by the STA issue you mentioned above as pending?

kuba2k2, 2022-12-11:
I don't know. That particular issue was related to Beken, not Realtek. And AP mode on Realtek doesn't work at all, from what I tested.

mihsu81, 2022-12-11:
> And AP mode on Realtek doesn't work at all, from what I tested.

Indeed, when i try to connect my phone to the Fallback SSID it remains at _**Connecting...**_.

> That particular issue was related to Beken, not Realtek.

True, I've shut down all the other APs broadcasting the SSID and the device still goes in AP mode.

mihsu81, 2022-12-20:
When adding ```  fast_connect: true``` the devices reconnect just fine. I assume that while rescanning for available SSIDs it encounters an issue and is not able to connect. This leads to it going in AP mode.

BTW, I was able to connect from my laptop to a device which went into AP mode (Fallback SSID), but it took a long time (15-20s).

kuba2k2, 2022-12-20:
I have an RTL8710BN running day and night now (finally mounted as a hacked+modified power switch), and it has no problems at all. Even when the home network was unavailable for a few minutes yesterday, it reconnected just fine and continued to work.

You say "devices", how many is that? Do all of them have the same problems?

mihsu81, 2022-12-20:
Sorry, it was a typo. I meant _device_.
I have multiple APs from Asus in a Mesh. Maybe it's related somehow? Even though I have tested last week only with the router and it had the same issues, of not being able to connect after a reboot.

Elkropac, 2023-03-19:
I have problems with lvl300 device again.
I recompiled firmware to enable homeassistant api encryption, uploaded and it does not associate to wifi anymore. Event fallback AP does not pop up. I resolved my last problems (from december) by disabling dhcp and setting fixed IP address, but now i don't know what to do
```
substitutions:
  <<: !include _common/subst.yaml
  dev_name: lvl300_001
  dev_name_network: lvl300-001.devices
  log_level: INFO
  boot_brightness: "0.9

packages:
  wifi: !include _packages/wifi_common.yaml

libretuya:
  board: cb3l
  framework:
    version: dev

wifi:
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.23.59
    gateway: 192.168.23.1
    subnet: 255.255.255.0
```
and wifi_common.yaml is
```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: yes
  use_address: "${dev_name_network}"
  ap:
    ssid: "${dev_name} AP"
    password: !secret ap_password

captive_portal:
```

kuba2k2, 2023-03-19:
Are you sure you have the latest version of LT? There has been an error recently, that prevented some devices from booting properly. If you've uploaded firmware having that problem, you'll have to serial flash it to working firmware.

Elkropac, 2023-03-19:
I think i forgot to do git pull, i was compiling version your provided for me when fixing issue #101

kuba2k2, 2023-03-19:
If you did a git pull around 5 days ago, then chances are you were left on the faulty commit.

Elkropac, 2023-03-19:
i think, i'm on latest commit in branch platform/libretuya, should i use another branch?

kuba2k2, 2023-03-19:
It's not about esphome, it's about LibreTuya itself. See the docs on how to update the PlatformIO package.

Elkropac, 2023-03-22:
Thanks, it worked with updated platformio

closed 2023-08-18 by kuba2k2