libretiny-eu/libretiny #14: BK7231 I2C driver

kuba2k2, 2022-09-08:


Marvin-D-Martian, 2023-08-26:
Hi.
Is there any way an implamentation of the "I2C" led driver KP18058 or KP18068 could be implemented?
A range of very affordable Bulbs is available with BK7231n using the 18058/18068 is now available. Everything works up to a point of ESPHome having no support for the driver. OpenBeken has the support, but ESPhome is the platform of  choice for many of us and except for the led driver support, it has all else inplace already.

p.kacmarek and his team made a huge progress and all the workd id bssically done:
[https://www.elektroda.com/rtvforum/viewtopic.php?p=20708687#20708687](url)

I am saying "I2C" as it's function is fully described here:
[https://www.elektroda.com/rtvforum/topic3991620.html](url)

I am more than happy to help in making this happen, unfortunately not throught my knowledge (limited) but I am happy to preovide a bulb for testing, buy coffee... :)
Many thanks

kuba2k2, 2023-08-26:
That's not really up to me - you'd have to ask ESPHome developers for this driver support. I don't own any bulbs with such a driver, so I'm not really able to write a component for it.

That being said, it's true that I2C is not yet supported in LibreTiny. There are, however, components that use software based I2C in ESPHome itself, such as BP5758 driver (if I got the symbol right).

Marvin-D-Martian, 2023-08-27:
Thanks for your reply. I will do as you suggest. Thanks for your time and your great work!

dwyschka, 2023-10-05:
is there any eta, when i2c will be available?

kuba2k2, 2023-10-05:
Probably soon. I'm waiting for new Beken boards to arrive, then I'll be able to work on I2C for BK7231. I've already implemented I2C and SPI for RTL8710, so it should be pretty straightforward.

dwyschka, 2023-10-05:
Would be awesome. 
i own a Klarstein Wonderwall, and it communicates over i2c with the other controller.

kuba2k2, 2023-10-05:
What other controller is that? Are you sure it uses I2C?

TuyaMCU controllers always use UART. If that's a PWM controller which uses I2C, there's a high chance that ESPHome has a component for it which uses software I2C.

dwyschka, 2023-10-05:
It uses the https://github.com/buzzer13/esphome-components TM1650 controller. 
Extended Information about the device: 
https://www.elektroda.com/rtvforum/topic3952083.html

Would be awesome to got it in esphome.

Marvin-D-Martian, 2023-10-05:
FWIW ...The problem with KP18058/18068 is that is a sort of a hack I2C. It's documented (now) but acan't be used with existing controllers supported by ESPHome. In their Discord channel one person mentioned thay looked into this a while ago a did quite a lot of work on a similar controller... so I'm hoping htis will become more popolar controller that will eventually get support...

sundeepgoel72, 2023-12-06:
am adding a lot of homemate 4 gang switches to my house. These have the  7231N controller with 5 free GPIO pins, so will be very cool to be able to use existing i2s sensors with controller

sundeepgoel72, 2023-12-07:
I would love to help / contribute to writing the i2c driver, but will need guidance to get started.

shakin89, 2024-01-26:
Just following to have a notification for when this will be implemented. Hope to see it soon. I would like to help, but i have a very poor knowledge on the argument. 
Thanks in advance

Marvin-D-Martian, 2025-07-19:
My post above points to elektroda web which explains the "i2c" integration in this chip. They did a good job taking this apart and it seems it's not a full i2c standard. It would most likely mean a custom implementation. That's as far as I can uderstand it. Yet I still hope it can be done.