libretiny-eu/libretiny #19: unable to compile

mario-almeida, 2022-09-16:
`python -m esphome compile workshop-8.yaml 
`
give this error

> ERROR Unexpected exception while reading configuration:
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
>     exec(code, run_globals)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/__main__.py", line 950, in <module>
>     sys.exit(main())
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/__main__.py", line 941, in main
>     return run_esphome(sys.argv)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/__main__.py", line 919, in run_esphome
>     config = read_config(dict(args.substitution) if args.substitution else {})
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 975, in read_config
>     res = load_config(command_line_substitutions)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 830, in load_config
>     return _load_config(command_line_substitutions)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 818, in _load_config
>     result = validate_config(config, command_line_substitutions)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 745, in validate_config
>     result.run_validation_steps()
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 124, in run_validation_steps
>     task.step.run(self)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config.py", line 500, in run
>     validated = schema(self.conf)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/voluptuous_schema.py", line 34, in __call__
>     res = super().__call__(data)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
>     return self._compiled([], data)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/validators.py", line 229, in _run
>     return self._exec(self._compiled, value, path)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/validators.py", line 353, in _exec
>     v = func(path, v)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable
>     return schema(data)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/voluptuous_schema.py", line 34, in __call__
>     res = super().__call__(data)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
>     return self._compiled([], data)
>   File "/home/remy/.local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
>     return base_validate(path, iteritems(data), out)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/voluptuous_schema.py", line 118, in validate_mapping
>     not isinstance(key.default, vol.Undefined)
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/config_validation.py", line 1457, in default
>     if CORE.is_esp8266:
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/core/__init__.py", line 593, in is_esp8266
>     return self.target_platform == "esp8266"
>   File "/home/remy/workspace/github.com/esphome_beken/libretuya-esphome/esphome/core/__init__.py", line 589, in target_platform
>     return self.data[KEY_CORE][KEY_TARGET_PLATFORM]
> KeyError: 'target_platform'

kuba2k2, 2022-09-16:
Post your .yaml file - most likely you have an error there.

mario-almeida, 2022-09-16:
```
esphome:
  name: workshop-8

libretuya:
  board: cb2s
  framework:
    version: latest
    sdk_silent: true
    gpio_recover: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "myssid"
  password: "password"

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Workshop-8 FH"
    password: "fA30OYwIpBMb"
```

kuba2k2, 2022-09-16:
Right here:
```yaml
libretuya:
  board: cb2s
  framework:
    version: latest
    sdk_silent: true
    gpio_recover: true
```
`sdk_silent` and `gpio_recover` should be directly in `libretuya` block, so remove one indentation level and it should work.

mario-almeida, 2022-09-16:
> Right here:
> 
> ```yaml
> libretuya:
>   board: cb2s
>   framework:
>     version: latest
>     sdk_silent: true
>     gpio_recover: true
> ```
> 
> `sdk_silent` and `gpio_recover` should be directly in `libretuya` block, so remove one indentation level and it should work.

silly mistake, now ok

```
libretuya:
  board: cb2s
  framework:
    version: latest
  sdk_silent: true
  gpio_recover: true
```
Thanks a lot

closed 2022-09-16 by kuba2k2