libretiny-eu/libretiny #20: unable to update the firmware to CB2S

mario-almeida, 2022-09-16:
`python -m esphome upload workshop-8.yaml`

> INFO Reading configuration workshop-8.yaml...
> Found multiple options, please choose one:
>   [1] /dev/ttyUSB0 (CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller)
>   [2] Over The Air (192.168.1.10)
> (number): 1
> **************************************************************************************************************************************************************************
> Obsolete PIO Core v6.0.2 is used (previous was 6.1.4)
> Please remove multiple PIO Cores from a system:
> https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
> **************************************************************************************************************************************************************************
> Processing workshop-8 (board: cb2s; framework: arduino; platform: libretuya)
> --------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> HARDWARE: BK7231N 120MHz, 256KB RAM, 1.03MB Flash
>  - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
>  - framework-beken-bdk @ 0.0.0+sha.c3533e8.sha.c3533e8
>  - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
>  - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.33191e0
>  - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
>  - tool-bk7231tools @ 1.0.0-eea4c3+sha.1eea4c3
>  - tool-ltchiptool @ 1.5.2+sha.f7a0850
>  - tool-openocd @ 2.1100.211028 (11.0)
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> Dependency Graph
> |-- ESPAsyncWebServer-esphome @ 3.0.0
> |   |-- AsyncTCP-esphome @ 2.0.0
> |-- ArduinoJson @ 6.18.5
> Building UF2 OTA image
> |-- firmware.uf2
> RAM:   [====      ]  35.8% (used 93740 bytes from 262144 bytes)
> Flash: [=======   ]  69.4% (used 751488 bytes from 1083136 bytes)
> Configuring upload protocol...
> AVAILABLE: uart
> CURRENT: upload_protocol = uart
> Looking for upload port...
> Using manually specified: /dev/ttyUSB0
> Uploading firmware.uf2
> |-- esphome 2022.10.0-dev @ 2022-09-16 21:18:48 -> cb2s
> |-- Using UART
> |   |-- Trying to link on /dev/ttyUSB0 @ 460800
> Connected! Chip info: 0x7231c
> |   |-- Writing 803182 bytes to 0x011000
> ERROR: RuntimeError: Failed to unprotect
>  - File "/home/remy/.platformio/packages/tool-bk7231tools/bk7231tools/serial/__init__.py", line 75, in program_flash
> *** [upload] Error 1
> ======================================================================= [FAILED] Took 3.45 seconds =======================================================================

yaml file
```
esphome:
  name: workshop-8

libretuya:
  board: cb2s
  framework:
    version: latest
  sdk_silent: true
  gpio_recover: true
  loglevel: verbose

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

web_server:
  port: 80

wifi:
  ssid: "myssid"
  password: "password"

  manual_ip:
    static_ip: 192.168.1.10
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Workshop-8 FH"
    password: "fA30OYwIpBMb"
```

The upload file is firmware.uf2, shouldn't it be the firmware.bin file?

Manually I am able to upload but unable to access the device
`uartprogram .esphome/build/workshop-8/.pioenvs/workshop-8/firmware.bin -d /dev/ttyUSB0 -w -s 0x0 -u`

kuba2k2, 2022-09-16:
Yes, the upload file should be `firmware.uf2`. `firmware.bin` is a **raw** ARM binary file - it's not supposed to be uploaded to the device. If you uploaded it using that `uartprogram` command, you removed the bootloader of the chip - so you won't be able to run any firmware until you flash that bootloader again.

There's a lot of confusion made by `hid_download_py` authors and OpenBeken's flashing instructions.
**Flashing to 0x0 is not needed nor recommended! At all. Never flash to 0x0, and don't use "QIO" binaries.**

Right now, you have to flash the dump of stock firmware (if you made one), because you don't have the bootloader. If you didn't, comment here and I'll tell you what you should do.

Does the "Failed to unprotect" error happen every time?

If you have Discord, you can [join our server](https://discord.gg/SyGCB9Xwtf) which would speed things up a bit, especially in the case of such a weird error.

kuba2k2, 2022-09-21:
Hi @mario-almeida 
Try with the latest commit (05cb7f79474c0a4ca6663c3feeb3f3a0671d0dcb). 
Commit 5b6b80e3cd45401501939053b30cc7d4309acdcf contains a new version of bk7231tools which should fix uploading.

dbuezas, 2022-10-03:
I've uploaded to CB2S hundreds of times in the last week during development, so this is most likely solved.

kuba2k2, 2022-10-03:
There was a problem with the Beken UART flasher, that didn't recognize the flash chip in @mario-almeida's module. That's why uploading could work for you with no problems.

kuba2k2, 2022-11-26:
I'll close this as we're probably not going to get a response anymore. If the issue still persists, please reopen it.

closed 2022-11-26 by kuba2k2