libretiny-eu/libretiny #23: Unable to flash RTL8710BX - Invalid OTA2 address on chip - found 1048576, expected 851968

mihsu81, 2022-09-22:
I'm trying to flash ESPHome LibreTuya on a RTL8710BX (Ezviz T31 https://www.ezviz.com/product/T31/2021), but I'm getting the following error: ```Invalid OTA2 address on chip - found 1048576, expected 851968```.
```
PS C:\Users\Bethayn-Dell\OneDrive - TUI\Documents\GitHub\esphome\libretuya-esphome> python -m ltchiptool uf2 upload "C:\Users\Bethayn-Dell\OneDrive - TUI\Documents\GitHub\esphome\libretuya-esphome\.esphome\build\plug-ezviz-test1\.pioenvs\plug-ezviz-test1\firmware.uf2" uart COM7
|-- esphome 2022.10.0-dev @ 2022-09-22 15:43:40 -> wr3l
|-- Using UART
|   |-- Connecting to COM7...
ERROR: ValueError: Invalid OTA2 address on chip - found 1048576, expected 851968
 - File "C:\Program Files\Python39\lib\site-packages\ltchiptool\soc\ambz\upload.py", line 36, in upload
```

If I try to upload it using esphome, I get ```Permission denied``` because of blank spaces in the path.

```
PS C:\Users\Bethayn-Dell\OneDrive - TUI\Documents\GitHub\esphome\libretuya-esphome> python -m esphome upload plug-ezviz-test1.yml
←[32mINFO Reading configuration plug-ezviz-test1.yml...←[0m
Found multiple options, please choose one:
  [1] COM7 (USB-SERIAL CH340 (COM7))
  [2] Over The Air (plug-ezviz-test1.local)
(number): 1
************************************************************************************************************************
Obsolete PIO Core v6.0.2 is used (previous was 6.1.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
************************************************************************************************************************
Processing plug-ezviz-test1 (board: wr3le; framework: arduino; platform: libretuya)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/libretuya/wr3le.html
PLATFORM: LibreTuya (0.10.0+sha.b404f9f) > WR3LE Wi-Fi Module
HARDWARE: RTL8710BX 62MHz, 256KB RAM, 788KB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-realtek-amb1 @ 750.0.0-b0bc+sha.750b0bc
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.0.0-amb1 @ 2.0.0-amb1+sha.c39c3f5
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-bk7231tools @ 1.0.0-eea4c3+sha.1eea4c3
 - tool-ltchiptool @ 1.5.2+sha.f7a0850
 - tool-openocd @ 2.1100.211028 (11.0)
 - toolchain-gccarmnoneeabi @ 1.50201.0 (5.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Building UF2 OTA image
|-- firmware.uf2
Checking size .pioenvs\plug-ezviz-test1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.1% (used 52780 bytes from 262144 bytes)
Flash: [=====     ]  48.7% (used 393164 bytes from 806912 bytes)
Configuring upload protocol...
AVAILABLE: uart
CURRENT: upload_protocol = uart
Looking for upload port...
Using manually specified: COM7
Uploading firmware.uf2
Usage: python -m ltchiptool uf2 upload [OPTIONS] FILE COMMAND [ARGS]...
Try 'python -m ltchiptool uf2 upload -h' for help.

Error: Invalid value for 'FILE': 'C:\Users\Bethayn-Dell\OneDrive': Permission denied
*** [upload] Error 2
============================================== [FAILED] Took 7.57 seconds ==============================================
```

Since it's not a Tuya device, I've attached a full firmware dump, in case it's needed.
[dump_ezviz_new.zip](https://github.com/kuba2k2/libretuya/files/9626444/dump_ezviz_new.zip)

Also some logs from UART
```
interface 0 is initialized

interface 1 is initialized

Initializing WIFI ...
<RTL8195A>
ROM:[V0.1]

FLASHRATE:4

BOOT TYPE:0 XTAL:40000000

IMG1 DATA[1128:10002000]

IMG1 ENTRY[8000521:100021ef]

IMG1 ENTER

CHIPID[000000f6]

read_mode idx:3, flash_speed idx:3

calibration_result:[1:25:13][1:19] 

calibration_result:[2:25:13][1:19] 

calibration_result:[3:0:0][ff:ff] 

calibration_ok:[1:25:13] 

FLASH CALIB[NEW OK]

OTA2 ADDR[8100000]

OTAx SELE[fffffffe]

OTA2 USE

OTA2 SIGN[35393138:31313738]

IMG2 DATA[0x817283c:11448:0x10005000]

IMG2 SIGN[RTKWin(10005008)]

IMG2 ENTRY[0x10005000:0x816bfb5]

===== Enter Image 2 ====

System_Init1

OSC8M: 8390004 

boot reason: 0 

System_Init2

#
```

kuba2k2, 2022-09-22:
Oh it's not a Tuya device, hmm that's a new one.

The board inside is a WR3LE? Made by Tuya? Or it has another module, or perhaps RTL soldered directly to the PCB?

From the dump it seems to actually use a different OTA2 address. Is that a complete dump? It's a few bytes under 2 MiB, which is uncommon. Are you sure the flash chip is 2 MiB and not larger?

Did you flash any custom firmware to that board yet? Or does it still successfully boot the stock firmware?

mihsu81, 2022-09-22:
Indeed, it's not compatible with the Tuya cloud.
It's a RTL8710BX soldered onto the PCB and a 4MB 25Q32CSIG flash memory (http://www.elm-tech.com/en/products/spi-flash-memory/gd25q32/gd25q32.pdf).
I have 2 dumps taken within 2 hours of each other. I've used your rtltool.py to dump the flash over UART.
This archive contains both. [dump_ezviz.zip](https://github.com/kuba2k2/libretuya/files/9626626/dump_ezviz.zip)
The device is still running the original firmware. And I successfully flashed a dump using Image_Tool v2.3.1 to address 0x08000000.

P.S. The device doesn't fully boot unless connected to 220V.

[image]
[image]
[image]

mihsu81, 2022-09-22:
And here is a 4MB dump. 
[dump_ezviz_3.zip](https://github.com/kuba2k2/libretuya/files/9627823/dump_ezviz_3.zip)

kuba2k2, 2022-09-23:
Board support added in 50fea2a8f042db8f22435bef74c23d600548af2c. I've put some additional info [in the board docs](https://docs.libretuya.ml/boards/generic-rtl8710bx-4mb-980k/#information).

mihsu81, 2022-09-23:
@kuba2k2 It threw an error because of the names of the ld files, but after changing them from ```rlx8711B-symbol-v02-img2_xip1_4M_980k_cpp.ld``` to ```rlx8711B-symbol-v02-img2_xip1_4M_cpp.ld``` and from ```rlx8711B-symbol-v02-img2_xip2_4M_980k_cpp.ld``` to ```rlx8711B-symbol-v02-img2_xip2_4M_cpp.ld``` it linked them just fine.

```
Linking .pioenvs\plug-ezviz-test1\firmware.elf
c:/users/bethayn-dell/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file rlx8711B-symbol-v02-img2_xip1_4M_cpp.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
|-- Image 1: firmware.ota1.elf
Generated file not found: .pioenvs\plug-ezviz-test1\firmware.ota1.elf
*** [.pioenvs\plug-ezviz-test1\firmware.elf] Error 1
============================================= [FAILED] Took 103.92 seconds =============================================
```

I was able to upload the uf2 file with ltchiptool and it worked fine. It connects to Wi-Fi.
Now I'll need to configure it.

closed 2022-09-23 by mihsu81