libretiny-eu/libretiny #25: generic-rtl8710bx-4mb-980k - cannot boot or enter flash mode after OTA update

mihsu81, 2022-10-01:
Hi @kuba2k2,

I had removed a binary_sensor from the yaml configuration file but it kept reappearing in the firmware.
So, I decided to erase with ImageTool all 4MB of the SPI flash. I flashed the last firmware.uf2 but it wouldn't boot.
I flashed a 2MB backup of the EZVIZ firmware and after that the firmware.uf2 and all went ok, the device booted up fine,
A couople of minutes later I tried performing an OTA update of a slightly modified yaml config (different pin for the relay) and realised that the device wasn't responding. 
This is all I'm getting now over serial, and the device won't enter flashing mode:
```
<RTL8195A>
ROM:[V0.1]

FLASHRATE:4

<RTL81ù
<RTL8195A>
ROM:[V0.1]

FLASHRATE:4
```
Did the SPI flash got wiped somehow together with the bootloader?
Is there a way to flash again the SPI flash from the full dumps I have of the original firmware?

The log after the last successful UART flash is below.
```
[22:03:44]<RTL8195A>ROM:[V0.1]
[22:03:44]FLASHRATE:4
[22:03:44]BOOT TYPE:0 XTAL:40000000
[22:03:44]IMG1 DATA[1128:10002000]
[22:03:44]IMG1 ENTRY[8000521:100021ef]
[22:03:44]IMG1 ENTER
[22:03:44]CHIPID[000000f6]
[22:03:44]read_mode idx:3, flash_speed idx:3
[22:03:44]calibration_result:[1:25:13][1:19]
[22:03:44]calibration_result:[2:25:13][1:19]
[22:03:44]calibration_result:[3:0:0][ff:ff]
[22:03:44]calibration_ok:[1:25:13]
[22:03:44]FLASH CALIB[NEW OK]
[22:03:44]OTA2 ADDR[8100000]
[22:03:44]OTAx SELE[fffffffe]
[22:03:44]OTA2 USE
[22:03:44]OTA2 SIGN[35393138:31313738]
[22:03:44]IMG2 DATA[0x818bbb0:3812:0x10005000]
[22:03:44]IMG2 SIGN[RTKWin(10005008)]
[22:03:44]IMG2 ENTRY[0x10005000:0x81309f1]
[22:03:44]System_Init1
[22:03:44]System_Init2
[22:03:44]I [      0.000] main():54: - LibreTuya v0.10.0+sha.b404f9f.dirty on generic-rtl8710bx-4mb-980k, compiled at Oct  1 2022 21:55:18
[22:03:44]I [      0.000] main():58: - Reset reason: 0
[22:03:44]I [      0.002] open():81: mainTask: Preferences initialized
[22:03:44][I][logger:283]: Log initialized
[22:03:44][C][status_led:014]: Setting up Status LED...
[22:03:44][C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[22:03:44][D][lt.preferences:104]: Saving 1 preferences to flash...
[22:03:44][V][lt.preferences:115]: sync: key: 233825507, len: 4
[22:03:44][D][lt.preferences:133]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:03:44][I][app:029]: Running through setup()...
[22:03:44][V][app:030]: Sorting components by setup priority...
[22:03:44][C][switch.gpio:011]: Setting up GPIO Switch 'PLUG-EZVIZ'...
[22:03:44][D][switch:017]: 'PLUG-EZVIZ' Turning OFF.
[22:03:44][D][switch:037]: 'PLUG-EZVIZ': Sending state OFF→<RTL8195A>ROM:[V0.1]
[22:03:44]FLASHRATE:4
[22:03:44]BOOT TYPE:0 XTAL:40000000
[22:03:44]IMG1 DATA[1128:10002000]
[22:03:44]IMG1 ENTRY[8000521:100021ef]
[22:03:44]IMG1 ENTER
[22:03:44]CHIPID[000000f6]
[22:03:44]read_mode idx:3, flash_speed idx:3
←[31mERROR Serial port closed!←[0m
```

kuba2k2, 2022-10-02:
If you wiped the flash, you erased the bootloader. You need to restore at least 0xB000 bytes from the backup dumps you've made. I don't know why would the device suddenly stop working, but try restoring everything first to see if that helps.

mihsu81, 2022-10-02:
It's not the first time I wipe the flash. I started the wipe at 0x08000000, and after the wipe I was able to enter download mode.
The issue is that after the last OTA update the device won't enter download mod anymore.
Any suggestion on how I can get download mode to work again? Or a way to flash the SPI module with a JLink?

Thanks in advance.

[image]

kuba2k2, 2022-10-02:
It's not possible to break the download mode - it's hard-baked into the chip's ROM. If you can't enter it, you're probably doing something wrong - i.e. maybe a wire cut loose, or some other connection problem.

If your module's SPI flash is easily accessible, and you have a SOIC8 programming clip (or you want to solder wires directly or temporarily desolder the chip), you can use flashrom on a Raspberry Pi, or a dedicated programmer if you have one.
This gives you access to the entire flash, just like the download mode.

mihsu81, 2022-10-03:
Thanks a lot for the hints. Indeed, it looks like I had shorted 2 pins of the CPU when checking the pins of the HLW8012.
Strange though that the issue only appeared a couple of hours later.

closed 2022-10-03 by mihsu81