libretiny-eu/libretiny #35: error: 'lwip_inet_ntop' was not declared in this scope

flohav, 2022-10-27:
When trying to compile esphome, if "api:" is declared in yaml config file, the following compilation error occures :

`src/esphome/components/socket/lwip_sockets_impl.cpp: In function 'std::__cxx11::string esphome::socket::format_sockaddr(const sockaddr_storage&)':
src/esphome/components/socket/lwip_sockets_impl.cpp:16:80: error: 'lwip_inet_ntop' was not declared in this scope
     const char *ret = lwip_inet_ntop(AF_INET, &addr->sin_addr, buf, sizeof(buf));
                                                                                ^
src/esphome/components/socket/lwip_sockets_impl.cpp: In member function 'virtual ssize_t esphome::socket::LwIPSocketImpl::readv(const iovec*, int)':
src/esphome/components/socket/lwip_sockets_impl.cpp:82:99: error: 'lwip_readv' was not declared in this scope
   ssize_t readv(const struct iovec *iov, int iovcnt) override { return lwip_readv(fd_, iov, iovcnt); }
                                                                                                   ^
*** [.pioenvs\yourdevice\src\esphome\components\socket\lwip_sockets_impl.cpp.o] Error 1`

After adding 
`socket:
    implementation: lwip_tcp`

it compiles fine

kuba2k2, 2022-11-26:
Fixed in 1172c1a. I'd appreciate testing it, as I don't even know if it works myself (I have a single RTL device that I can use, and it's powered by mains so debugging it is a bit harder).

flohav, 2022-12-11:
> Fixed in [1172c1a](https://github.com/kuba2k2/libretuya/commit/1172c1a852cccd394946bc0d00eac0ce54dca76a). I'd appreciate testing it, as I don't even know if it works myself (I have a single RTL device that I can use, and it's powered by mains so debugging it is a bit harder).
> 
> You don't need the `socket:` line now, in fact it might not work if you add it.

I confirm it compiles and runs fine without the `socket:` line in 0.11. 👍(actually it also still works if I leave it)

closed 2022-12-11 by kuba2k2