libretiny-eu/libretiny #36: Unexpected error with generic-rtl8720cf-2mb-992k board

rkorn86, 2022-11-03:
Hi there,

as stated on https://docs.libretuya.ml/boards/generic-rtl8720cf-2mb-992k/, this board should be supported.

Adopting the ESPHome example from https://docs.libretuya.ml/docs/projects/esphome/ and changing `board` accordingly results in an error:

```
$ python -m esphome compile example.yml 
INFO Reading configuration example.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing example (board: generic-rtl8720cf-2mb-992k; framework: arduino; platform: libretuya)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/usr/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/run/cli.py", line 144, in cli
    process_env(
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/run/cli.py", line 201, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/run/processor.py", line 83, in process
    install_project_env_dependencies(
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 131, in install_project_env_dependencies
    _install_project_env_platform(project_env, options),
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 148, in _install_project_env_platform
    PlatformPackageManager().install(
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/package/manager/platform.py", line 60, in install
    p.configure_project_packages(project_env, project_targets)
  File "/home/robert/.local/lib/python3.10/site-packages/platformio/platform/base.py", line 184, in configure_project_packages
    self.configure_default_packages(options, targets or [])
  File "/home/robert/.platformio/platforms/libretuya/platform.py", line 132, in configure_default_packages
    framework = next(fw for fw in frameworks if framework in fw)
StopIteration

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

Other generic boards, like `generic-rtl8710bx-4mb-980k`, compile without errors.

Do you have any idea?

kuba2k2, 2022-11-03:
No, this board is not supported. As stated in "board families", this family doesn't support the Arduino framework yet, which is needed for ESPHome.

Regarding the second issue, this means that LibreTuya you're using is outdated. See "getting started" on how to install the latest version.

rkorn86, 2022-11-03:
Thanks for clarification.
Regarding second issue: I am using the latest version of https://github.com/kuba2k2/libretuya-esphome commit `090809cfb6affa3b64a9e715dac5e337295feabc (HEAD -> platform/libretuya, origin/platform/libretuya, origin/HEAD)` and installed platformio platform using `platformio platform install -f https://github.com/kuba2k2/libretuya`

Nethertheless thanks for your great work. Seems, like I have to resolder this goddamn WBR3 with an ESP or just return my `ME81H
 Floor Heating Thermostats`

kuba2k2, 2022-11-03:
Well, whatever happened during the compilation made LT install the older version. These two functions `wdtEnable` and `wdtFeed` were introduced some time ago, so if you update again using `platformio platform install -f https://github.com/kuba2k2/libretuya` it will work just fine.

> Nethertheless thanks for your great work. Seems, like I have to resolder this goddamn WBR3 with an ESP or just return my ME81H Floor Heating Thermostats

If you don't mind sticking with the Tuya firmware for some more time, you can wait until LT supports that. (yes, it's planned as well)

rkorn86, 2022-11-04:
Sounds great. I will apply as beta tester, as I am not in a hurry with replacing my old thermostats.

PS: running `platformio platform install -f https://github.com/kuba2k2/libretuya` again did the trick - example successfully compiled - `INFO Successfully compiled program.`

kuba2k2, 2023-01-19:
Closing this as duplicate of #44.

closed 2023-01-19 by kuba2k2