libretiny-eu/libretiny #39: BL602

Minims, 2022-11-15:
Hello,

will you support BL602 chips too in the future ?

kuba2k2, 2022-11-16:
Yes, that's a plan, but it's of the lowest priority.

kuba2k2, 2022-11-16:
To clarify: BL602 boards are not really popular, there's a much higher demand on RTL8720Cx chips (as well as RTL8720Dx). When I find some time to work on new platforms, that's probably where I'll start. If you (or anybody) are feeling brave enough, you can always try writing the new platform port yourself, based on the already available platforms.

blakadder, 2022-11-25:
I'd just chime in on that with a remark that Sonoff is keen on using BL602 chips in some of their devices like bulbs and the MINIR3

ardenking, 2022-12-06:
Was about to post the same as they avail as a dev board

kuba2k2, 2022-12-06:
I see! Indeed a more interesting topic :)
There's even a partial Arduino core.

Minims, 2023-08-20:
Hello, Is there any news or ETA for bl602 ? 
I don’t know if this can help
https://github.com/openshwprojects/OpenBK7231T_App

kuba2k2, 2023-08-21:
I could try to start working on BL602 core in a few days. But for now it's more important to address the changes with ESPHome that were requested in the PR review.

Minims, 2023-08-22:
Thanks !! will wait a bit more, i know there is some priority :-)

4rianton, 2023-09-08:
Just received a BL602 board and a BL0937 energy meter combo in a socket from Aliexpress. Would be looking forward to any updates here :)

BadAd84, 2023-10-11:
Also seeing more BL602 devices, led controllers etc. Bought a bunch that were previously ESP, but now switched to BL602. Congrats on the HA merge too!

wenoptics, 2023-10-11:
Also eager to see progress on BL602.

Would like to help to move this forward, any suggestions? I have a few LED controllers and switches (Sonoff S40) with BL602

Minims, 2024-02-18:
@kuba2k2 Do you have any update on the BL602 integration ? Thx

cap9qd, 2024-03-05:
@kuba2k2 I just got a device from Ali that has the BL602 so if you need any help testing please get in touch.

aletzi1, 2024-04-10:
I also have couple of bl602 devices. Just writing to get notification when we have breakthrough 🤭

Minims, 2024-12-03:
Hello @kuba2k2 ,

Any update on BL602 integration ? Or this is definitely cancelled ? 

Thx

kuba2k2, 2024-12-03:
No, I don't have any updates.
Due to work and university assignments, I don't have that much time for other projects these days, and that includes LibreTiny.
I'm not saying the project won't be developed in the future, because it certainly will at some point.
But if you're waiting for LibreTiny to be ready in order to use a particular device with BL602, you should consider using OpenBeken on that device instead.

nschimme, 2025-08-20:
For anyone looking into this, there is a related feature-request here to be aware of: https://github.com/esphome/feature-requests/issues/1049

The original BL602 SDK seems unmaintained and there is a community fork now at: https://github.com/openshwprojects/OpenBL602

Xmister, 2026-11-02:
Let me leave this here, more info later

[image]

Xmister, 2026-11-05:
Works pretty stable for me for a few days now.
Changes:
https://github.com/Xmister/esphome/pull/1
https://github.com/Xmister/libretuya/pull/1
https://github.com/Xmister/ltchiptool/pull/1

Using that specific esphome version with the example yaml will pull in the proper libretiny changes.
ltchiptool can be used to flash the image.
GPIO8 needs to be connected to 3.3 to boot into flash mode.
OTA works fine after that.
BLE and most of what I tried works fine.

felipecrs, 2026-11-05:
That's great news, @Xmister! I hope you are able to upstream all these contributions smoothly.

Looking forward to being able to re-flash some smart plugs I have lying around.