libretiny-eu/libretiny #40: RTL8710BN - Stuck at reboot

flohav, 2022-11-16:
Everytime I send an OTA update or ask for a reboot from Home Assistant (software "Restart" button), my RTL8710BN based device is stuck.
Here are the logs (level: VERY_VERBOSE):

```
[I][restart.button:012]: Restarting device...
[I][app:134]: Rebooting safely...
[VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[D][lt.preferences:104]: Saving 4 preferences to flash...
[VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[V][lt.preferences:115]: sync: key: 233825507, len: 4
[VV][lt.preferences:113]: Checking if FDB data 3130601430 has changed
[V][lt.preferences:115]: sync: key: 3130601430, len: 44
[VV][lt.preferences:113]: Checking if FDB data 2162300045 has changed
[V][lt.preferences:115]: sync: key: 2162300045, len: 44
[VV][lt.preferences:113]: Checking if FDB data 3551080420 has changed
[V][lt.preferences:115]: sync: key: 3551080420, len: 1
[D][lt.preferences:133]: Saving 4 preferences to flash: 0 cached, 4 written, 0 failed
[VV][api.service:025]: send_disconnect_request: DisconnectRequest {}
[VV][api.service:025]: send_disconnect_request: DisconnectRequest {}
===== Enter Image 1.5 ====
Img2 Sign: RTKWin, InfaStart @ 0x080f9fbd
System_Init1
System_Init2
I [      0.000] LibreTuya v0.10.0+sha.a79a2d6 on wr3, compiled at Nov 16 2022 14:50:56
[I][logger:283]: Log initialized
[VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[D][lt.preferences:104]: Saving 1 preferences to flash...
[VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[V][lt.preferences:115]: sync: key: 233825507, len: 4
```

The device is stuck at this step, so I do a hard reboot (unplug/replug power) and it goes fine:

```
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:3, flash_speed idx:0
calibration_result:[1:5:11][9:d]
calibration_result:[2:13:7][1:d]
calibration_result:[3:7:5][1:7]
calibration_ok:[2:13:7]
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[ffffffc0]
OTA1 USE
IMG2 DATA[0x80774a0:3672:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x80314d5]
System_Init1
System_Init2
I [      0.000] LibreTuya v0.10.0+sha.a79a2d6 on wr3, compiled at Nov  8 2022 12:01:09
[I][logger:283]: Log initialized
[VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[D][lt.preferences:104]: Saving 1 preferences to flash...
[VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[V][lt.preferences:115]: sync: key: 233825507, len: 4
[D][lt.preferences:133]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
```
=> Everything goes well after that hard reboot. It happens everytime.

kuba2k2, 2022-11-16:
There's something weird going on here: the first log shows a build date of Nov 16 2022, while the second one is Nov  8 2022.
Are these logs from an OTA attempt? Or just a software reset without upgrading anything?

flohav, 2022-11-16:
Oh yes sorry I made several attemps and swaped the logs, but the behaviour is the same. I can make another extract with correct data if needed.

The logs an behaviour are the same from an OTA attempt than from a software reset without updrading anything.
The OTA goes well by the way, after the hard reboot it get the upragdes.

kuba2k2, 2022-11-16:
Fine, I don't need exact logs, just wanted to figure out if this was about OTA going wrong.

This part is concerning:
```
===== Enter Image 1.5 ====
Img2 Sign: RTKWin, InfaStart @ 0x080f9fbd
System_Init1
System_Init2
```

I don't usually see that when cold-booting the chip. The current method of rebooting is the second I've tried, and apparently also doesn't work. Try changing the `arduino/realtek-ambz/cores/arduino/LibreTuyaAPI.cpp` file locally:
- remove `sys_cpu_reset()` on line 13 - so that only `sys_reset()` is left

You can find the file in `~/.platformio/platforms/libretuya/`

flohav, 2022-11-16:
Actually, there is something strange with the build dates, here is a log output from a build I made at 15:10 today:

```
[D][ota:143]: Starting OTA Update from 192.168.1.10...
[D][ota:312]: OTA in progress: 0.1%
[D][ota:312]: OTA in progress: 41.3%
[D][ota:312]: OTA in progress: 80.9%
[I][ota:345]: OTA update finished!
[I][app:134]: Rebooting safely...
===== Enter Image 1.5 ====
Img2 Sign: RTKWin, InfaStart @ 0x080314d5
System_Init1
System_Init2
I [      0.000] LibreTuya v0.10.0+sha.a79a2d6 on wr3, compiled at Nov 16 2022 14:53:52
[I][logger:283]: Log initialized
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[D][lt.preferences:104]: Saving 1 preferences to flash...
```

Stuck, hard reboot:

```
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1168:10002000]
IMG1 ENTRY[8000541:100021ef]
IMG1 ENTER
CHIPID[000000ff]
read_mode idx:3, flash_speed idx:0
calibration_result:[1:5:11][9:d]
calibration_result:[2:13:7][1:d]
calibration_result:[3:7:5][1:7]
calibration_ok:[2:13:7]
FLASH CALIB[NEW OK]
OTA2 ADDR[80d0000]
OTAx SELE[fffffe00]
OTA2 USE
OTA2 SIGN[35393138:31313738]
IMG2 DATA[0x8142a90:3672:0x10005000]
IMG2 SIGN[RTKWin(10005008)]
IMG2 ENTRY[0x10005000:0x80f9fbd]
System_Init1
System_Init2
I [      0.000] LibreTuya v0.10.0+sha.a79a2d6 on wr3, compiled at Nov 16 2022 15:10:51
[I][logger:283]: Log initialized
[VV][lt.preferences:064]: fdb_kv_get_blob: key: 233825507, len: 4
[C][ota:473]: There have been 0 suspected unsuccessful boot attempts.
[VV][lt.preferences:041]: s_pending_save: key: 233825507, len: 4
[D][lt.preferences:104]: Saving 1 preferences to flash...
[VV][lt.preferences:113]: Checking if FDB data 233825507 has changed
[V][lt.preferences:115]: sync: key: 233825507, len: 4
[D][lt.preferences:133]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
```

flohav, 2022-11-16:
> Fine, I don't need exact logs, just wanted to figure out if this was about OTA going wrong.
>
> This part is concerning:
>
> ```
> ===== Enter Image 1.5 ====
> Img2 Sign: RTKWin, InfaStart @ 0x080f9fbd [...]

Works perfectly! Thank you!

kuba2k2, 2022-11-16:
Really? That's it...? Huh.
If you test OTA, as well as simple software reboots, I'll merge that change into master.

mihsu81, 2022-11-17:
Doesn't seem to make any difference on my "generic-rtl8710bx-4mb-980k" (EZVIZ TT1 Smart Plug CS-T31-16B-EU), which needs to be manually reset after each OTA or after pressing the restart button.

kuba2k2, 2022-11-18:
Yeah, that's what I thought. Once it works, then it doesn't...

mihsu81, 2022-11-20:
After finalizing the configuration of my test device (EZVIZ T31), making that change in ```arduino/realtek-ambz/cores/arduino/LibreTuyaAPI.cpp``` seems to have helped with OTA reboots, which now get the device back up consistently.
Restarts triggered by using the _**restart**_ platform still put the device in an unusable state.

kuba2k2, 2022-11-26:
@flohav @mihsu81 That should be fixed in 9e4bbe3e36e0953f0361f4e713069e656d475b74, *this time for good*.

Tested on a RTL8710BN (WR3), along with ESPHome (after today's updates), using OTA as well as a soft-reboot button.

flohav, 2022-12-11:
> @flohav @mihsu81 That should be fixed in [9e4bbe3](https://github.com/kuba2k2/libretuya/commit/9e4bbe3e36e0953f0361f4e713069e656d475b74), _this time for good_.
> 
> Tested on a RTL8710BN (WR3), along with ESPHome (after today's updates), using OTA as well as a soft-reboot button.

I confirm, working on 0.11 using OTA as was as software button.
Thanks !

closed 2022-12-11 by kuba2k2