libretiny-eu/libretiny #41: CB2S Not restoring saved state consistently

descipher, 2022-11-18:
Noticed that preferences are not saved consistently.

  **this->rtc_.save(&this->state);**
  ESP_LOGD(TAG, "'%s': Sending state %s", this->name_.c_str(), ONOFF(this->state));
  this->state_callback_.call(this->state);

On power loss the last state does not always persist.

Device: Aubess MINI with the BL0942.

yaml:

~~~
esphome:
  name: relay-x1-2

libretuya:
  board: cb2s
  framework:
    version: latest

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: 
  password: 

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: 
    password: 

web_server:
  port: 80

output:
  - platform: gpio
    id: led1
    inverted: True
    pin: D2

# relay output, exposed as switches in Home Assistant
switch:
  - platform: gpio
    pin: D7
    name: Relay-x1-2
    id: relay_x1_2
    on_turn_on:
      - output.turn_on: led1
    on_turn_off:
      - output.turn_off: led1
    restore_mode: RESTORE_DEFAULT_OFF

binary_sensor:
  - platform: gpio
    pin:
      number: D3
      mode: INPUT_PULLUP
    id: button1
    on_press:
      then:
        - switch.toggle: relay_x1_2
  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
    id: button2
    on_press:
      then:
        - switch.toggle: relay_x1_2

uart:
  id: uart_bus
  tx_pin: D5
  rx_pin: D4
  baud_rate: 4800
  stop_bits: 1

sensor:
  - platform: bl0942
    uart_id: uart_bus
    voltage:
      name: 'Relay-x1-2 Voltage'
    current:
      name: 'Relay-x1-2 Current'
    power:
      name: 'Relay-x1-2 Power'
      filters:
        multiply: -1
    energy:
      name: 'Relay-x1-2 Energy'
    frequency:
      name: "Relay-x1-2 Frequency"
      accuracy_decimals: 2

    update_interval: 30s
~~~

kuba2k2, 2022-11-18:
Note how it says "RTC" in the code you highlighted. Beken chips, unlike ESP, don't have any RTC memory for data preservation. Even if they did, a power loss would make it go away.

Thus, preferences are stored in the flash memory. Because of wear leveling, they are saved every 10 minutes or so, not upon each change. I believe ESP chips do the same and that they won't recover the very latest state after a power loss either.

descipher, 2022-11-18:
Ok, found the save interval define, it 60 seconds, that works for almost every case, unfortunatly its "global" not sure thats the best way to handle it. Also restore_from_flash appears to be required for esp8266 based devices.

descipher, 2022-11-18:
Not a bug in any case, closing this, thanks.

descipher, 2022-11-18:
Retested with the new parameters, we do have a bug. It failed to save an off state after 10min. Seems to be good on saving an on state consistently but not always an off state.

descipher, 2022-11-18:
It's repeatable.
1. Starting from a off state after a power cycle.
2. Set state to on and wait 60+ seconds.
3. Power cycle and verify the on state occurs.
4. Set state to off and wait 60+ seconds.
5. Power cylcle. State save of off fails an will be on in the on state. (this will occur every time)
6. Now from the last point set the state to off, immediately followed by on and then finnish with an off state waiting 60+ seconds.
7. Power cycle, the state will be succesfully restored as the off state.

descipher, 2022-11-19:
Parameters were in error, it works as expected given enough time.

closed 2022-11-19 by descipher