libretiny-eu/libretiny #43: Add RFIO pin for generic-rtl8710bx-4mb-980k

mihsu81, 2022-11-20:
Hi @kuba2k2,

Please add the RFIO pin (Pin 9) present in the RTL8710BX (generic-rtl8710bx-4mb-980k). In my device EZVIZ T31 it is mapped to a physical button and was used by EZVIZ to trigger a reset of the SSID.
https://jlcpcb.com/partdetail/RealtekSemicon-RTL8710BX_A0CG/C398719

[image]

kuba2k2, 2022-11-20:
RFIO is not a GPIO input/output. Are you absolutely sure this pin is mapped to a button?

mihsu81, 2022-11-20:
It seemed to be. I've verified the continuity between that pin and Pin1 of the physical switch.

kuba2k2, 2022-11-20:
Isn't that pin GND by any chance? RFIO may be closely wired to GND so your continuity tester may show it's connected.

mihsu81, 2022-11-20:
You're absolutely right. I guess I'll have to do more tests to identify the switch. Thanks.

closed 2022-11-20 by mihsu81