libretiny-eu/libretiny #46: esphome + tuya causes unresponsive device

rickyelopez, 2022-11-27:
Just flashed a BK7321T (WB3S) module with libretuya and it seems to be working, until I add the `tuya:` module to my esphome yaml, after which the API becomes unresponsive. I can still upload new firmware with esphome, so it's not like it's completely bricked. I guess it's probably boot looping?

Is the TuyaMCU module not supported yet? Serial seems to be working so I figured this would too, but I guess not?

kuba2k2, 2022-11-27:
We haven't tested TuyaMCU yet. What is the exact config you're using? Maybe something's wrong with the UART settings.

rickyelopez, 2022-11-27:
Ah ok, that's unfortunate. Hopefully I messed something up. I'm using the same config I used for the version of this that has an esp8266.

```yaml
substitutions:
  device_name: mtl-light-dining-room-fixture
  friendly_name: Dining Room Light Fixture
  ip_addr: 10.5.25.144
  icon: "mdi:chandelier"

esphome:
  name: ${device_name}

libretuya:
  board: wb3s
  framework:
    version: latest

# Make sure logging is not using the serial port
logger:
  baud_rate: 0
  # level: VERY_VERBOSE

api:
  encryption:
    key: !secret encryption_key

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid_mtl
  password: !secret wifi_password_mtl
  use_address: ${ip_addr}
  domain: !secret domain_mtl
  fast_connect: on

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ${device_name}
    password: !secret fallback_ap_password_mtl

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200
  debug:
  # rx_buffer_size: 512

# tuya:

# light:
#   - platform: tuya
#     name: ${friendly_name}
#     dimmer_datapoint: 2
#     min_value_datapoint: 3
#     switch_datapoint: 1
#     min_value: 100
#     max_value: 1000

# [17:20:33][C][tuya:048]:   Datapoint 1: switch (value: ON)     -- Switch State
# [17:20:33][C][tuya:050]:   Datapoint 3: int value (value: 100) -- Minimum value
# [17:20:33][C][tuya:050]:   Datapoint 102: int value (value: 0)
# [17:20:33][C][tuya:054]:   Datapoint 4: enum (value: 0)        -- Dimming mode
# [17:20:33][C][tuya:050]:   Datapoint 104: int value (value: 1)
# [17:20:33][C][tuya:050]:   Datapoint 2: int value (value: 722) -- Dimmer value
# [17:20:33][C][tuya:046]:   Datapoint 101: raw (value: 00.00.00)
# [17:20:33][C][tuya:046]:   Datapoint 105: raw (value: 06.00.00)
```

kuba2k2, 2022-11-27:
As per [WB3S docs on LibreTuya](https://docs.libretuya.ml/boards/wb3s/), UART pins are D10+D11 (UART1) and D6+D7 (UART2).
Your `GPIO1` and `GPIO3` don't really point to anything. How do you know where the TuyaMCU is connected?

rickyelopez, 2022-11-27:
Ahhhh thank you for pointing that out, I think I just hit the same problem on the SS01S. Will test and report back.

rickyelopez, 2022-11-27:
Yup that fixed it, thanks for the quick help as usual! Out of curiosity, do you know if that's the failure mode that regular esphome has? It just doesn't start up? Seems strange to me, because as soon as I comment out the `tuya:` line, it boots fine, even with the wrong uart pins. Anyway, not an issue. Thanks again!

closed 2022-11-27 by rickyelopez

kuba2k2, 2022-11-27:
I don't know. Maybe it waits for something on the UART line, or maybe the pins are used by something else and that creates problems.