libretiny-eu/libretiny #49: [CB2S] Issue with power readings

dj-fiorex, 2022-12-08:
hello to everyone, i have a Smart Plug marked Elivco LSPA9, thanks to [this website](https://zry.io/archives/799)  i know that it has a BL0937 power meter connected with pin 34, 7, 6. I already flashed it with libretuya-esphome and i can turn on/off but i can't read the power meter. This is my yml config if someone could help me
```
esphome:
  name: elivcolspa9cb2s

libretuya:
  board: cb2s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""
  
# Enable web server
web_server:
  port: 80

wifi:
  ssid: ""
  password: ""
  manual_ip:
    static_ip: x.x.x.188
    gateway: x.x.x.1
    subnet: x.x.x.0

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Elivcolspa9Cb2SFallbackHotspot"
    password: "IHxBKsc7WQM4"
    
switch:  # the socket relay
  - platform: gpio
    pin: D7
    id: plug_1
    name: 'Plug 1'
    restore_mode: RESTORE_DEFAULT_OFF  # attempt to restore state and default to OFF if not possible to restore

binary_sensor:  # the button
  - platform: gpio
    pin:
      number: D4
      inverted: true
    id: button
    internal: true  # don't expose it to HASS, only use it locally
    on_release:
      then:
        - switch.toggle: plug_1  # toggle the relay

sensor:
  - platform: hlw8012
    model: BL0937  # note that the model must be specified to use special calculation parameters
    voltage_divider: 1600  # adjust it according to the actual resistor values on board
    sel_pin:
      number: D6
      inverted: true  # the logic of BL0937 is opposite from HLW8012
    cf_pin: D1
    cf1_pin: D0
    current:
      name: 'Plug 1 Current'
    voltage:
      name: 'Plug 1 Voltage'
    power:
      name: 'Plug 1 Power'
    energy:
      name: 'Plug 1 Energy'
      # convert it to kWh
      filters:
        - multiply: 0.001
      unit_of_measurement: 'kWh'
      accuracy_decimals: 4
    update_interval: 60s
    change_mode_every: 3

  
```

descipher, 2022-12-08:
p24 = pin 16 on the BK7231N qfn32 which is virtually mapped to arduino D3 in libretuya.

descipher, 2022-12-08:
also p6 and p7 are not D0 and D1

cb2s.json
```
			},
			"13": {
				"IC": 24,
				"ARD": "D8"
			},
			"14": {
				"IC": 23,
				"ARD": "D9"
```

kuba2k2, 2022-12-08:
There's no pin 34, so I believe this is a typo.

@descipher If you take a look at the docs [here](https://docs.libretuya.ml/boards/cb2s/), you can see that D0 and D1 are indeed P6 and P7.

Since yesterday, you can actually use the P* pins directly in esphome config (make sure you have the latest commits for both esphome and LibreTuya).

dj-fiorex, 2022-12-08:
Sorry guys, i don't understand, is there an error in my config? Or this type of power meter is not supported?

Thanks

kuba2k2, 2022-12-08:
You can try adding a basic pulse_counter component, and see if it reports anything:
```yaml
sensor:
  - platform: pulse_counter
    pin:
      number: D0
      mode: input
    name: "D0 Pulse Counter"
    update_interval: 10s
  - platform: pulse_counter
    pin:
      number: D1
      mode: input
    name: "D1 Pulse Counter"
    update_interval: 10s
```
(adjust the pin numbers as needed)

dj-fiorex, 2022-12-08:
Tried with your suggests but i receive 0 pulses/min in the two pins, so i think the problem might be in pulse_counter also

dj-fiorex, 2022-12-27:
Thanks to the help of @catalin2402 i successfully read the current/voltage/power from that BL0937
I will post the updated yaml soon, the trick was to invert cf and cf1 pin

catalin2402, 2022-12-27:
more precise: inverted signal on CF and CF1

dj-fiorex, 2022-12-29:
Like promised it is the yaml
```
sensor:
  - platform: hlw8012
    model: BL0937  # note that the model must be specified to use special calculation parameters
    voltage_divider: 780  # adjust it according to the actual resistor values on board
    current_resistor: 0.00107
    # initial_mode: "CURRENT"
    # change_mode_every: 4294967295 
    update_interval: 10s
    sel_pin:
      number: P24
      # mode:
      #  input: true
      #  pullup: true
      inverted: true  # the logic of BL0937 is opposite from HLW8012
    cf_pin:
      number: P7
      inverted: true
      mode:
        input: true
        pulldown: true
    cf1_pin:
      number: P6
      inverted: true
      mode:
        input: true
        pulldown: true  
    current:
      name: 'Plug 1 Current'
    voltage:
      name: 'Plug 1 Voltage'
    power:
      name: 'Plug 1 Power'
    energy:
      name: 'Plug 1 Energy'
      # convert it to kWh
      filters:
        - multiply: 0.001
      unit_of_measurement: 'kWh'
      accuracy_decimals: 4
```

closed 2023-01-01 by dj-fiorex