libretiny-eu/libretiny #50: PWM output on libretuya

hunhejj, 2022-12-22:
I have a light bulb with a BK7231N board in it. How can I define PWM outputs in ESPHome for using it for a light component?  More precisely which platform should I use for the output?

```yaml
# ...

light:
  - platform: rgbww
    name: "my_room"
    red: output_component1
    green: output_component2
    blue: output_component3
    cold_white: output_component4
    warm_white: output_component5
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K

output:
  - platform: ???
    id: output_component1
    pin: P6
  #...
```

kuba2k2, 2022-12-22:
Use `ledc`.

hunhejj, 2022-12-22:
Thank you for the quick answer @kuba2k2, that indeed worked! Crossing fingers that esphome finally approves your MR-s and your awesome work will officially be part of esphome!

closed 2022-12-22 by hunhejj