libretiny-eu/libretiny #52: error installing platformio platform

dj-fiorex, 2022-12-30:
```
platformio platform uninstall https://github.com/kuba2k2/libretuya

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg uninstall` instead.

Error: Could not find the package with 'https://github.com/kuba2k2/libretuya' requirements for your system 'linux_x86_64'
carmelo@carmelo-desktop:~/Desktop/tuya-cloudcutter$ platformio platform install -f https://github.com/kuba2k2/libretuya

WARNING: This command is deprecated and will be removed in the next releases. 
Please use `pio pkg install` instead.

Platform Manager: Installing git+https://github.com/kuba2k2/libretuya
git version 2.34.1
Cloning into '/home/carmelo/.platformio/.cache/tmp/pkg-installing-i8188zqg'...
remote: Enumerating objects: 650, done.
remote: Counting objects: 100% (650/650), done.
remote: Compressing objects: 100% (446/446), done.
remote: Total 650 (delta 188), reused 529 (delta 173), pack-reused 0
Receiving objects: 100% (650/650), 438.17 KiB | 207.00 KiB/s, done.
Resolving deltas: 100% (188/188), done.
Platform Manager: libretuya@0.12.0+sha.b85f6f8 has been installed!
Installing/updating ltchiptool
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: ltchiptool in /home/carmelo/.local/lib/python3.10/site-packages (1.7.0)
Collecting ltchiptool
  Downloading ltchiptool-2.0.2-py3-none-any.whl (118 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 118.9/118.9 KB 213.1 kB/s eta 0:00:00
Requirement already satisfied: prettytable<4.0.0,>=3.3.0 in /home/carmelo/.local/lib/python3.10/site-packages (from ltchiptool) (3.5.0)
Requirement already satisfied: click<9.0.0,>=8.1.3 in /home/carmelo/.local/lib/python3.10/site-packages (from ltchiptool) (8.1.3)
Requirement already satisfied: importlib-metadata<5.0.0,>=4.12.0 in /home/carmelo/.local/lib/python3.10/site-packages (from ltchiptool) (4.13.0)
Requirement already satisfied: pycryptodomex<4.0.0,>=3.15.0 in /home/carmelo/.local/lib/python3.10/site-packages (from ltchiptool) (3.16.0)
Requirement already satisfied: colorama<0.5.0,>=0.4.5 in /home/carmelo/.local/lib/python3.10/site-packages (from ltchiptool) (0.4.6)
Collecting bk7231tools<2.0.0,>=1.2.1
  Downloading bk7231tools-1.2.3-py3-none-any.whl (27 kB)
Collecting xmodem<0.5.0,>=0.4.6
  Downloading xmodem-0.4.6.tar.gz (32 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: pyserial<4.0,>=3.5 in /home/carmelo/.local/lib/python3.10/site-packages (from bk7231tools<2.0.0,>=1.2.1->ltchiptool) (3.5)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3/dist-packages (from importlib-metadata<5.0.0,>=4.12.0->ltchiptool) (1.0.0)
Requirement already satisfied: wcwidth in /home/carmelo/.local/lib/python3.10/site-packages (from prettytable<4.0.0,>=3.3.0->ltchiptool) (0.2.5)
Building wheels for collected packages: xmodem
  Building wheel for xmodem (setup.py) ... done
  Created wheel for xmodem: filename=xmodem-0.4.6-py3-none-any.whl size=34564 sha256=18ed5de0122c8065d5317eb90efbe230352103e469b2e0bbdaf84843659b14e9
  Stored in directory: /home/carmelo/.cache/pip/wheels/19/00/c7/cff2d8b8981235aa42bcd09109a401169f9726df47518ada6b
Successfully built xmodem
Installing collected packages: xmodem, bk7231tools, ltchiptool
  Attempting uninstall: bk7231tools
    Found existing installation: bk7231tools 1.1.5
    Uninstalling bk7231tools-1.1.5:
      Successfully uninstalled bk7231tools-1.1.5
  Attempting uninstall: ltchiptool
    Found existing installation: ltchiptool 1.7.0
    Uninstalling ltchiptool-1.7.0:
      Successfully uninstalled ltchiptool-1.7.0
Successfully installed bk7231tools-1.2.3 ltchiptool-2.0.2 xmodem-0.4.6
Error: Traceback (most recent call last):
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/cli.py", line 71, in invoke
    return super().invoke(ctx)
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/commands/platform.py", line 177, in platform_install
    ctx.invoke(
  File "/home/carmelo/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 65, in package_install_cmd
    install_global_dependencies(options)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/package/commands/install.py", line 77, in install_global_dependencies
    pm.install(
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/package/manager/platform.py", line 52, in install
    p = PlatformFactory.new(pkg)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/platform/factory.py", line 78, in new
    cls.load_platform_module(
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/platform/factory.py", line 37, in load_platform_module
    return load_python_module("platformio.platform.%s" % name, path)
  File "/home/carmelo/.local/lib/python3.10/site-packages/platformio/compat.py", line 79, in load_python_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/carmelo/.platformio/platforms/libretuya/platform.py", line 48, in <module>
    ltchiptool.util.lt_set_path(dirname(__file__))
AttributeError: module 'ltchiptool.util' has no attribute 'lt_set_path'

```
ubuntu 22 not virtualized! worked before latest update

kuba2k2, 2022-12-30:
Post the output of `python -V`, `pip -V` and `ltchiptool -V`.

Cossid, 2022-12-30:
Confirming I get the same error.

```
PS C:\Dev\libretuya-esphome> python -V
Python 3.10.8
PS C:\Dev\libretuya-esphome> pip -V
pip 22.3.1 from C:\Users\Cossid\AppData\Local\Programs\Python\Python310\lib\site-packages\pip (python 3.10)
PS C:\Dev\libretuya-esphome> ltchiptool -V
ltchiptool v2.0.2
```

Cossid, 2022-12-30:
Strange, ran the install a second time and it looks like it worked fine.

kuba2k2, 2022-12-30:
Most likely `importlib.reload()` didn't work properly (it didn't load the new version of ltchiptool).

dj-fiorex, 2022-12-30:
i can confirm rerunning the installer fix the problem

closed 2022-12-30 by dj-fiorex