libretiny-eu/libretiny #54: Publish libretuya-esphome as a standalone Home Assistant add-on

felipecrs, 2023-01-01:
This should make things a lot easier for people who want to get started, so they can simply install it in their HassOS instead of having to tinker with PlatformIO, ESPHome cloning/installing/building and so on.

It should not be complicated, I think you only need to start building and publishing docker images from libretuya-esphome project and then fork https://github.com/esphome/home-assistant-addon renaming the add-on to something like ESPHome Libretuya and pointing to your own docker image.

What do you think?

kuba2k2, 2023-01-01:
It should be possible, but I'd need to set up a Docker registry and somehow push the images there. I'm not familiar with HA addons, do you just point it to the github repository and it installs the addon from there?

felipecrs, 2023-01-01:
I will send you some PRs. See you soon!

closed 2023-07-13 by kuba2k2

felipecrs, 2023-07-13:
Awesome!