libretiny-eu/libretiny #55: LibreTuya ESPHome fails to compile when using `time`

felipecrs, 2023-01-02:
If I put this on my LibreTuya ESPHome `device.yaml`:

```yaml
time:
  - platform: sntp
    id: my_time
```

The compilation fails with:

[image]

Full logs are:

```
INFO Reading configuration /config/libretuya-esphome/portao-tuya.yaml...
INFO Detected timezone 'America/Sao_Paulo'
INFO Generating C++ source...
INFO Compiling app...
Processing portao-tuya (board: wr2e; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-realtek-amb1 @ 750.0.0-b0bc+sha.750b0bc 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.5d6edaa 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/api_connection.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/api_server.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/list_entities.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/proto.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/api/user_services.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/button/button.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/factory_reset/button/factory_reset_button.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/json/json_util.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/libretuya/core.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/libretuya/gpio_arduino.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/libretuya/lt_component.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/libretuya/preferences.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/logger/logger.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/md5/md5.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/network/util.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/restart/button/restart_button.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/sensor/automation.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/sensor/filter.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/socket/socket.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/status/status_binary_sensor.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/text_sensor/filter.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/time/real_time_clock.cpp.o
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/uptime/uptime_sensor.cpp.o
src/esphome/components/time/real_time_clock.cpp: In member function 'void esphome::time::RealTimeClock::apply_timezone_()':
src/esphome/components/time/real_time_clock.cpp:51:42: error: 'setenv' was not declared in this scope
   setenv("TZ", this->timezone_.c_str(), 1);
                                          ^
src/esphome/components/time/real_time_clock.cpp:52:9: error: 'tzset' was not declared in this scope
   tzset();
         ^
Compiling /data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/web_server/list_entities.cpp.o
*** [/data/portao-tuya/.pioenvs/portao-tuya/src/esphome/components/time/real_time_clock.cpp.o] Error 1
========================= [FAILED] Took 80.30 seconds =========================
```

kuba2k2, 2023-01-02:
Most likely they changed something in ESPHome latest versions, which requires setenv() and tzset() now. I'll need to provide dummy implementations of these methods.

catalin2402, 2023-01-02:
I think this is something related to RTL chips, on Beken the compilation works without a problem using the sntp platform.

blakadder, 2023-01-02:
confirmed, compiled `time` for Beken yesterday with no problem

kuba2k2, 2023-01-02:
I don't know what even defines `tzset()` on Beken. VSCode can't find it (and underlines the function as undefined), but somehow it compiles properly. It's in `time.h` which has an `#ifndef __STRICT_ANSI__` statement, and `__STRICT_ANSI__` is apparently defined, so I don't know why that function still exists.

felipecrs, 2023-01-02:
Oh, I forgot to mention, this is my libretuya esphome config:

```mention
libretuya:
  board: wr2e
  framework:
    version: dev
```

kuba2k2, 2023-01-02:
Should be fixed in da95cc30d3818eea5a44aa10bdf34d947d3290f4.

felipecrs, 2023-01-02:
Is there anything I need to do to make ESPHome be aware of the fix?

kuba2k2, 2023-01-02:
Yes, you need to install LibreTuya again. `platformio platform install -f https://github.com/kuba2k2/libretuya` should do the job.

felipecrs, 2023-01-02:
Hm... As I'm using the hass add-on, I guess this is part of the docker build. I'll rebuild with no cache and try again.

kuba2k2, 2023-01-02:
No, this is part of PlatformIO. I don't know how to update PlatformIO packages within the container.

felipecrs, 2023-01-02:
I think PlatformIO packages are installed during docker build too:

https://github.com/kuba2k2/libretuya-esphome/blob/19970729a96203d3eec376936ae45e8f17a2ffdd/docker/platformio_install_deps.py#L2

felipecrs, 2023-01-02:
Well, either I wasn't able to pull in the latest changes or they didn't fix the issue for me. I will investigate it a little more later today.

kuba2k2, 2023-01-02:
This is for global libraries, not platforms. It would be quite useless to include all platforms (esp32, esp8266, pi pico) in every docker image.

felipecrs, 2023-01-02:
Does the `libretuya.framework.version` ESPHome config attribute makes any difference in this case?

kuba2k2, 2023-01-02:
`latest` is using whatever the installed version is, and `dev` is using the latest version (at the time of installation) from the repo. I don't know if it makes any difference.

felipecrs, 2023-01-02:
Got it. It makes sense then. I uninstalled the add-on and installed it again. This causes the persistent `/data` folder to be deleted and therefore, next time ESPHome executes, it needs to pull in all platforms again.

Using `latest`:

[image]

Using `dev`:

[image]

So, I guess the correct option is `dev` in this case. Now I'll check if the build works.

kuba2k2, 2023-01-02:
Right. `latest` doesn't touch the installed version, unless there's nothing installed, in which case it pulls from PIO registry.

felipecrs, 2023-01-02:
Done. No errors this time. Thanks a lot!

closed 2023-01-02 by felipecrs

kuba2k2, 2023-01-02:
Lovely! Thanks for the report.