libretiny-eu/libretiny #59: BK7231 MQTT

catalin2402, 2023-01-03:
MQTT won't compile for BK7231, Can't compile https://github.com/OttoWinter/async-mqtt-client, maybe we should fork it and change 
`#ifdef ESP32 `  in https://github.com/OttoWinter/async-mqtt-client/blob/master/src/AsyncMqttClient.hpp

kuba2k2, 2023-01-19:
MQTT should work now, starting with https://github.com/kuba2k2/libretuya-esphome/commit/65864298a3afcf951d6a5b6329d273b3f40c6457.

dskw, 2023-12-15:
I'm running into the same problem with a cb3s board.
I'm using the current ESPHome addon in HA and even tried directly listing the library from your linked commit, but it still complains about being unable to find AsyncTCP during compilation.

Do I have to switch to any dev versions of ESP home or bk72xx/libretiny for this to work? (your post is from January, so I hope not)

Configuration yaml:
[bkwindow1.yaml.txt](https://github.com/libretiny-eu/libretiny/files/13686512/bkwindow1.yaml.txt)

Failed compile log:
[logs_bkwindow1_run.txt](https://github.com/libretiny-eu/libretiny/files/13686509/logs_bkwindow1_run.txt)

Any advice is greatly appreciated.

olemmela, 2023-12-17:
@dskw it seems esphome/AsyncTCP-esphome dependency is missing.
Adding following lines to config should force it to be downloaded

```
esphome:
  libraries:
    - "esphome/AsyncTCP-esphome" 
```

dskw, 2023-12-18:
Ahh I tried that before but did not add the esphome prefix/suffix! That works indeed, thank you @olemmela !