libretiny-eu/libretiny #60: [esphome] wizard error

cyber-junkie9, 2023-01-03:
I am getting an error while using the wizard for LibreTuya, while ESP8266 / ESP32 works fine
although manually created config compiles fine

Also please add _**LibreTuya**_ in this line  **.. for your custom ESP8266/ESP32 firmware. Yay!**

```
venom@venom-vm:~/Desktop/libretuya-esphome$ python3 -m esphome wizard yourdevice.yml
Hi there!
I'm the wizard of ESPHome :)
And I'm here to help you get started with ESPHome.
In 4 steps I'm going to guide you through creating a basic configuration file for your custom ESP8266/ESP32 firmware. Yay!

============= STEP 1 =============
    _____ ____  _____  ______
   / ____/ __ \|  __ \|  ____|
  | |   | |  | | |__) | |__
  | |   | |  | |  _  /|  __|
  | |___| |__| | | \ \| |____
   \_____\____/|_|  \_\______|

===================================
First up, please choose a name for your node.
It should be a unique name that can be used to identify the device later.
For example, I like calling the node in my living room livingroom.
(name): livingroom
Great! Your node is now called "livingroom".

============= STEP 2 =============
      ______  _____ _____
     |  ____|/ ____|  __ \\
     | |__  | (___ | |__) |
     |  __|  \___ \|  ___/
     | |____ ____) | |
     |______|_____/|_|

===================================

Now I'd like to know what microcontroller you're using so that I can compile firmwares for it.
Are you using an ESP32, ESP8266 or LibreTuya platform? (Choose ESP8266 for Sonoff devices)

Please enter either ESP32, ESP8266 or LibreTuya.

(ESP32/ESP8266/LibreTuya): LibreTuya

Thanks! You've chosen LIBRETUYA as your platform.
Next, I need to know what board you're using.
Please go to https://docs.libretuya.ml/docs/status/supported/ and choose a board.

For example "cb2s".
Options:
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/venom/Desktop/libretuya-esphome/esphome/__main__.py", line 975, in <module>
    sys.exit(main())
  File "/home/venom/Desktop/libretuya-esphome/esphome/__main__.py", line 966, in main
    return run_esphome(sys.argv)
  File "/home/venom/Desktop/libretuya-esphome/esphome/__main__.py", line 931, in run_esphome
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/home/venom/Desktop/libretuya-esphome/esphome/__main__.py", line 341, in command_wizard
    return wizard.wizard(args.configuration)
  File "/home/venom/Desktop/libretuya-esphome/esphome/wizard.py", line 365, in wizard
    for group in boards.values():
AttributeError: 'list' object has no attribute 'values'
```

kuba2k2, 2023-01-03:
Fixed, thanks for the report. Do a git pull on esphome and try again.

cyber-junkie9, 2023-01-03:
> Fixed
Working now, Thank you very much.
Any workaround for MQTT (beken board) ?

cyber-junkie9, 2023-01-03:
>Fixed

Working now, Thank you very much.
Any workaround for MQTT (beken board) ?

kuba2k2, 2023-01-03:
No, it's a separate issue #59.

closed 2023-01-03 by kuba2k2