libretiny-eu/libretiny #61: On RTL8710BX OTA updates start failing after 40-50 successful ones with message: ERROR Error receiving acknowledge Update end: timed out

mihsu81, 2023-01-07:
After a number of 40-50 successful OTA updates, they start failing with error: ```ERROR Error receiving acknowledge Update end: timed out``` and continue failing until I flash the device over UART.

```
INFO Reading configuration /config/libretuya-esphome/plug-ezviz.yaml...
INFO Detected timezone 'Europe/Bucharest'
INFO Generating C++ source...
INFO Compiling app...
Processing plug-ezviz (board: generic-rtl8710bx-4mb-980k; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
HARDWARE: RTL8710BX 62MHz, 256KB RAM, 980KB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-realtek-amb1 @ 750.0.0-b0bc+sha.750b0bc 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.5d6edaa 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
|-- ArduinoJson @ 6.18.5
Compiling /data/plug-ezviz/.pioenvs/plug-ezviz/src/main.cpp.o
Linking /data/plug-ezviz/.pioenvs/plug-ezviz/firmware.elf
|-- Image 1: firmware.ota1.elf
|-- Image 2: firmware.ota2.elf
|   |-- image_0x00B000.ota1.bin
|   |   |-- ota1.ram_2.r.bin
|   |   |-- ota1.xip_image2.bin
|   |   |-- ota1.rdp.bin
|   |-- image_0x100000.ota2.bin
|   |   |-- ota2.ram_2.r.bin
|   |   |-- ota2.xip_image2.bin
|   |   |-- ota2.rdp.bin
Building UF2 OTA image
|-- firmware.uf2
RAM:   [==        ]  20.7% (used 54192 bytes from 262144 bytes)
Flash: [======    ]  58.1% (used 583148 bytes from 1003520 bytes)
========================= [SUCCESS] Took 8.77 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of plug-ezviz.local
INFO  -> 192.168.45.123
INFO Uploading /data/plug-ezviz/.pioenvs/plug-ezviz/firmware.uf2 (1167360 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...

ERROR Error receiving acknowledge Update end: timed out
```

kuba2k2, 2023-01-07:
In order for me to check what's wrong, do this:
- make a full dump of the device in a state of "broken" OTA updating
- -> install `ltchiptool` and run `ltchiptool flash read ambz dump_file.bin`
- post the dump here, along with an UF2 file causing that failure

mihsu81, 2023-01-07:
The archive contains a dump and the first uf2 which failed. The last working uf2 is not available because it was overwritten by the compilation process.
[libretuya.zip](https://github.com/kuba2k2/libretuya/files/10366355/libretuya.zip)

mihsu81, 2023-01-07:
I forgot to mention that I also have to flash a dump of the original ezviz firmware before OTA updates start working again.
I'm attaching again the 3 dumps I have. I'm usually flashing "dump_ezviz_3.bin".
[ezviz dumps.zip](https://github.com/kuba2k2/libretuya/files/10366405/ezviz.dumps.zip)

kuba2k2, 2023-01-07:
There is a 32-bit counter which indicates which firmware (1/2) to use. It's 0xFFFFFFFF by default, and each OTA attempt clears one bit from the number. Just like I thought, the mechanism of OTA switching is broken and doesn't know what to do when the counter reaches 0.

kuba2k2, 2023-01-19:
@mihsu81 
Should be fixed in 4b25ef75168d1d8beb25f0b59f1dbf3783833d00 :tada:
(note that you need to upload the fix first, after that OTA will work correctly)

mihsu81, 2023-01-19:
@kuba2k2 Flashed the original dump after which I flashed my latest uf2 for that device over UART.
I've also did a couple of OTA updates which went well.
Now only time will tell if the issue is fixed. I'll reopen the issue if something goes wrong.

Thank you again for your help.

closed 2023-01-19 by mihsu81