libretiny-eu/libretiny #63: Refactor project structure

kuba2k2, 2023-01-19:
While developing LT, one can notice at least a few oddities and nuisances about the project structure:
- `arduino/<family>/cores/arduino/...` - weird path
- WiFi impl is in `libraries/WiFi` (correct), but common headers are in `libretuya/api/WiFi` (weird), while other libraries are in `libretuya/libraries`
    - since WiFi is not really LT's API (does LT even have its own API?), naming the folder `api` is not good
- OTA utilities are in `LibreTuya` class. I think they could be moved to a separate OTA library (same applies to Watchdog)
- LT class is named `LibreTuyaClass.cpp` in common core, and `LibreTuyaAPI.cpp` in family cores
- libraries should be split between:
    1. family-independent/"final" (those not requiring **any** family code, like `Flash`, `Update`)
    2. family-specific (common interface but separate impl, optionally with common functions, like `WiFi`, `SoftwareSerial`)
    3. LT-independent (those ~~literally copied~~ from ESP32 core or other places, having code not related to LT at all - probably compatible with every Arduino core. Like `WiFiMulti`, `StreamString`, `HTTPClient`, `base64`...)
- `NetUtils` is a bit of a mess - it's mostly WiFi clients/servers, so why not in one of the `WiFi` folders?
- Arduino core extensions (classes which are not yet in ArduinoCore-API, like IPv6Address) should have their own directory

Generally, the current structure is forgetting a bit about non-ArduinoCore usage of LT. All code is in `arduino/`, and it's completely ignored when using bare-SDK builders, leaving the developer with nothing at all. The project could be refactored to separate the `port/` directory (C library ports, non-Arduino code, like `FlashDB` + FAL and `printf`), as well as the partial POSIX API, the LT logger and LT API, from the Arduino cores. Builders would then need to be adapted, to include that common code in bare-SDK builds. I think LT should be considered a C-only SDK (as in, adding only C-style functions) when used without Arduino. When using Arduino, the developer gets access to classes like OTA.

This refactor should be done together with the one described in TODO (`Environment stability`). Bumping the major version number is considerable.

- `platform/` - could this be merged with that refactored `cores` directory in any way?

kuba2k2, 2023-01-22:
Currently proposed project structure:
<details>
<summary>Click to expand</summary>
<pre>
cores/
├── common/
│   ├── arduino/
│   │   ├── inc/
│   │   │   ├── Arduino.h
│   │   │   ├── HardwareI2C.h
│   │   │   ├── LibreTuyaAPI.h
│   │   │   └── wiring_custom.h
│   │   ├── libraries/
│   │   │   ├── api/
│   │   │   │   ├── LT/
│   │   │   │   │   ├── LT.cpp
│   │   │   │   │   └── LT.h
│   │   │   │   ├── SoftwareSerial/
│   │   │   │   │   ├── SoftwareSerial.cpp
│   │   │   │   │   └── SoftwareSerial.h
│   │   │   │   ├── OTA/
│   │   │   │   │   ├── OTA.cpp
│   │   │   │   │   └── OTA.h
│   │   │   │   └── WiFi/
│   │   │   │       ├── WiFi.cpp
│   │   │   │       ├── WiFi.h
│   │   │   │       ├── WiFiAP.cpp
│   │   │   │       ├── ...
│   │   │   │       └── WiFiType.h
│   │   │   ├── common/
│   │   │   │   ├── Flash/
│   │   │   │   │   ├── Flash.cpp
│   │   │   │   │   └── Flash.h
│   │   │   │   ├── IPv6Address/
│   │   │   │   │   ├── IPv6Address.cpp
│   │   │   │   │   └── IPv6Address.h
│   │   │   │   ├── MD5/
│   │   │   │   │   ├── MD5.h
│   │   │   │   │   ├── MD5MbedTLSImpl.cpp
│   │   │   │   │   ├── MD5MbedTLSImpl.h
│   │   │   │   │   └── ...
│   │   │   │   ├── mDNS/
│   │   │   │   │   ├── LwIPmDNS.cpp
│   │   │   │   │   ├── mDNS.cpp
│   │   │   │   │   └── mDNS.h
│   │   │   │   ├── Update/
│   │   │   │   │   ├── Update.cpp
│   │   │   │   │   └── Update.h
│   │   │   │   ├── WiFiClient/
│   │   │   │   │   ├── LwIPClient.cpp
│   │   │   │   │   ├── ...
│   │   │   │   │   └── WiFiClient.h
│   │   │   │   └── WiFiServer/
│   │   │   │       ├── LwIPServer.cpp
│   │   │   │       ├── ...
│   │   │   │       └── WiFiServer.h
│   │   │   └── ext/
│   │   │       ├── base64/
│   │   │       ├── cbuf/
│   │   │       ├── HTTPClient/
│   │   │       ├── mDNS/
│   │   │       ├── StreamString/
│   │   │       ├── WebServer/
│   │   │       └── WiFiMulti/
│   │   └── src/
│   │       ├── common/
│   │       │   ├── abi.cpp
│   │       │   ├── ...
│   │       │   └── WMath.cpp
│   │       ├── posix/
[... 97 lines cut ...]
│   │   └── config/
│   │       └── sys_config.h
│   └── src/
│       └── fixups/
│           └── boot_handlers_105_bk7231s.S
├── beken-7251/
│   └── inc/
│       └── config/
│           └── sys_config.h
├── realtek-amb/
│   └── ...
├── realtek-ambz/
│   ├── ...
│   └── misc/
│       ├── boot_all_77F7.bin
│       └── ...
└── realtek-ambz2/
    ├── ...
    └── misc/
        ├── bootloader.axf
        └── ...
</pre>
</details>

Generated [here](https://tree.nathanfriend.io/?s=(%27options!(%27fancy!true~fullPath!false~trailingSlash!true~rootDot!false)~source!(%27source!%27-%20cores%5Cn9common%26YRHardwareI2CY%3FY%23_custom%60apiQLTJLTKJLTYQVJVKJVYQOTAJOTAKJOTAYQZJZKJZYJZAP%7FTypeYRcommonQFlashJFlashKJFlashYQ%2CJ%2CKJ%2CYQMD5JMD5Y%3CK%3CYJ...QmDNSJLwIPmDNSKJmDNSKJmDNSYQUpdateJUpdateKJUpdateYQZ%5BJLwIP%5B%7F%5BYQZ%5DJLwIP%5D%7F%5DYRextQbase64%2FQcbuf%2FQHTTP%5B%2FQmDNS%2FQStreamString%2FQWeb%5D%2FQZMulti%2FEsrcRcommonQabiKQ...QWMathKRposixQtime.c%3FKRmainK%23_custom%3AY%24lwipQerrnoYRerrnoYEproxyRlibretuyaY%20-%3E%20..%2Flt_apiYRcertsYRnetifYRsocketsYR...Elt_apiYElt_XYElt_loggerYElt_posix_apiYGsrcEposixRstrcasecmp.cRstrdup.cRstrptime.cElt_api.cElt_logger.cElt_main%2BcfgYRfdb_cfgYE%3BR%3B_XYRputchar.cRputs.cjxx%26FamilyY%23%60%40Q%40ClassYQ%40ClassKRVQVKRZQZKQ...%20%7Bno%20other%20headers%7DQZPrivateYRWireQWireKQWireYEsrcRmainK%23_analog.c%23_digital.c%23_irq.c%23%3AY%24param_XYRuart_pubYElt_defsYEsdk_privateYGmiscEbk7231%5E31n%5Exx.cfg%7Carch_main.cRate_app.cR...Rintc.cEwrapsRBkDriverFlash.cRwlan_ui%2Bflash_bk72xx_port.cE%3BR%3B_portYR%3B.cj31nG%25%7Ctemp_detect.cj31uG%25%7Cboot_handlers_105_bk7231s.Sj51G%25%22G...%22z%3D_all_77F7.binE...%22z2%3Dloader.axfE...%5Cn%27)~version!%271%27)*%20%208%5Cn***9*-%20E8-%20G%5Cn*9J8**9K.cppQ8*9R89VSoftware%40XconfigY.hZWiFij%5Cn9beken-72%22%5Cn9realtek-amb%23Rwiring%24EfixupsR%25incEXRsys_XY%26GarduinoEincRArduino%2B.cGportEflashdbRfal_%2CIPv6Address%3A.cGincEXRlwipopts%3Bprintf%3CJMD5MbedTLSImpl%3DG...GmiscEboot%3FRLibreTuyaAPI%40Serial%5BClient%5DServer%5E_bsp.ldEbk72%60YElibrariesR%7CGsrc%24%7FKJ...JZ%01%7F%7C%60%5E%5D%5B%40%3F%3D%3C%3B%3A%2C%2B%26%25%24%23%22jZYXVRQKJGE98*).

kuba2k2, 2023-01-31:
Moving all non-Arduino sources to a `platform`/`sdk`/`libretuya` subdirectory, to align it better with `arduino`.

kuba2k2, 2023-02-21:
`base` - that's the word I needed

lonerzzz, 2023-03-26:
How is the refactor coming along? I want to contribute for the Beken chips once you have a layout defined.

kuba2k2, 2023-03-26:
Let me crosspost the message I posted today on our Discord server:

---
Testers needed for the new `structure-refactor` branch :slightly_smiling_face: 
I think I've changed everything (or most) of what I wanted to change - what's left now (except for a few cosmetic changes) is testing. I've been testing for compilation errors, and if it boots (but roughly). That being said, there shouldn't be any difference in stability or compilation issues (compared to the current LibreTuya master branch, after GCC updates).

If you'd like to help testing the new version:
- get a device to which you have easy UART access - just in case something goes wrong in the OTA process. The OTA format is changed a bit (under the hood); it is backwards compatible (meaning you can update old devices), but **you can't revert to LT v0.12.6 or older**
- install LT v1.0.0 (alpha) - `platformio platform install -f https://github.com/kuba2k2/libretuya#structure-refactor`
- clone updated ESPHome - `git clone https://github.com/kuba2k2/libretuya-esphome -b platform/lt-refactor`
- compile your YAMLs, as usual - make sure to uninstall old esphome (or install the new one, using `pip install` - see the docs)

Let me know about any problems/questions in the thread here. Happy testing!

PR #94 (2023-04-27): Refactor project structure
Closes #63, #48, #65, #66, #67, #78, #87, #89, #91.
closed 2023-04-27 by kuba2k2