libretiny-eu/libretiny #65: Dual-OTA confusion

kuba2k2, 2023-01-19:
Dual OTA means either:
- having two OTA partitions, with different binaries (e.g. RTL8710B)
- having two OTA partitions, with identical binaries (e.g. RTL8720C)
- supporting OTA at all - having a `download` partition (e.g. BK72xx)

Currently, there's only one variable/function to check that - `otaSupportsDual()`/`has_dual_ota()`, and even I'm not sure which of these three values does that represent. Please reduce the confusion by naming it better, or adding more values.

kuba2k2, 2023-03-05:
- "is dual OTA" - having two OTA partitions
- "is download OTA" - having a separate upgrade partition

mczerski, 2023-04-16:
is there any description how to perform OTA update and how it is working (technically) ? I'm using libretuya to build BACnet device and I would like it to have OTA update capability but I don;t know where to start.

kuba2k2, 2023-04-16:
There are some pages about technical low-level aspects of UF2OTA in the docs. There's also a small usage example of uf2ota, but have in mind that these haven't been updated in a long time.

The best place to start would be to check the Update library in LT. If you're using the Arduino Core, the Update library is all you need - it works the same way as on other Arduinos supporting OTA. If you're not using Arduino Core, or want to do things manually, then check the implementation of Update.

mczerski, 2023-04-16:
thanks ! I'll start with Update library and see where it goes :)