libretiny-eu/libretiny #69: Remote transmitter/receiver component support

i2r, 2023-01-30:
Hi. Thank you for great work.

Trying to compile firmware for bk7231.

Adding component:
```
remote_transmitter:
  pin: D7
  carrier_duty_percent: 50%
```
Causes compiling error:
```
undefined reference to 'vtable for esphome::remote_transmitter::RemoteTransmitterComponent'
```

kuba2k2, 2023-01-31:
Not implemented in esphome yet. You could try changing #ifdef ESP32 to LIBRETUYA and there's a slight chance it will work.

RoganDawes, 2023-01-31:
I honestly didn't have to work too hard to get the `remote_receiver:` integration working when I did the STM32 port. If the GPIO's are working for input and output, then the remote* integrations should also work, albeit less efficiently than when using the ESP32 RMT peripheral. So you might want to aim at the ESP8266 implementation rather than the ESP32 version, as the ESP8266 was purely using the GPIOs.

ferbulous, 2023-01-31:
@catalin2402 is there specific instruction to do that? How do I modify the remote_transmitter & receiver .cpp files?

catalin2402, 2023-01-31:
where you see #define ESP8266, change it with `#if defined(USE_ESP8266) || defined(USE_LIBRETUYA)`

ferbulous, 2023-01-31:
thanks! @catalin2402 , modified the line in remote_receiver_esp8266.cpp
just that I'm getting this error when compiling
```
←[32mINFO Reading configuration smatrul_remote.yml...←[0m
←[32mINFO Generating C++ source...←[0m
Traceback (most recent call last):
  File "C:\Users\theba\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\theba\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 975, in <module>
    sys.exit(main())
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 966, in main
    return run_esphome(sys.argv)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 953, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 361, in command_compile
    exit_code = write_cpp(config)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 177, in write_cpp
    generate_cpp_contents(config)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 189, in generate_cpp_contents
    CORE.flush_tasks()
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\core\__init__.py", line 625, in flush_tasks
    self.event_loop.flush_tasks()
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\__main__.py", line 169, in wrapped
    await coro(conf)
  File "C:\Users\theba\Desktop\esp\libretuya-esphome\esphome\components\remote_receiver\__init__.py", line 63, in to_code
    cg.add(var.set_buffer_size(config[CONF_BUFFER_SIZE]))
KeyError: 'buffer_size'
sys:1: RuntimeWarning: coroutine 'add_arduino_global_workaround' was never awaited
```

i2r, 2023-01-31:
@ferbulous try to set `buffer_size: 1kB` in your device config, like so:
```
remote_receiver:
  ...
  buffer_size: 1kB
```

i2r, 2023-01-31:
@catalin2402 thank you! Compiled successfully. Now trying to find pin with transmitter connected (without opening device).

```(git)
diff --git a/esphome/components/remote_transmitter/remote_transmitter.h b/esphome/components/remote_transmitter/remote_transmitter.h
index 560d8380..91cfb2af 100644
--- a/esphome/components/remote_transmitter/remote_transmitter.h
+++ b/esphome/components/remote_transmitter/remote_transmitter.h
@@ -28,7 +28,8 @@ class RemoteTransmitterComponent : public remote_base::RemoteTransmitterBase,

  protected:
   void send_internal(uint32_t send_times, uint32_t send_wait) override;
-#ifdef USE_ESP8266
+/* #ifdef USE_ESP8266 */
+#if defined(USE_ESP8266) || defined(USE_LIBRETUYA)
   void calculate_on_off_time_(uint32_t carrier_frequency, uint32_t *on_time_period, uint32_t *off_time_period);

   void mark_(uint32_t on_time, uint32_t off_time, uint32_t usec);
diff --git a/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp b/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp
index 1c0eb94e..1aed7977 100644
--- a/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp
+++ b/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp
@@ -2,7 +2,7 @@
 #include "esphome/core/log.h"
 #include "esphome/core/application.h"

-#ifdef USE_ESP8266
+#if defined(USE_ESP8266) || defined(USE_LIBRETUYA)

 namespace esphome {
 namespace remote_transmitter {
```

samneirinck, 2023-01-31:
I've been running a `remote_receiver` and `remote_transmitter` for a while, based on the esp8266 code. (using a [Moes Tuya Remote Control IR Controller](https://templates.blakadder.com/moes_UFO-R6.html))

The receiver seems to work flawlessly: the IR commands get recognized in the correct format. Transmitting however I couldn't get to work yet reliably, I can see the receiver receive some signals when I transmit them, but the timing is off, and the actual devices don't recognize them.

If I retry a couple of times (e.g. via setting [repeat](https://esphome.io/components/remote_transmitter.html#remote-transmitter-actions) on the remote transmitter action), then it works in about 1 out of 10 attempts. 

I'm still planning to dive in a bit deeper, but would be interesting to know if you're facing similar issues once you get it running.

i2r, 2023-01-31:
@samneirinck i got it! Found pinout for my device at https://www.elektroda.com/rtvforum/topic3940249.html

And i have exactly the same problem now. Transmitter works in about 1 out of 6—7 attempts.

RoganDawes, 2023-01-31:
Do you perhaps have a logic analyser or oscilloscope? I know you were trying to avoid opening the device, but if you could tap that pin to get some debugging information (timing of the waveform, really), that would be very helpful to figure out what is actually going wrong.

i2r, 2023-01-31:
@RoganDawes no, i do not have it =(

i2r, 2023-01-31:
@RoganDawes is it possible to analyse transmitted signal by recording raw data by the device's receiver?

ferbulous, 2023-02-01:
> If I retry a couple of times (e.g. via setting [repeat](https://esphome.io/components/remote_transmitter.html#remote-transmitter-actions) on the remote transmitter action), then it works in about 1 out of 10 attempts.

I noticed the same bug on [openbk](https://github.com/openshwprojects/OpenBK7231T_App/issues/643) (for the console command)

i2r, 2023-02-01:
> Any other changes I need to make to the yaml or LT files?

If not, you need to change *.h file too:
https://github.com/esphome/esphome/blob/e847766514957f6ceb81de0b678399918ed532a2/esphome/components/remote_receiver/remote_receiver.h#L9
https://github.com/esphome/esphome/blob/e847766514957f6ceb81de0b678399918ed532a2/esphome/components/remote_receiver/remote_receiver.h#L58

i2r, 2023-02-01:
If you give me code fragment, I can use it to log generated signal. So we can understand where is a problem: (1) at signal generation or (2) transmittion.

ferbulous, 2023-02-01:
> If not, you need to change *.h file too:
> https://github.com/esphome/esphome/blob/e847766514957f6ceb81de0b678399918ed532a2/esphome/components/remote_receiver/remote_receiver.h#L9 [...]

Thanks! that fixed it, on both the receiver.h & transmitter.h
Will test it out later if I have the same issue with the repeated attempts

ferbulous, 2023-02-01:
@i2r hi, not much success yet with transmitting (i keep pressing the button repeatedly)
tested with sony and pronto codes. I'll be testing the same codes on esp just to verify if the code does work
Do you mind sharing how you added the raw codes in the yaml

```
//sony vol down raw
22:56:08	[D]	[remote.raw:041]	
Received Raw: 2219, -667, 1125, -657, 1145, -636, 531, -688, 531, -688, 1114, -668, 531, -656, 532, -695, 544, -668, 500, -656, 531, -688, 536, -2695, 1082
22:56:08	[D]	[remote.raw:041]	
Received Raw: 4148, -681, 1094, -708, 1105, -687, 500, -688, 537, -695, 1081, -688, 531, -657, 551, -668, 500, -688, 531, -688, 500, -708, 479, -688, 1125
22:56:08	[D]	[remote.raw:041]	
Received Raw: 2212, -688, 1145, -668, 1094, -687, 532, -708, 480, -687, 1125, -688, 472, -2729, 487
22:56:08	[D]	[remote.raw:041]	
Received Raw: 498, -688, 500, -708, 511, -688, 499, -688, 1125
````

dbuezas, 2023-02-01:
You may find this tool useful for visualizing, comparing and debugging raw patterns 
https://github.com/dbuezas/esphome-remote_receiver-oscilloscope

i2r, 2023-02-01:
@ferbulous

Example:
```
button:
  - platform: template
    name: "PH450 Z Send Raw"
    on_press:
      remote_transmitter.transmit_raw:
        code: [ 8940, -4463, 628, -493, 625, -500, 625, -1663, 556, -531, 625, -469, 625, -500, 628, -497, 625, -500, 625, -1663, 556, -1687, 594, -1660, 590, -1656, 594, -510, 615, -469, 625, -500, 593, -532, 635, -490, 625, -1656, 594, -500, 631, -494, 625, -500, 625, -500, 625, -503, 622, -500, 625, -1625, 601, -493, 625, -1656, 594, -1663, 588, -1655, 601, -1649, 594, -1633, 585, -1657, 600 ]
```

But for tests i use `transmit_lg`
```
button:
  - platform: template
    name: "PH450 Volume Up"
    on_press:
      remote_transmitter.transmit_lg:
        data: 0x20F040BF
        nbits: 32
```

dbuezas, 2023-02-01:
When transmitting, it is generally better to repeat the signal a bunch of times to increase the chances of the receiver getting it. Most rc remotes do exactly that, and the receivers debounce the commands so they only pick them up once.
```yaml
    - remote_transmitter.transmit_raw:
          code: [...]
          repeat:
              times: 10
              wait_time: 4.733ms 
```

You can find an appropriate wait_time by increasing the `idle` option in the remote_receiver to a very high number. This will make it pick up multiple repetitions as a signal stream, separated by spaces as long as the remote's idle spacing. The UI tool above is particularly useful for that.

I think the first two numbers in the code `8940, -4463` are just spacing between repetitions actually. It doesn't make much of a difference when transmitting, but it is worth removing for receiving, or if you will do repeated transmissions as I suggest.

i2r, 2023-02-01:
Anyway it sends random numbers:
```
[22:22:32][D][button:013]: 'PH450 Volume Up' Pressed.
[22:22:32][D][remote_transmitter:075]: Sending remote code...
[22:22:33][D][remote.raw:041]: Received Raw: 144, -523, 4160, -2776, 344, -3937, 532, -624, 563, -562, 625, -1594, 593, -563, 625, -531, 562, -563, 624, -531, 625, -562, 626, -1531, 593, -1625, 657, -1530, 625, -1563, 656, -500, 2313, -468, 438, -188
[22:22:33][D][remote.raw:041]: Received Raw: -2501, 688, -1532, 656, -500, 655, -469, 688, -468, 656, -499, 625, -531, 656, -500, 657, -1531, 656, -499, 657, -1532, 875, -3532, 625, -1562, 625, -1563, 687, -1532, 2563

[22:22:36][D][button:013]: 'PH450 Volume Up' Pressed.
[22:22:36][D][remote_transmitter:075]: Sending remote code...
[22:22:36][D][remote.raw:028]: Received Raw: 171, -368, 695, -649, 3467, -1744, 286, -4436, 625, -968, 219, -532, 625, -4272, 352, -376, 655, -500, 2344, -5375, 656, -1562, 1194, -4376, 934, -1435, 655, -532, 593, -532, 656, -469, 657, -500, 624, -532, 656, -531, 625, -1781, 375,
[22:22:36][D][remote.raw:041]:   -531, 625, -1563, 562, -1656, 626, -1563, 624, -1594, 594, -1603, 584, -1625, 628

[22:22:39][D][button:013]: 'PH450 Volume Up' Pressed.
[22:22:39][D][remote_transmitter:075]: Sending remote code...
[22:22:40][D][remote.raw:041]: Received Raw: 2654, -1718, 751, -313, 813, -3909, 625, -563, 625, -531, 562, -1657, 656, -474, 214, -2688
[22:22:40][D][remote.raw:041]: Received Raw: -1407, 249, -532, 625, -500, 625, -1594, 625, -3916, 525, -1532, 625, -532, 625, -500, 688, -657, 437, -532, 563, -594, 2313, -157, 2187, -2657
[22:22:40][D][remote.raw:041]: Received Raw: -5128, 156, -564, 594, -562, 626, -1563, 2344, -1001, 625, -1595, 594, -1595, 657, -1532, 657, -1562, 687, -1846, 342

[22:22:44][D][button:013]: 'PH450 Volume Up' Pressed.
[22:22:44][D][remote_transmitter:075]: Sending remote code...
[22:22:44][D][remote.raw:028]: Received Raw: 2342, -4112, 667, -500, 2344, -1407, 281, -468, 1094, -2344, 594, -562, 594, -1594, 687, -1531, 2874, -1532, 625, -531, 688, -844, 219, -531, 688, -469, 2844, -501, 625, -531, 593, -562, 562, -563, 656, -532, 593, -531, 625, -532, 624,
[22:22:44][D][remote.raw:041]:   -1564, 625, -501, 625, -1594, 625, -1562, 656, -1563, 625, -1563, 656, -1531, 562, -1656, 707
```

i2r, 2023-02-01:
Config:
```yaml
remote_transmitter:
  pin:
    number: P7
    mode:
      output: true
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: P8
    inverted: true
  buffer_size: 1kB
  idle: 200ms
  dump: raw

button:
  - platform: template
    name: "PH450 Z Send Raw"
    on_press:
      remote_transmitter.transmit_raw:
        code: [ 8000, -4000, 600, -550, 600, -550, 600, -1600, 600, -550, 600, -550, 600, -550, 600, -550, 600, -550, 600, -1600, 600, -1600, 600, -1600, 600, -1600, 600, -550, 600, -550, 600, -550, 600, -550, 600, -550, 600, -1600, 600, -550, 600, -550, 600, -550, 600, -550, 600, -550, 600, -550, 600, -1600, 600, -550, 600, -1600, 600, -1600, 600, -1600, 600, -1600, 600, -1600, 600, -1600, 600 ]
        carrier_frequency: 38kHz
        repeat:
          times: 10
          wait_time: 4.733ms
```

Results:
```yaml
[22:37:47][D][button:013]: 'PH450 Z Send Raw' Pressed.
[22:37:47][D][remote_transmitter:075]: Sending remote code...
[22:37:47][D][remote.raw:041]: Received Raw: 202, -399, 2567, -1769, 384, -937, 2625, -2158, 688, -469, 656, -500, 657, -1532, 2375, -2844
[22:37:47][D][remote.raw:041]: Received Raw: -3906, 624, -1563, 655, -1531, 625, -1593, 656, -1530, 656, -499, 3156, -282, 2343
[22:37:47][D][remote.raw:041]: Received Raw: 344, -1562, 687, -470, 2344, -437, 406, -250
[22:37:47][D][remote.raw:028]: Received Raw: -2468, 688, -468, 686, -1531, 657, -468, 657, -1562, 656, -1531, 625, -1561, 688, -1531, 656, -1530, 2875, -5108, 2043, -9494, 563, -625, 593, -563, 593, -1595, 594, -593, 573, -551, 562, -593, 531, -657, 531, -656, 500, -1781, 406, -1687,
[22:37:47][D][remote.raw:041]:   500, -1749, 469, -1780, 2281, -63522, 156, -11152, 156, -13274, 219, -2030, 187, -1999, 187, -9652, 231, -13942, 249, -6622, 125, -1000, 187, -1906, 187, -4248, 249, -1938, 250, -4435, 187, -3060, 281, -2064, 250, -2092, 219

[22:37:58][D][button:013]: 'PH450 Z Send Raw' Pressed.
[22:37:58][D][remote_transmitter:075]: Sending remote code...
[22:37:59][D][remote.raw:028]: Received Raw: 205, -463, 1064, -1803, 4163, -6222, 335, -1917, 583, -531, 657, -532, 594, -531, 656, -500, 657, -501, 656, -1530, 625, -1562, 687, -1499, 657, -1531, 655, -500, 624, -562, 625, -545, 579, -532, 624, -563, 593, -1593, 562, -594, 593, -531,
[22:37:59][D][remote.raw:028]:   625, -530, 626, -532, 593, -594, 532, -593, 601, -1617, 563, -625, 532, -1624, 563, -1655, 563, -1686, 500, -1687, 500, -1718, 468, -1781, 156, -5248, 2250, -83302, 250, -2123, 156, -2061, 250, -1938, 250, -6341, 281, -999, 125, -967, 188, -906, 250,
[22:37:59][D][remote.raw:028]:   -905, 250, -3154, 188, -906, 250, -875, 281, -937, 219, -905, 250, -843, 156, -1062, 219, -2000, 219, -937, 188, -1936, 281, -1905, 281, -1937, 281, -4156, 250, -1906, 249, -5154, 250, -18383, 198, -906, 250, -905, 219, -968, 281, -1875, 2250, -5687,
[22:37:59][D][remote.raw:041]:   218, -783, 375, -687, 468, -688, 468, -719, 438, -1687, 249, -968, 438, -749, 344, -750, 468, -656, 499, -656, 468, -719, 249, -1968, 2250
[22:37:59][D][remote.raw:041]: Received Raw: 204, -1031, 250, -813, 428, -2887, 2246, -1217, 468, -688, 188, -2031, 437, -1812, 2281, -2061, 219, -7685, 2283
[22:37:59][D][remote.raw:041]: Received Raw: 187, -2156, 156, -687, 500, -874, 281, -624, 500, -1686, 532, -656, 468, -1688, 530, -1688, 499, -1687, 500, -1717, 500, -1687, 249, -1969, 499

[22:38:07][D][button:013]: 'PH450 Z Send Raw' Pressed.
[22:38:07][D][remote_transmitter:075]: Sending remote code...
[22:38:08][D][remote.raw:028]: Received Raw: 209, -399, 1562, -1872, 310, -566, 343, -594, 1282, -3907, 621, -562, 657, -469, 593, -1626, 594, -562, 625, -500, 593, -562, 594, -562, 626, -532, 656, -1530, 624, -1594, 656, -1531, 656, -1531, 656, -500, 594, -562, 562, -594, 656, -500,
[22:38:08][D][remote.raw:041]:   593, -531, 688, -1532, 656, -499, 625, -500, 688, -469, 656, -501, 687, -469, 656, -500, 2312, -3219
[22:38:08][D][remote.raw:041]: Received Raw: -5593, 688, -1501, 687, -1534, 656, -1564, 594, -1594, 687, -4627, 8062, -3938, 626, -531, 595, -563, 2844, -501
[22:38:08][D][remote.raw:028]: Received Raw: -2564, 593, -531, 625, -562, 2314, -3252, 562, -1626, 562, -1594, 2344, -1159, 2282, -11400, 1636, -1658, 2281, -6597, 2250, -17222, 531, -878, 281, -594, 219, -3127, 530, -656, 500, -657, 500, -625, 532, -593, 563, -1657, 532, -1657, 532,
[22:38:08][D][remote.raw:041]:   -1688, 531, -1658, 531, -656, 469, -625, 563, -656, 469, -657, 500, -656, 2281, -1033
[22:38:08][D][remote.raw:041]: Received Raw: -9628, 563, -657, 500, -1656, 564, -1625, 563, -3985, 239, -1872, 2218, -5441, 218

[22:38:19][D][button:013]: 'PH450 Z Send Raw' Pressed.
[22:38:19][D][remote_transmitter:075]: Sending remote code...
[22:38:20][D][remote.raw:028]: Received Raw: 174, -431, 813, -436, 1064, -1368, 3378, -3936, 657, -500, 688, -468, 656, -1563, 656, -469, 687, -469, 656, -501, 625, -532, 656, -500, 625, -1564, 657, -1532, 2376, -2034, 657, -500, 656, -469, 688, -469, 689, -468, 656, -501, 656, -1501,
[22:38:20][D][remote.raw:028]:   687, -500, 625, -531, 470, -657, 625, -531, 688, -469, 626, -531, 656, -1532, 687, -469, 687, -1501, 687, -1532, 626, -1564, 2437, -157, 187, -1634, 650, -1531, 594, -4721, 8032, -3970, 656, -532, 563, -563, 626, -1595, 562, -563, 625, -531, 625, -532,
[22:38:20][D][remote.raw:041]:   594, -563, 594, -563, 593, -1595, 595, -1626, 2312, -2160, 250

[22:38:24][D][button:013]: 'PH450 Z Send Raw' Pressed.
[22:38:24][D][remote_transmitter:075]: Sending remote code...
[... cut ...]
[22:39:14][D][remote.raw:028]:   -531, 593, -562, 607, -549, 2314, -2190, 2309, -1312, 376, -656, 531, -593, 563, -607, 549, -1657, 531, -2863, 480, -1718, 250, -2250, 124, -195443, 187, -5379, 156, -4255, 124, -10037, 187, -11377, 156, -1002, 125, -3126, 250, -7691, 250, -14289, 281,
[22:39:14][D][remote.raw:041]:   -969, 188, -908, 249, -2125, 156, -2095, 219, -1906, 282, -3066, 250, -4188, 250, -1908, 250, -1939, 250, -99645, 146, -20130, 208, -10843, 250, -31650, 250, -4187, 249, -2062, 188, -2001, 187, -3157, 218, -2093, 187, -937, 219, -3032, 250
```

samneirinck, 2023-02-01:
I have a feeling it's mostly the beginning of the sequence that is causing problems. For testing purposes, I'm using the following:
```yaml
      remote_transmitter.transmit_raw:
        code: [1000, -200, 500, -500, 750, -750, 1000]
        carrier_frequency: 38kHz

```
This was the output:
```
[20:27:25][D][remote.raw:041]: Received Raw: 194, -222, 278, -282, 406, -469, 782, -719, 1041
[20:27:50][D][remote.raw:041]: Received Raw: 174, -219, 250, -281, 469, -500, 781, -718, 1042
[20:28:01][D][remote.raw:041]: Received Raw: 205, -219, 219, -218, 469, -500, 781, -750, 981
[20:28:09][D][remote.raw:041]: Received Raw: 667, -219, 500, -469, 781, -718, 1041
[20:28:27][D][remote.raw:041]: Received Raw: 170, -222, 279, -218, 469, -469, 750, -753, 1070
[20:28:37][D][remote.raw:041]: Received Raw: 170, -218, 969, -500, 781, -719, 1009
[20:28:50][D][remote.raw:041]: Received Raw: 164, -250, 969, -469, 813, -688, 1041
```

Using @dbuezas's excellent tool, this looks like this:
(1st row is the reference)
[image]

With a different made-up sequence:
```1000, -200, 500, -500, 750, -750, 1000, -200, 500, -500, 750, -750, 1000, -200, 500, -500, 750, -750, 1000, -200, 500, -500, 750, -750```

(1st row is the reference)
[image]

I'll see to add some additional logging to see if this reveals more.

kuba2k2, 2023-02-01:
I'm confused, is that remote transmitter/receiver for IR remotes, or 433 MHz remotes?

ferbulous, 2023-02-02:
@i2r 
i'm using transmitter_sony
Do I just add to each line for this command with multiple data entry?

```
//vol up
20:25:09	[D]	[remote.sony:065]	
Received Sony: data=0x00000481, nbits=12
20:25:09	[D]	[remote.sony:065]	
Received Sony: data=0x00000681, nbits=12
20:25:10	[D]	[remote.sony:065]	
Received Sony: data=0x00000581, nbits=12
```

```
button:
  - platform: template
    name: "Sony Volume Up"
    on_press:
      remote_transmitter.transmit_sony:
        data: 0x00000481
        nbits: 12
      remote_transmitter.transmit_sony:
        data: 0x00000681
        nbits: 12
      remote_transmitter.transmit_sony:
        data: 0x00000581
        nbits: 12
```

For my raw codes, i should just output all the codes to a single line?
But I noticed the end of each & start of next line of codes are both positive values instead of positive & negative
```
//sony vol down raw
22:56:08	[D]	[remote.raw:041]	
Received Raw: 2219, -667, 1125, -657, 1145, -636, 531, -688, 531, -688, 1114, -668, 531, -656, 532, -695, 544, -668, 500, -656, 531, -688, 536, -2695, 1082
22:56:08	[D]	[remote.raw:041]	
Received Raw: 4148, -681, 1094, -708, 1105, -687, 500, -688, 537, -695, 1081, -688, 531, -657, 551, -668, 500, -688, 531, -688, 500, -708, 479, -688, 1125
22:56:08	[D]	[remote.raw:041]	
Received Raw: 2212, -688, 1145, -668, 1094, -687, 532, -708, 480, -687, 1125, -688, 472, -2729, 487
22:56:08	[D]	[remote.raw:041]	
Received Raw: 498, -688, 500, -708, 511, -688, 499, -688, 1125
```

ferbulous, 2023-02-02:
> I'm confused, is that remote transmitter/receiver for IR remotes, or 433 MHz remotes?

the model we're using is for IR only

catalin2402, 2023-02-02:
But the code is for both IR & RF 433

ferbulous, 2023-02-02:
> But the code is for both IR & RF 433

i see, well this model doesn't have RF

[image]

dbuezas, 2023-02-02:
> For my raw codes, i should just output all the codes to a single line?

No, it's normal that it logs it in multiple lines

i2r, 2023-02-02:
I made separate stands.

1) Libretuya Transmitter
```yaml
      remote_transmitter.transmit_raw:
        code: [ 1000, -500, 500, -500, 250, -500, 500, -500, 750, -500, 2000 ]
        carrier_frequency: 38kHz
        repeat:
          times: 10
          wait_time: 5ms
```

2) ESP32 Receiver
```
remote_receiver:
  pin:
    number: GPIO19
    inverted: true
  idle: 500ms
  dump: raw
```

Result
[image]

Start signal is broken every time.

i2r, 2023-02-02:
I planned to use libretuya remote with [IR Remote Climate Component](https://esphome.io/components/climate/climate_ir.html) — there is no option to send signal multiple times.

Only L1 is working =(
[image]

dbuezas, 2023-02-02:
It looks like you are almost there. How does the circuit look like? Maybe there's a pull up or pull down resistors, or something like like that could be tweaked a bit.
There is also a filter option in esphome that can discard pulses too short to be real. You could take something like 10 to 50% of the shortest pulse you want to receive and it should be more resilient. Or maybe some smoothing capacitor on the receiver

i2r, 2023-02-02:
Source of device photos: https://www.elektroda.com/rtvforum/topic3940249.html

I can not modify receiver cause it is AC that I want to control with the bk7231n based remote with esphome firmware. 

[image]
[image]
[image]
[image]

dbuezas, 2023-02-03:
Oh, right, sorry I forgot it was a pre made one. And what is not working well is transmitting to the AC or receiving from the remote?

i2r, 2023-02-03:
> And what is not working well is transmitting to the AC or receiving from the remote?

With esp32s2 signal is correct every time, I tested.

So **transmitting** at **bk7231 is broken**. Looks like there are interrupts at transmission time.

dbuezas, 2023-02-03:
I see. Is reception also affected?
In esp32 and 8266, time is tracked using a cpu ticks counter which I believe is not present in libretuya chips, so a hardware timer is used for that instead. It is possible that the transmitter is disabling interrupts during time critical operations and that that results in the time keeping logic missing timer overflow events

i2r, 2023-02-03:
Seems reception is not effected.

kuba2k2, 2023-02-03:
There's a feed_wdt() on line 88 of remote_transmitter_esp8266.cpp. It's possible that this function adds some delay to the process. Can you try with this line removed?

i2r, 2023-02-03:
> There's a feed_wdt() on line 88 of remote_transmitter_esp8266.cpp. It's possible that this function adds some delay to the process. Can you try with this line removed?

I already commented it out. All my tests are without `App.feed_wdt();` call.

dbuezas, 2023-02-03:
The only thing I see that can go wrong is an interrupt taking very long and disrupting the carrier frequency. The issue is that if you were to disable interrupts while transmitting, the interrupts that keep track of time would also break down.

dbuezas, 2023-02-03:
Now that I think about it, it may work. The carrier frequency is 38kHz, so micros() is called way more often than once every 2 ms (500hz). Therefore, micros should be able to stay on track. It will de-synchronize with millis, but the remote transmitter doesn't use millis anyway, so who cares.

dbuezas, 2023-02-03:
Putting an interrupt lock [here](https://github.com/kuba2k2/libretuya-esphome/blob/platform/libretuya/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp#L79) like this:
```cpp
[...]
  for (uint32_t i = 0; i < send_times; i++) {
    InterruptLock lock;
    for (int32_t item : this->temp_.get_data()) {
[...]
```

Other things may get very unhappy about that, but I think it is worth the try :)

dbuezas, 2023-02-03:
Oh, I think you'll also need to replace the delayMicroseconds with a busy loop:
replace [this](https://github.com/kuba2k2/libretuya-esphome/blob/platform/libretuya/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp#L40)
with:
```cpp
  } else while (this->target_time_ > micros()){ /*busy loop that ensures micros is constantly called*/};
```

samneirinck, 2023-02-03:
👏  Your assumptions are totally correct. With the above changes it works reliably!

Now the next question is, what interrupt handling is interfering here, because I guess putting in an interrupt lock for a couple of milliseconds is not the ideal way to fix this. Is there a way to log what interrupts are received?

dbuezas, 2023-02-03:
Oh wow I got lucky, that was a long shot 😁. I think I modified the micros() function for that uC specifically so it keeps working during long interrupts.

Unfortunately that's where my knowledge ends, I don't know what consequences that would have or which other interrupts may be interfering. Maybe with an oscilloscope one could check how much and how often the carrier is distorted. AFAIK, the wifi stack gets processing time when one calls delay or yield, but it may be doing something with interrupts too. I think only @kuba2k2 may know this. I also have no idea on how much overhead an interrupt in these chips has.

What you could do to reduce risks is to only disable interrupts while the pulse is high (or low if inverted). When the led's off, the carrier frequency doesn't matter, it's just off.

My guess is that it will work fine. It may lose some packets while transmitting, but I think the web sockets use TCP, so those will just be resent. And it's not like you'll be transmitting once a second all the time so ...

dbuezas, 2023-02-03:
> a couple of milliseconds

a couple of ms would be quite ok, but the signal you are sending is over 200ms... Anyway, as I said before, it should be able to recover from that just fine actually.

dbuezas, 2023-02-03:
Can you share how the new screenshots look like? I'd like to see how much more accurate it becomes

samneirinck, 2023-02-03:
> Can you share how the new screenshots look like? I'd like to see how much more accurate it becomes

Unfortunately I only have a single device with both a receiver and a transmitter, maybe @i2r can test out the above and report back.

dbuezas, 2023-02-03:
Oh, I think it would be wise to try re-adding the feed_wdt calls. Failling to feed the watchdog for too long will reboot the cpu, that's way worse than losing some packets

i2r, 2023-02-04:
> But I will check with transmitting by esp32s2 and receipting with bk7231.

**Receiver is broken** too. Sending signals with ESP32, receiving with bk7231.

Test signal:
[image]

Real signal:
[image]

i2r, 2023-02-04:
@dbuezas the signal is correct with your tweaks!

Receiver is disabled.

Test signal with 10 repeats
[image]

Real signal (projector volume up, short)
[image]

Real signal (AC, long)
[image]

dbuezas, 2023-02-04:
Oh, interesting, the receiver is affected too? 
I'm not sure I got you right: Does the transmitter also works when the wdt is fed (as long as interrupts are disabled)?

i2r, 2023-02-04:
> Does the transmitter also works when the wdt is fed (as long as interrupts are disabled)?

Yes. Right. Transmitter works when the wdt is fed as long as interrupts are disabled.

samneirinck, 2023-02-06:
Created a PR @ https://github.com/kuba2k2/libretuya-esphome/pull/6, so at least it doesn't just live on one of our machines.

kuba2k2, 2023-03-13:
This got closed along with the PR that I merged... thanks to @samneirinck the remote transmitter/receiver should work now :smile: 
If someone confirms that's the case, I'll close this issue.

catalin2402, 2023-03-13:
Confirm, tested both receiver/transmitter

closed 2023-08-29 by kuba2k2

olivluca, 2024-08-12:
The receiver is not working here. I reported [this bug](https://github.com/esphome/issues/issues/6138) in the esphome repository.

olivluca, 2024-08-12:
Fix [here](https://github.com/libretiny-eu/libretiny/pull/290)