libretiny-eu/libretiny #70: bdk_freertos repository not found

LeonReinPMD, 2023-02-01:
Error Message:

```Tool Manager: Installing git+https://github.com/bekencorp/bdk_freertos#c3533e8
INFO Installing git+https://github.com/bekencorp/bdk_freertos#c3533e8
git version 2.25.1
Cloning into '/home/dereinle/.platformio/.cache/tmp/pkg-installing-d9u0q2xv'...
Username for 'https://github.com':
Password for 'https://github.com':
remote: Repository not found.
fatal: Authentication failed for 'https://github.com/bekencorp/bdk_freertos/'
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/bekencorp/bdk_freertos', '/home/dereinle/.platformio/.cache/tmp/pkg-installing-d9u0q2xv']
```

Seems like Beken has removes its FreeRtos repository:
[https://github.com/bekencorp/bdk_freertos](https://github.com/bekencorp/bdk_freertos)

lonerzzz, 2023-02-01:
That is indirectly my fault. I just asked them if they were supporting it since no updates had been posted after the initial submission. Hopefully you have a clone of it  in your workspace...

kuba2k2, 2023-02-01:
Did you at least get any response? :smile:

lonerzzz, 2023-02-01:
No, no response at all. Did you have a clone you could repost?

evlo, 2023-02-02:
And if we find a copy and upload here, how can i then build esphome libretuya or does this need also fixing in the code so it does not refer to beken repo? (I'm trying to build for a first time and come across this)

kuba2k2, 2023-02-02:
I have a backup in the LibreTuya organization on GitHub (see pinned repos). There's a platform.json file in LT repo, which contains the repo URL which needs to be changed.

LeonRein, 2023-02-02:
I have manually replaced [https://github.com/bekencorp/bdk_freertos repository](https://github.com/bekencorp/bdk_freertos repository) with [https://github.com/SberHome/bdk_freertos_old](https://github.com/SberHome/bdk_freertos_old). 
Compiled successfully.

divisini, 2023-02-02:
The pinned repo didn't work out. Using commit #c3533e8 from the repo you forked was successful.

evlo, 2023-02-02:
on windows local cache of the repo file is located at `C:\Users\<usr>\.platformio\platforms\libretuya`

kuba2k2, 2023-02-02:
Maybe it didn't work out, because you were using the default branch. It has to be pinned to a specific tag (there's only one I think). I'll fix this later, possibly today.

kuba2k2, 2023-02-02:
Should be fixed in 949dfe7266adef02ef55f0e4038dbed2a7c7c744.

closed 2023-04-28 by kuba2k2