libretiny-eu/libretiny #71: example config does have incorrect indents

evlo, 2023-02-02:
Example config on this https://docs.libretuya.ml/docs/projects/esphome/ page does not have correct formatting and is not buildable
```
esphome:
name: yourdevice

libretuya:
board: wr3  # THIS IS YOUR BOARD CODE
framework:
    version: latest

logger:
api:
password: ""
ota:
password: ""

wifi:
ssid: "YourWiFiSSID"
password: "SecretPa$$w0rd"
ap:
    ssid: "Yourdevice Fallback Hotspot"
    password: "Dv2hZMGZRUvy"
```

should be

```
esphome:
  name: yourdevice

libretuya:
  board: wr3  # THIS IS YOUR BOARD CODE
  framework:
    version: latest

logger:
api:
  password: ""
ota:
  password: ""

wifi:
  ssid: "YourWiFiSSID"
  password: "SecretPa$$w0rd"
  ap:
    ssid: "Yourdevice Fallback Hotspot"
    password: "Dv2hZMGZRUvy"

web_server:
captive_portal:
```

kuba2k2, 2023-02-02:
Oh, thanks, I didn't notice that before.

kuba2k2, 2023-02-02:
Fixed in f0d490aef952899b4fa2e9619c053a6f8eb66ac6.

closed 2023-02-03 by kuba2k2