libretiny-eu/libretiny #73: What other modules do i need to run libretuya-esphome ?

donburch888, 2023-02-03:
Having installed tuya-cloudcutter on a new Raspberry Pi OS LITE SDcard, I am trying to install and run LibreTuya-esphome. on the same card.  
My objective is just to install libretuya-esphome on my Arlec PC191HA power plugs, not to get into full-blown development. 

Sorry for this being so long-winded, but maybe I missed a step somewhere, or (more likely) didn't have the same development environment already set up and aren't experienced enough to make the right assumptions. 

Starting at https://github.com/kuba2k2/libretuya, the first step is "Install PlatformIO", which sounds simple. 
Your link is to the PlatformIO IDE, which appears to require VScode.  I don't want to fill my RasPi with apps that I won't use ... so do I need to install VScode and the IDE in order to run libretuya, or will the CLI version install and work as well ? 

OK, so the CLI method has its problems, starting with having to manually "sudo apt-get install python3-venv" because the script doesn't have permission.  Then the "Install shell commands" section tries to create symlinks in a ~/.local/bin directory which doesn't exist yet.  I am unsure whether installing 99-platform-udev rules is necessary, so have skipped it for now.  PlatformIO isn't your package, so ignore this gripe. 

I believe I have PlatformIO installed now, and  followed your instructions for installing libretuya - which appeared to complete smoothly.  However **running "python -m esphome" is missing modules**.  I have manually installed python3-pip and voluptuous, and now need yaml
```
pi@raspi4:~/esphome/libretuya-esphome $ python -m esphome
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pi/esphome/libretuya-esphome/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/home/pi/esphome/libretuya-esphome/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/home/pi/esphome/libretuya-esphome/esphome/config.py", line 13, in <module>
    from esphome import core, yaml_util, loader
  File "/home/pi/esphome/libretuya-esphome/esphome/yaml_util.py", line 9, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'
pi@raspi4:~/esphome/libretuya-esphome $ 

pi@raspi4:~/esphome/libretuya-esphome $ pip install yaml
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement yaml
ERROR: No matching distribution found for yaml
pi@raspi4:~/esphome/libretuya-esphome $ sudo apt-get install yaml
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package yaml
pi@raspi4:~/esphome/libretuya-esphome $
```
and pypi.org shows many 472 pages of versions :-(  Which yaml file do I need, and what other dependencies are there ?  Would they have automatically have been installed by going the PlatformIO IDE route ?

Cossid, 2023-02-03:
I think he skipped over anything you would also need to do on regular esphome.  You likely need to run `pip install .` (which should install all the packages in requirements.txt which should have all version requirements handled as well) on the directory before you can use `python -m esphome`

donburch888, 2023-02-03:
Yes.  I followed the Tip for if it doesn't work...
```
pi@raspi4:~/esphome/libretuya-esphome $ pip uninstall esphome
WARNING: Skipping esphome as it is not installed.
pi@raspi4:~/esphome/libretuya-esphome $ pip install -e .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/esphome/libretuya-esphome
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting libretuya-esphome-dashboard
  Downloading https://www.piwheels.org/simple/libretuya-esphome-dashboard/libretuya_esphome_dashboard-

...

Successfully built zeroconf reedsolo
Installing collected packages: pycparser, sniffio, cffi, tzdata, typing-extensions, packaging, ifaddr, h11, cryptography, click, async-timeout, anyio, zeroconf, wsproto, uvicorn, tabulate, starlette, semantic-version, reedsolo, pytz-deprecation-shim, pyserial, pyelftools, protobuf, noiseprotocol, marshmallow, ecdsa, colorama, bottle, bitstring, ajsonrpc, aiofiles, tzlocal, tornado, PyYAML, platformio, paho-mqtt, libretuya-esphome-dashboard, kconfiglib, esptool, aioesphomeapi, esphome
  Running setup.py develop for esphome
Successfully installed PyYAML-6.0 aioesphomeapi-13.1.0 aiofiles-22.1.0 ajsonrpc-1.2.0 anyio-3.6.2 async-timeout-4.0.2 bitstring-3.1.9 bottle-0.12.23 cffi-1.15.1 click-8.1.3 colorama-0.4.6 cryptography-39.0.0 ecdsa-0.18.0 esphome esptool-4.4 h11-0.14.0 ifaddr-0.2.0 kconfiglib-13.7.1 libretuya-esphome-dashboard-20230120.0 marshmallow-3.19.0 noiseprotocol-0.3.1 packaging-23.0 paho-mqtt-1.6.1 platformio-6.1.5 protobuf-4.21.12 pycparser-2.21 pyelftools-0.29 pyserial-3.5 pytz-deprecation-shim-0.1.0.post0 reedsolo-1.5.4 semantic-version-2.10.0 sniffio-1.3.0 starlette-0.21.0 tabulate-0.9.0 tornado-6.2 typing-extensions-4.4.0 tzdata-2022.7 tzlocal-4.2 uvicorn-0.19.0 wsproto-1.2.0 zeroconf-0.47.1
pi@raspi4:~/esphome/libretuya-esphome $ python -m esphome
usage: __main__.py [-h] [-v] [-q] [-s key value] command ...
__main__.py: error: the following arguments are required: command
pi@raspi4:~/esphome/libretuya-esphome $ 
```
I guess that this is "working" ?  So far so good. 

I started on a fresh RasPi SDcard, so no previous esphome or other packages installed.  Biggest problem I have found with FOSS is the unwritten assumptions making me pull my hair out.

closed 2023-02-03 by donburch888

kuba2k2, 2023-02-03:
It will all get better when we get it merged into upstream. It's hard to predict all OS configurations ever, to know for sure what needs to be installed.