libretiny-eu/libretiny #75: esp8266_restore_from_flash: true  equivalent?

Stef-90, 2023-02-04:
Is there an option like this. It would be a nice feature. 

for example for light switches, in case power goes down, it would be nice to turn everything back to what it was before power out, instead of starting everything as off.

kuba2k2, 2023-02-04:
This is enabled by default. I believe the interval of saving settings is configurable, and defaults to something like 10 minutes. There's no option of saving state to RTC memory, so the only possibility is flash.

Stef-90, 2023-02-04:
Oh that's nice, I just found the project last night so I just quickly flashed one device, but didn't keep it on long enough for save.
I have bunch of beken devices collecting dust, because i didn't like other "tasmota like" approach as i use esphome for all my esp devices, now i can finally put those devices to good use.
Is there a way to change default save interval from yaml, or I would need to look into backend.

kuba2k2, 2023-02-04:
You can change it in yaml: https://esphome.io/components/esphome.html#adjusting-flash-writes

How can ESP devices boot in milliseconds? It takes time to scan for WiFi, connect to it, and boot everything up. The code itself starts running after few hundred milliseconds, but WiFi tasks take time. I believe it should be the same for ESP chips.

Stef-90, 2023-02-06:
Thanks this should do the trick.
I use fastconnect, so it doesn't  actually scan for wifi it connects on blind. I phrase it wrong you're right it's few hundreds ms, but still way faster then beken device, which looks like it takes 2-3seconds.
Here is an example i used smart touch wall switches as a test, pretty much identical besides mcu. Old model with esp and new version with beken.
I wired them on pieces of cord with power plug for test and when i plug esp device it goes into previous state in the blink of the eye and i can immediately as i plug it with one hand tap on it with other and it toggles relay right away.

Same procedure with beken it takes 2-3sec to go to previous state and intil then tapping doesn't toggle relay, no reaction at all.
 Looks strange, so it's my assumption that it takes longer time for rtos to boot, as code that runs after is identical .

kuba2k2, 2023-02-06:
I don't think that 2 seconds is a huge problem, unless you're turning it on with a manual switch, rather than a breaker. Fast connect is not supported yet.

mihsu81, 2023-02-07:
@kuba2k2 it does cause the relay in power plugs to toggle. During the reboot of a device you can hear the relay clicking for a second. Just enough to switch off the appliance plugged into it. Happens on a RTL8710BX and a CB2S.

Stef-90, 2023-02-09:
> @kuba2k2 it does cause the relay in power plugs to toggle. During the reboot of a device you can hear the relay clicking for a second. Just enough to switch off the appliance plugged into it. Happens on a RTL8710BX and a CB2S.

Yep, this is how i realized the delay in boot time initially. The other alternative firmware behave exactly same, I have no idea if it happened on stock all my device were flashed before actually trying them. But I'm starting to think either hardware or beken rtos is trash.

kuba2k2, 2023-02-10:
ESP32 has so-called RTC GPIOs, which are controlled by the RTC and aren't touched during a software reset. RTL and BK don't have these (not that I know) so every reboot will reset all the registers and clear all GPIOs.

Why do you need to reboot it in the first place? The stock firmware probably had that problem too, but since it didn't need to reboot, it wasn't an issue.