libretiny-eu/libretiny #76: WiFiUDP reference missing and needs to be included

lonerzzz, 2023-02-05:
In some code I am attempting to port from ESP32, there is a failure to resolve WiFiUDP.h. The file is not present in Arduino-API, but is present in [parent path]\platformio\platforms\libretuya\arduino\beken-72xx\libraries\WiFi

Where should it be included according to the current LibreTuya structure?

kuba2k2, 2023-02-05:
It can be added in libretuya/arduino/libretuya/libraries/.../WiFi.h I think. There's a file that includes all wifi functions (AP, STA, Generic,...) as well as WiFiClient and WiFiServer.

lonerzzz, 2023-02-05:
I saw that [parent path]\platformio\platforms\libretuya\arduino\libretuya\api\WiFi\WiFi.h has definitions for the classes defined in the same location. What I didn't see was external references from the entries in that folder. This would be the first as best I can tell so didn't want to add linkages that are unexpected.

lonerzzz, 2023-02-05:
Submitted the pull request to add for the Beken chips based on the existing, but not included entry.

https://github.com/kuba2k2/libretuya/pull/77

Not sure for other chipsets as I am not using these at this point.

closed 2023-03-02 by kuba2k2