libretiny-eu/libretiny #79: "sh: 1: arm-none-eabi-g++: not found" when compiling ESPHome

donburch888, 2023-02-06:
My first LibreTuya-ESPHome compile... 

Simple initial configuration for my Arlec PC191HA power points (uses WB2S module) is 
```
esphome:
  name: pc191ha

libretuya:
  board: wb2s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "dagoba"
  password: "gt49lkxyzzy"

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pc191Ha Fallback Hotspot"
    password: "7j78Uhtsl4Hi"

captive_portal:

# Example configuration entry
web_server:
  port: 80
```
gives
```
pi@raspi4:~ $ python -m esphome compile PC191HA.yaml
INFO Reading configuration PC191HA.yaml...
INFO Generating C++ source...
INFO Compiling app...
*****************************************************************************************************************************
Obsolete PIO Core v6.1.5 is used (previous was 6.1.6)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/en/latest/core/installation/troubleshooting.html
*****************************************************************************************************************************
Processing pc191ha (board: wb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
-----------------------------------------------------------------------------------------------------------------------------
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.33191e0 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/pc191ha/src/esphome/components/api/api_connection.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/pc191ha/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/pc191ha/src/esphome/components/api/api_pb2.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/pc191ha/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/pc191ha/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/pc191ha/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/pc191ha/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/pc191ha/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
================================================ [FAILED] Took 3.53 seconds ================================================
pi@raspi4:~ $ 
```

Am I missing something obvious here ?

kuba2k2, 2023-02-06:
See #48

@catalin2402

kuba2k2, 2023-02-06:
Please run Python and type:
```
import platform
print(platform.machine())
```

donburch888, 2023-02-06:
Its in my Raspberry Pi 4 with fresh install of Raspberry Pi OS Lite.
```
pi@raspi4:~ $ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> print(platform.machine())
aarch64
>>> 
```

catalin2402, 2023-02-06:
Ok, It's a 64 bit ARM installation, the toolchains are using the x86 executables for compilation, but Debian/ Raspberry PI OS doesn't allow them to run by default. If you want to compile using this machine to the following thing:

```
dpkg --add-architecture armhf
apt-get update
apt-get install -y libc6:armhf
```

donburch888, 2023-02-06:
I'm not wedded to using the RasPi 4 for this - just seemed an easy virtual machine. 

I can setup a VM on my intel i7 desktop PC running Ubuntu to do the ESPHome builds, if that would be better/easier. 
Or would it be feasible to install LibreTuya-esphome on my HAOS machine (a DELL intel i5 pc) instead of ESPHome ?

catalin2402, 2023-02-06:
you can install the Libretuya-ESPHome addon as a Home Assistant addon, you don't have to uninstall the original one (Libretuya-ESPHome is a separate addon). Libretuya-ESPHome Addon is not updated with the latest additions from ESPHome base repository, but it should work anyway

kuba2k2, 2023-02-06:
I wonder how could we make PIO toolchains a little bit less of a mess.

Btw. X86 strictly refers to Intel/AMD CPUs, 32 bit ARM is not x86.

catalin2402, 2023-02-06:
Yes, I know, I don't know why I'm always refering to arm 32 bit as x86 :)

dkaukov, 2023-02-07:
Hi, I have similar problem:
```
INFO Reading configuration /config/sw09.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing sw09 (board: wr2; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-realtek-amb1 @ 750.0.0-b0bc+sha.750b0bc 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.5d6edaa 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling .pioenvs/sw09/src/esphome/components/binary_sensor/automation.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/sw09/src/esphome/components/binary_sensor/binary_sensor.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/sw09/src/esphome/components/binary_sensor/filter.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/sw09/src/esphome/components/binary_sensor/automation.cpp.o] Error 127
*** [.pioenvs/sw09/src/esphome/components/binary_sensor/binary_sensor.cpp.o] Error 127
*** [.pioenvs/sw09/src/esphome/components/binary_sensor/filter.cpp.o] Error 127
Compiling .pioenvs/sw09/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/sw09/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o] Error 127
========================== [FAILED] Took 0.83 seconds ==========================
```

---

```
root@7c8d873a83a4:/config# python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> print(platform.machine())
x86_64
```

It is failing only on RT platform, Beken is fine.

dkaukov, 2023-02-07:
Little bit more info:
Libretuya installs 32bit version of the toolchain:
```
root@docker:/home/ziss/esphome-libretuya/.esphome/platformio/packages/toolchain-gccarmnoneeabi/bin# file ./arm-none-eabi-c++
./arm-none-eabi-c++: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.8, stripped
```
But docker image does not have 32bit version libc. Running this inside the docker fixes the problam:
```
apt-get install lib32z1
```

catalin2402, 2023-02-07:
@dkaukov Check this PR to make it work in docker: https://github.com/kuba2k2/libretuya-esphome/pull/5

dkaukov, 2023-02-07:
👍 LGTM

donburch888, 2023-02-11:
> you can install the Libretuya-ESPHome addon as a Home Assistant addon

I assume that requires adding "https://github.com/kuba2k2/libretuya-esphome" as a repository either in HA Add-ons or in HACS ... but neither of these work. for me.

In https://docs.libretuya.ml/docs/projects/esphome/#find-your-devices-board under Download ESPHome and GUI it gives CLI commands.  I am using HAOS, with "Terminal and SSL add-on".  Do I enter these at this terminal (which I believe is in the HA container); at the machine's HAOS console, or where ?

kuba2k2, 2023-02-11:
No, this "GUI" guide is not for the HA addon, but standalone container with the ESPHome dashboard. The HA addon was linked above.

donburch888, 2023-02-11:
Got it !

For any other newbies trying to follow this, the relevant post was 
> If you want to test, just add https://github.com/felipecrs/libretuya-esphome-hass-addon to your HA Addon Store and install it. It should work.

closed 2023-02-11 by donburch888