libretiny-eu/libretiny #81: compile errors

Fredfishface, 2023-02-14:
Hi,

I have been following the tutorial at https://docs.libretuya.ml/docs/projects/esphome/#create-your-device-config for CLI. 

When I build the project I have several MDNS errors. Specifically:

src/esphome/components/mdns/mdns_rp2040.cpp:21:54: error: no matching function for call to 'mDNS::begin(const char*, uint32_t)'

src/esphome/components/mdns/mdns_rp2040.cpp:43:26: error: no 'void esphome::mdns::MDNSComponent::loop()' member function declared in class 'esphome::mdns::MDNSComponent'

src/esphome/components/mdns/mdns_rp2040.cpp:46:8: error: 'MDNSResponder' has no member named 'close'

Fredfishface, 2023-02-15:
Hi, Found the glitch with verbose. Works as expected.

closed 2023-02-15 by Fredfishface

kuba2k2, 2023-02-15:
This was an error introduced in the last merge, I fixed it about 13 hours ago.