libretiny-eu/libretiny #82: compile errors

donburch888, 2023-02-17:
I'm on to my 3rd attempt - this time using my desktop PC (an i7 CPU running latest Ubuntu).  I have installed Docker and VScode. 

> platformio platform install https://github.com/kuba2k2/libretuya
> git clone https://github.com/kuba2k2/libretuya-esphome

seem to have installed correctly, and  "python3 -m esphome" gives me the option to choose LocalTuya board type.  

I have added the recommended options giving my pc191ha.yaml file:
```
esphome:
  name: pc191ha

libretuya:
  board: wb2s
  framework:
    version: dev

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "dagoba"
  password: "redacted"

  # Enable fallback hotspot in case wifi connection fails
  ap:
    ssid: "Pc191Ha Fallback Hotspot"
    password: "7j78Uhtsl4Hi"

captive_portal:

# Example configuration entry
web_server:
  port: 80
```

and compiling it gives massive output including...
```
pi@raspi4:~/esphome/libretuya-esphome $ python -m esphome compile PC191HA.yaml

...

Compiling .pioenvs/pc191/lwip2.1.3_bdk/src/api/netbuf.c.o
Compiling .pioenvs/pc191/lwip2.1.3_bdk/src/api/netdb.c.o
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/port/net.c: In function 'sta_ip_start':
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/port/net.c:387:12: warning: missing braces around initializer [-Wmissing-braces]
     struct wlan_ip_config address = {0};
            ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/port/net.c:387:12: warning: (near initialization for 'address.ipv4') [-Wmissing-braces]
Compiling .pioenvs/pc191/lwip2.1.3_bdk/src/api/netifapi.c.o
Compiling .pioenvs/pc191/lwip2.1.3_bdk/src/api/sockets.c.o

...

Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls-port/src/timing_alt.c.o
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls-port/src/tls_certificate.c.o
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_certificate.c:56:76: warning: backslash and newline separated by space [enabled by default]
 #define GLOBALSIGN_ROOT_CA                                                 \ 
 ^
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls-port/src/tls_client.c.o
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls-port/src/tls_hardware.c.o

...

Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls/library/aes.c.o
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls/library/aesni.c.o
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c: In function 'mbedtls_net_accept':
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c:349:5: warning: passing argument 5 of 'lwip_getsockopt' from incompatible pointer type [enabled by default]
     if( getsockopt( bind_ctx->fd, SOL_SOCKET, SO_TYPE,
     ^
In file included from /home/don/.platformio/platforms/libretuya/platform/common/fixups/lib_inc/sockets.h:5:0,
                 from /home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c:81:
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:582:5: note: expected 'socklen_t *' but argument is of type 'int *'
 int lwip_getsockopt (int s, int level, int optname, void *optval, socklen_t *optlen);
     ^
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c:360:50: warning: passing argument 3 of 'lwip_accept' from incompatible pointer type [enabled by default]
                                          (struct sockaddr *) &client_addr, &n );
                                                  ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:615:65: note: in definition of macro 'accept'
 #define accept(s,addr,addrlen)                    lwip_accept(s,addr,addrlen)
                                                                 ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:577:5: note: expected 'socklen_t *' but argument is of type 'int *'
 int lwip_accept(int s, struct sockaddr *addr, socklen_t *addrlen);
     ^
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c:368:33: warning: passing argument 6 of 'lwip_recvfrom' from incompatible pointer type [enabled by default]
                         (struct sockaddr *) &client_addr, &n );
                                 ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:639:81: note: in definition of macro 'recvfrom'
 #define recvfrom(s,mem,len,flags,from,fromlen)    lwip_recvfrom(s,mem,len,flags,from,fromlen)
                                                                                 ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:590:9: note: expected 'socklen_t *' but argument is of type 'int *'
 ssize_t lwip_recvfrom(int s, void *mem, size_t len, int flags,
         ^
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls-port/src/tls_net.c:409:34: warning: passing argument 3 of 'lwip_getsockname' from incompatible pointer type [enabled by default]
                          (struct sockaddr *) &local_addr, &n ) != 0 ||
                                  ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:623:70: note: in definition of macro 'getsockname'
 #define getsockname(s,name,namelen)               lwip_getsockname(s,name,namelen)
                                                                      ^
/home/don/.platformio/packages/library-lwip@2.1.3-bdk/src/include/lwip/sockets.h:581:5: note: expected 'socklen_t *' but argument is of type 'int *'
 int lwip_getsockname (int s, struct sockaddr *name, socklen_t *namelen);
     ^
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls/library/arc4.c.o
Compiling .pioenvs/pc191/bdk_mbedtls/mbedtls/library/asn1parse.c.o

...

Compiling .pioenvs/pc191/bdk_driver_spi/spi_master.c.o
Compiling .pioenvs/pc191/bdk_driver_spi/spi_slave.c.o
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls_ui/sl_tls.c: In function 'ssl_create':
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls_ui/sl_tls.c:69:3: warning: return makes pointer from integer without a cast [enabled by default]
   return -1;
   ^
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls_ui/sl_tls.c:72:20: warning: assignment makes pointer from integer without a cast [enabled by default]
  tls_session->host = os_strdup(url);  ///os_strdup(DEFAULT_MBEDTLS_WEB_SERVER);
                    ^
/home/don/.platformio/packages/framework-beken-bdk/beken378/func/mbedtls/mbedtls_ui/sl_tls.c:79:23: warning: assignment makes pointer from integer without a cast [enabled by default]
     tls_session->port = os_strdup(port);///os_strdup(DEFAULT_MBEDTLS_WEB_PORT);
                       ^
Compiling .pioenvs/pc191/bdk_ble_4_2/ble.c.o
Archiving .pioenvs/pc191/libbdk_driver_spi.a

...

Archiving .pioenvs/pc191/libbdk_ble_4_2.a
Indexing .pioenvs/pc191/libbdk_ble_4_2.a
Linking .pioenvs/pc191/firmware.elf
|-- Image 1: firmware.elf
|   |-- bk7231t_app_0x011000.rbl
|   |   |-- firmware.bin
|   |-- bk7231t_app_0x011000.crc
|   |-- bk7231t_app_0x129F0A.rblh
|   |-- bk7231t_app.ota.rbl
|   |-- bk7231t_app.ota.ug.bin
Building UF2 OTA image
|-- firmware.uf2
RAM:   [====      ]  44.0% (used 115320 bytes from 262144 bytes)
Flash: [=====     ]  54.8% (used 593612 bytes from 1083136 bytes)
====================================================== [SUCCESS] Took 118.86 seconds ======================================================
INFO Successfully compiled program.
don@muscle:~/esphome/libretuya-esphome$ 
```

So, has it compiled correctly ?  Will uploading the resulting file brick my device ?
And if not, is there anything I can do to fix the errors ?

Cossid, 2023-02-17:
Those are warnings, not errors.  Errors would result in a failed compile.  You should be able to upload that firmware without issues, it should not brick your device.

donburch888, 2023-02-17:
OK, thanks - and so quick a reply too !

Now to move the file (the "build directory" is well and truly hidden) to my RasPi so I can upload it with cloudcutter to my device.

kuba2k2, 2023-02-17:
It is a bit hidden, yes. That's why I added paths of the directory to the docs.

About the warnings: the part where it says `[enabled by default]` means that the compiler always shows these warnings, and it's not possible to disable them with any CLI flag, sadly. I'm not sure why some warnings are always visible.

donburch888, 2023-02-17:
Unfortunately the fact that the compile worked has not stopped it bricking 2 of my devices :-(  Not happy

At first I thought it was cloudcutter not returning the device to the WLAN after uploading the firmware; but no the device woulddn't even turn on when power turned on.  Thinking maybe it was a single unit going flakey after so many attempts I tried a second unit and after first upload it won't turn on.

donburch888, 2023-02-18:
FYI: 
I posted here last night that this has bricked 2 of my devices ... they are Arlec PC191HA power plugs, and when power to the plug is turned on they do not pass power through, the switch does not work, and neither does the LED in the switch.  They appear dead. 

Then after dinner I noticed 2 new ESPHome devices had been detected by HA ;-)

However the HA LibreTuya-ESPHome integration provides no useful information, and the power plug itself doesn't work (eg to put it into slow blink mode) - no doubt because of my absolutely minimal configuration.  

The device does display a basic web interface at its IP address, and it is listed on the Wireless statistics page of the old AP it connects to, and in the ARP cache table of my DrayTek router - but strangely does not show on the (otherwise user friendly) Network Map of my shiny new ASUS RT-AX55 router (which likes to think it's really in charge of the network, despite being in AP mode). 

More research required ...

closed 2023-02-18 by donburch888

kuba2k2, 2023-02-18:
So the devices are working correctly. You flashed a minimum configuration, that's why the plug doesn't power on. "Slow blink mode" is a Tuya thing, and it's gone as soon as you flash custom firmware.

You have a web interface, which is good, since using that you can flash an OTA update with actual components added to the YAML (like switch, button, etc). That's how users usually do this, after cloudcutting a device they write configs that control the switches.