libretiny-eu/libretiny #83: Support smartconfig (airkiss)?

dinnaAPP, 2023-02-17:
Hi, does it support Smartconfig (ESP airkiss)?
Thaks!

kuba2k2, 2023-02-18:
There's some code related to Airkiss in the Beken SDK, but it's not officially supported nor tested by us.

dinnaAPP, 2023-02-22:
Thank you very much for your answer. I have seen that library in the SDK, could you tell me how to use it?

kuba2k2, 2023-02-22:
Not really, since I don't know how to use it :smile:

dinnaAPP, 2023-02-22:
Jajaj ok. Airkiss is a protocol to send router credentials to the BK. Both the APP and the BK must be in airkiss mode, basically packets are sent and a sniffing process takes place to collect them for a certain time. It is very useful.
On the ESP8266/32 it is called smartconfig and esp-touch.
You should find a way to put the BK in that mode.
Greetings.

kuba2k2, 2023-02-22:
I'm just saying that I've never used this library, so I don't know how to guide you to use it - you have to find out by yourself.