libretiny-eu/libretiny #85: Support reset using DTR

zultron, 2023-02-19:
Could LibreTuya support programmers that can reset the modules for programming with the DTR signal, like ESPHome does?  That would make flashing WB3S modules as brain-dead easy as flashing ESP-12F modules with ESPHome.

Thanks SO MUCH for this system.  Before now, I was transplanting ESP-12F modules onto Treatlife DS02S dimmers with WB3S modules.  Even after several operations, I was still very slow, and still ripped off solder pads.

LibreTuya helped speed up integrating these switches into HASSIO by 300%.  Now I just do enough desoldering to slip a piece of paper under the EN and VCC pads, solder five wires to the programmer, and run LibreTuya.  If LibreTuya toggled DTR before upload, I wouldn't even have to wait for the right moment to ground the EN pin.  Then I just solder the EN and VCC pads back to the main board and it's done!

The other roadbump I ran into was `bk7231tools` breaking on my Linux system, but that one was [easy to fix][1].

[1]:  https://github.com/tuya-cloudcutter/bk7231tools/pull/11

kuba2k2, 2023-02-19:
Thanks for the PR!
LibreTuya's flasher is based on [ltchiptool](https://github.com/libretuya/ltchiptool), our custom utility for flashing chips and managing binaries. When it establishes connection to the chip, first it checks if it's available already. Then, after 2 seconds, it does a hardware reset by toggling RTS and DTR lines, and tries to connect again. Does it work after waiting 2 seconds?
After next 2 seconds, it should print a connection guide in the terminal. Does it do that?

Cossid, 2023-02-19:
@zultron If you weren't aware, the Treatlife DS02S is supported by [Tuya-CloudCutter](https://github.com/tuya-cloudcutter/tuya-cloudcutter) for OTA flashing and avoids soldering altogether.

zultron, 2023-02-19:
Thanks for the tips, that helped direct me as I did a little digging.  TLDR; I've gotten this to work, but lost track of exactly what might have changed, and now can't repeat the failures.  ;(  I'm out of town next week, but I'll be back at this afterwards.  Feel free to close this out.

-----

Notes on what I did for when I pick this up again.  Some of this is from memory, so it's possible some facts are incorrect or omitted.

I'm using a burner [like this][1].  I'm not 100% sure the schematic is the same as my burner, but experimenting with combinations of setting/clearing RTS and DTR give the following outputs:

```
DTR RTS  EN RST GPIO0
1   1   1  1   0
1   0   1  1   0
0   1   0  0   0
0   0   1  1   1
```

Flashing the WB3S by jumpering only GND, 3V3, TX, RX and EN (i.e. the above-described method of lifting 3V3 and EN pads  from the DS02S board and soldering wires) wasn't working yesterday without pressing the burner's RST button, which pulls EN & RST low.

Today, I tried flashing a few WB3S module plugged directly into the burner (i.e. WB3S modules completely removed from the DS02S board), and same results, burning failed.  I don't remember if I tried hitting the RST button during this activity.

Then I went and cloned the `ltchiptool` repo.  I couldn't find where DTR/RTS were manipulated.  (The `ltchiptool.soc.ambz.util.rtltool.RTLXMD.connect()` method manipulates DTR/RTS, but IIUC that's N/A for the WB3s, which uses the `ltchiptool.soc.bk72xx` module instead.)  In any case, I installed `ltchiptool` into the venv from git.

Next, I found the `bk7231tools.serial.protocol.BK7231Protocol.reset()` method, and tried toggling only RTS (using the above truth table), with no success.

Next, I flashed using the RST button, success.  Then I lose track, and can't remember whether I got any more failures, but unexpectedly, the flashing started succeeding at some point.  I found that when the chip detection succeeds, the `ltchiptool.soc.ambz.util.rtltool.RTLXMD.connect()` method never runs, so that's a red herring.

When I get back to this, I'll try flashing a few more modules still attached to the DS02S PCB.  If that succeeds, I'll wipe the entire install and see if I can retrace my steps (install following the LibreTuya docs from `pip`, no cloned git repos).

[1]: https://www.pcbway.com/project/shareproject/esp8266_burning_rack.html

kuba2k2, 2023-02-19:
The buttons on the burner might be wired via transistors, which change the logic levels. I'm not sure how do you differentiate EN/CEN and RST, but there's only one "chip enable" pin on BK7231. 

The CEN pin, short for chip enable, resets the chip when low, and allows it to run when high. The DTR and RTS pins are commonly used on ESP8266, which also requires toggling GPIO0 to put it in download mode.

In theory, only one line is needed to reset the chip. People are used to go with DTR->CEN; I like using RTS instead, because it's kind of like RST which makes me remember "this is the one to reset the chip" 😆
LibreTuya (and ltchiptool) toggles both DTR and RTS simultaneously. The code you found in RTLXMD is not applicable here, as it's for Realtek chips. The code in `bk72xx` has a `flash_hw_reset()` method, which is called after first 2 seconds of being unable to communicate with the chip. This method calls BK7231Tools's `reset()`, which in turn toggles the pins.

I have used that method some time ago; it worked fine, then it didn't work, then it worked again, then I didn't use it anymore, as I was flashing LT over LT, so I had the auto-reset feature. In the end, I'd say it should still work, but it can also be broken.

zultron, 2023-02-26:
Back in town now, I just burned three more modules.  One thing that was confusing me, the LibreTuya firmware doesn't need reset by pulling EN low to flash, but the original firmware does.

I submitted a pull request against the bk7231tools repo.  `BK7231Protocol.hw_reset()` now puts a short pause between clearing DTR and RTS (also note the changed order) for these wacky burners.  That fixes the problem for me.
https://github.com/tuya-cloudcutter/bk7231tools/pull/14

kuba2k2, 2023-02-26:
>  One thing that was confusing me, the LibreTuya firmware doesn't need reset by pulling EN low to flash, but the original firmware does.

And that's a feature! :smiley: 
Read more here: https://docs.libretuya.ml/docs/flashing/tools/adr/

Thanks for the PR, I'll try to review it soon, but remind me if I forget.

zultron, 2023-02-26:
> The buttons on the burner might be wired via transistors, which change the logic levels. I'm not sure how do you differentiate EN/CEN and RST, but there's only one "chip enable" pin on BK7231.

Sorry, the "RST" and "PROG" buttons on the burner got me confused, ignore that.  The burner controls EN and GPIO0 outputs separately, depending on combination the DTR and RTS signal inputs, according to the truth table above.

> In theory, only one line is needed to reset the chip. People are used to go with DTR->CEN; I like using RTS instead, because it's kind of like RST which makes me remember "this is the one to reset the chip" laughing LibreTuya (and ltchiptool) toggles both DTR and RTS simultaneously.

That's a nice mnemonic.  OTOH, consider sticking with established convention (assuming DTR is the convention) so this sort of unexpected problem doesn't crop up.  Maybe a mnemonic for DTR, "Data Terminal Ready", is when it goes low, it resets the chip, which becomes not Ready.  :D

 The code you found in RTLXMD is not applicable here, as it's for Realtek chips. The code in `bk72xx` has a `flash_hw_reset()` method, which is called after first 2 seconds of being unable to communicate with the chip. This method calls BK7231Tools's `reset()`, which in turn toggles the pins.

Thanks for the pointer!  I think I'd already found it when you posted, but that would have saved me some grepping around.

> I have used that method some time ago; it worked fine, then it didn't work, then it worked again, then I didn't use it anymore, as I was flashing LT over LT, so I had the auto-reset feature. In the end, I'd say it should still work, but it can also be broken.

Please keep it!  :D 

BTW, this code is very well organized.  I spend a lot of time with a large pile of spaghetti Python code, so I really appreciate this code.  Thanks!

closed 2023-02-26 by zultron