libretiny-eu/libretiny #86: Please add to the list of supported chips WBLC9

esnet146, 2023-02-20:
Successfully flashed and working
https://developer.tuya.com/en/docs/iot/wblc9-module-datasheet?id=K9hgglry2jp5h

Tested on device RGB+CW Lapm Tuya WBLC9 (BK7231T) + BP1658
https://www.elektroda.com/rtvforum/topic3925796.html

kuba2k2, 2023-02-20:
WBLC9 is not a chip, it's a "board" - a small PCB with BK7231T *chip* on it. "Supported" in this context means more like "has board definition files" than "works". Any board having a supported chip (which you can find a list of [here](https://docs.libretuya.ml/docs/status/supported/#cpu-list)) will work.

esnet146, 2023-02-22:
You are right, but there is a full description of the board. URL attached. or is something missing?

kuba2k2, 2023-02-22:
Yes, the board documentation on docs.libretuya.ml, as well as pin definitions in code are missing.