libretiny-eu/libretiny #87: missing some macros in Arduino framework

mczerski, 2023-02-20:
There is no digitalPinToInterrupt macro, also other simmilar macros are missing. Also I cannot use D# pin names, like D17 - insead I need to use raw numbers like 17.

kuba2k2, 2023-02-20:
Why do you need this macro? The attachInterrupt() function accepts the D# number, there's no special value for interrupts.

mczerski, 2023-02-20:
according to the arduino documentation digitalPinToInterrupt should be used to pass interrupt number:
https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/

I think that compliance with arduino is enough reason. 

Regarding D#. Where are those pin names defined ?? My code does not compile if I use those pin names.

kuba2k2, 2023-02-21:
The D# pin names are not defined; normally you should simply use the number without the D. 

What CPU are you building for? I think the macros were defined for Realtek at least. I'll add them when I get the chance.

mczerski, 2023-02-21:
I'm building for beken-72xx. But the libraries I use are made for wide range of plaforms and one of them uses digitalPinToInterrupt.
Regarding D# pin names, that`s fine if they are not defined but it is quite misleading since in the documentation and in issues on github those names are used (event in this issue You mentioned to use D# pin names with attachInterrupt :)) So it would be nice if either those D# pin names were defined or it was made clear in the documentation to use raw numbers.

kuba2k2, 2023-02-21:
How does that work for normal Arduino programs? Can you use D# names there? I'm not sure as I haven't used Arduino for a long time.
I mentioned using the D# *number* 😀
Sure, we can define these names as macros, but I'm not sure if that won't conflict with something else that might be named "D0".

mczerski, 2023-02-21:
Hmm, does libretuya arduino framework not produce "normal Arduino programs" ? Maybe I'm confused but on the main page of guthub repo I ca read "the project focuses on developing working Arduino-compatible cores for supported families". Based on that I would assume that Arduino core supplied by this project should be similar to other Arduino cores - correct me if I'm wrong.

So going back to Your question, in other Arduino cores pin names like D1 are just #defines or enums for the raw pin numbers. Also some Arduino cores (like stm32duino) define additional pin names related to actual pins on the MCU package. So one could refer to the pin trough D# or PA# names. This double naming would be nice in this project also since for every board there is table with assignment of t he D# pin to actual pin of the MCU package. So for example I could write the code like:
```
pinMode(D17, OUTPUT);
```
or 
```
pinMode(P26, OUTPUT);
```
for Generic BK7231N board.

mczerski, 2023-02-21:
You can look [here](https://github.com/stm32duino/Arduino_Core_STM32/blob/main/cores/arduino/pins_arduino.h) to se how stm32duino is doing that.

kuba2k2, 2023-02-21:
Okay, I see. The P## mapping is supported for ESPHome (where you supply the pin numbers in YAML configuration files). The actual macro is called PIN_FUNCTION_P##, which you can even use in Arduino code.
The reason for not calling the macro simply "P##" is that the Beken and Realtek SDKs define similar macros/enums, which have different values than LT's macros.
Do you have an idea how would such a macro look like? Something like PIN_P##?

mczerski, 2023-02-25:
Those P## pin names are not required or defined by Arduino. They are just to make code more readable and are platform dependent. So You can choose any name that not clashes with SDK names. I think that PIN_P## is very good name for it.

kuba2k2, 2023-03-02:
Including this change in the refactor :smiley: 

I found some macros here:
https://garretlab.web.fc2.com/en/arduino/inside/hardware/arduino/avr/cores/arduino/Arduino.h/
so I'm going to add them as well. Do you have any other macros that I should add?

closed 2023-07-13 by kuba2k2