libretiny-eu/libretiny #88: WR2 - RTL8710BN

esnet146, 2023-02-21:
device is not started.

module: wr2 Chip: RTL8710BN

Config:
substitutions:
  board_name: "rtl"

esphome:
  name: $board_name

libretuya:
  board: wr2
  framework:
    version: dev

logger:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "sid"
  password: "password"

  ap:
    ssid: $board_name
    password: "passswrd"

button:
  - platform: restart
    name: Reset.$board_name

log:

INFO Reading configuration testRTL.yml...
INFO Generating C++ source...
INFO Compiling app...
Processing rtl (board: wr2; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/libretuya/wr2.html
PLATFORM: LibreTuya (0.12.6+sha.8f447a4) > WR2 Wi-Fi Module
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.5d6edaa
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-ltchiptool @ 2.0.2+sha.7559033
 - toolchain-gccarmnoneeabi @ 1.50201.0 (5.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
`.pioenvs\rtl\firmware.elf' is up to date.
Building UF2 OTA image
Checking size .pioenvs\rtl\firmware.elf
|-- firmware.uf2
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.2% (used 52972 bytes from 262144 bytes)
Flash: [=====     ]  51.3% (used 413964 bytes from 806912 bytes)
========================= [SUCCESS] Took 10.53 seconds =========================
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] COM4 (USB Serial Port (COM4))
  [2] Over The Air (rtl.local)
(number): 1
Processing rtl (board: wr2; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/libretuya/wr2.html
PLATFORM: LibreTuya (0.12.6+sha.8f447a4) > WR2 Wi-Fi Module
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
DEBUG: Current (openocd) External (custom, openocd)
PACKAGES:
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.5d6edaa
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1
 - tool-ltchiptool @ 2.0.2+sha.7559033
 - tool-openocd @ 2.1100.211028 (11.0)
 - toolchain-gccarmnoneeabi @ 1.50201.0 (5.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Building UF2 OTA image
|-- firmware.uf2
Checking size .pioenvs\rtl\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.2% (used 52972 bytes from 262144 bytes)
Flash: [=====     ]  51.3% (used 413964 bytes from 806912 bytes)
Configuring upload protocol...
AVAILABLE: uart
CURRENT: upload_protocol = uart
Looking for upload port...
Using manually specified: COM4
Uploading firmware.uf2
|-- Detected file type: UF2 - esphome 2023.3.0-dev
|-- Connecting to 'Realtek AmebaZ' on COM4 @ 1500000
|   |-- Success! Chip info: Realtek RTL87xxB
|-- Writing 'D:\libretuya\libretuya-esphome\.esphome\build\rtl\.pioenvs\rtl\firmware.uf2'
|   |-- esphome 2023.3.0-dev @ 2023-02-21 20:41:51 -> wr2
#################################################################
|   |-- Finished in 12.675 s
========================= [SUCCESS] Took 24.64 seconds =========================
INFO Successfully uploaded program.
INFO Starting log output from COM4 with baud rate 115200

reboot: power off
uart2:
ROM:[V0.1]
FLASHRATE:4
UARTIMG_Download 2
Open xModem Transfer on Log UART...

flash ltchiptool-v3.0.0.exe 
[image]
RESULT analogous

kuba2k2, 2023-02-21:
Is that the only log you're getting on the UART after flashing?

Do you have any GPIOs pulled down? Like GPIO0?

esnet146, 2023-02-21:
1.Yes!
2.No gpio. free

esnet146, 2023-02-21:
[image]
Gpio0 ?

kuba2k2, 2023-02-21:
Is the board soldered to the original device, or is it desoldered?

Can you dump the flash using ltchiptool?

esnet146, 2023-02-21:
Disconnected from the device. only uart
Full dump available spi flash

esnet146, 2023-02-21:
P.S.
[image]

esnet146, 2023-02-22:
Fantastic. Turn off gnd uart
[19:43:37]FLASHRATE:4
[19:43:37]BOOT TYPE:0 XTAL:40000000
[19:43:37]IMG1 DATA[1168:10002000]
[19:43:37]IMG1 ENTRY[8000541:100021ef]
[19:43:37]IMG1 ENTER
[19:43:37]CHIPID[000000ff]
[19:43:37]read_mode idx:0, flash_speed idx:0
[19:43:37]calibration_result:[1:9:9][5:d]
[19:43:37]calibration_result:[2:13:7][1:d]
[19:43:37]calibration_result:[3:3:3][1:3]
[19:43:37]calibration_ok:[2:13z7]
[19:43:37]FLASH CALIB[NEW OK]
[19:43:37]OTA2 ADDR[80d0000]
[19:43:37]OTAx SELE[fffffffc]
[19:43:37]OTA1 USE
[19:43:37]IMG2 DATA[0x807a5f0:3608:0x10005000]
[19:43:37]IMG2 SIGN[RTKWin(10005008)]
[19:43:37]IMG2 ENTRY[0x10005000:0x802f3b1]
[19:43:37]System_Init1
[19:43:37]System_Init2
The log freezes, there is nothing on the wifi network

kuba2k2, 2023-02-22:
Dump the firmware and attach the dump here, otherwise I'm not able to tell what's wrong. Also, try flashing it again and then disconnect GND from uart properly.

esnet146, 2023-02-22:
[test.zip](https://github.com/kuba2k2/libretuya/files/10807324/test.zip)
Step:
Flash original_firmvare.bin - tested - fully functional
flash testRTL.yml : python -m esphome run testRTL.yml 
comport
not started
firmware.uf2 
Read ltchiptool-v3.0.0.exe  full dump: flash_libertuya.bin

esnet146, 2023-03-07:
My friends, the project is very interesting, but so far I can not understand what the problem is. I see that many people are flashing realtek, but I can’t figure out how to do this or what else to check. So far, I just realized that the uart did not work quite right. this is fixed, but the result is the same.

kuba2k2, 2023-03-07:
It's hard for me to check without having the device. I didn't yet have time to flash your dump and test it, sorry for that.

esnet146, 2023-05-10:
Problem solved.
The chip is incorrectly specified in the documentation for the module.
https://developer.tuya.com/en/docs/iot/wifiwr2module?id=K9605tko0juc3
_It consists of a highly integrated wireless RF chip (RTL8710BN)_
[image]
After soldering the cover, I saw another chip. W302 
[image]
It is successfully flashed as wr3le.
Thanks for your work!

closed 2023-05-10 by esnet146

esnet146, 2023-05-10:
Yes, the problem was in the wrong value of the clock frequency. The update didn't help. Only change chip to BX. W302=RTL8710BX