libretiny-eu/libretiny #89: arm-none-eabi-g++ not found / libc6-i386 missing on e.g. Debian Buster

hn, 2023-02-23:
When compiling e.g. on Debian Buster, you might get the following error:

```
Building in release mode
Compiling .pio/build/generic-rtl8710bn-2mb-788k/src/main.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pio/build/generic-rtl8710bn-2mb-788k/src/main.cpp.o] Error 127

```

which is caused by 

```
$ ldd ~/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-c++
        not a dynamic executable
$ file ~/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-c++
ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked,
interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.8, stripped

```

and can be fixed by

```
apt install libc6-i386
```

Hope this saves you some time :)

kuba2k2, 2023-02-23:
Yeah, PlatformIO's compilers are absolutely broken.

esnet146, 2023-02-28:
Debian GNU/Linux 11, docker 23.0.1
1. apt install libc6-i386
2. git clone https://github.com/kuba2k2/libretuya-esphome
3. python docker/build.py --tag libretuya --arch amd64 --build-type docker build
4. docker-compose up
5. in log:
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-amb1 @ 2.1.3-amb1+sha.6297b80 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
No dependencies
Compiling .pioenvs/rtl/src/esphome/components/api/api_connection.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/rtl/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/rtl/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/rtl/src/esphome/components/api/api_frame_helper.cpp.o] Error 127

hn, 2023-02-28:
My bug report relates to a non-docker setup.

@esnet146 Probably you are missing the libc6 /lib/ld-linux.so.2 _inside_ the docker container?

kuba2k2, 2023-02-28:
Wait, so building on x86 is also broken...? I thought it only had problems for arm64...
Anyway, yes, installing libc6 outside the docker container will not install it inside it.

diogopms, 2023-04-28:
I'm using a pi4 using the `libretuya-esphome`, and I'm seeing the same issue. 

[image]

kuba2k2, 2023-04-28:
This is inside the docker container, are you sure you built the 64-bit version of the container?

diogopms, 2023-05-02:
I'm using pi4 with raspberry pi os 32 bits.

blfriedman, 2023-05-04:
I'm getting a similar issue on a RPi 4, see build.txt attached.

This was built using the docker container that I built by following the instructions located at: https://docs.libretiny.eu/docs/projects/esphome/#download-esphome and substituting "armv7" for "amd64".

[build.txt](https://github.com/kuba2k2/libretiny/files/11391996/build.txt)
[gs-sw5bk-aff8fc.yaml.txt](https://github.com/kuba2k2/libretiny/files/11392007/gs-sw5bk-aff8fc.yaml.txt)

for architecture, I check with lscpu and it shows:
`
brucef@raspberrypi:~ $ sudo lscpu
Architecture:                    aarch64
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
Vendor ID:                       ARM
Model:                           3
Model name:                      Cortex-A72
Stepping:                        r0p3
`

kuba2k2, 2023-05-04:
This seems to be a problem with PlatformIO GCC compiler being downloaded for a different architecture. Not sure if there's anything to fix here on our side.

For now try using the prebuilt docker images from here:
https://github.com/libretiny-eu/esphome-hass-addon