libretiny-eu/libretiny #90: read/write data in flash memory

dinnaAPP, 2023-02-24:
Hi, I need to write and read data and I understand that RTC memory cannot be used, therefore I must use flash memory. Could you give me an example of how to do it?
Basically I want to store wifi ssid and pass.
Regards.

kuba2k2, 2023-02-24:
There's a Flash library:
https://docs.libretuya.ml/ltapi/class_flash_class/

And there's also FlashDB, which is used in ESPHome to store Key-Value preferences (which looks like a perfect fit for your use case):
https://armink.github.io/FlashDB/#/sample-kvdb-basic
https://github.com/kuba2k2/libretuya-esphome/blob/platform/libretuya/esphome/components/libretuya/preferences.cpp

dinnaAPP, 2023-02-24:
Thanks! I'm going to check it

dinnaAPP, 2023-02-27:
Hi kuba2k2, I am using FlashClass for reading/writing. I would need help with the offset, since I can read but when I write, the program freeze, I guess I am writing over the program.

kuba2k2, 2023-02-27:
What CPU are you using?
Flash offset tables are in the docs. For example, for Realtek:
https://docs.libretuya.ml/boards/wr3/#flash-memory-map

dinnaAPP, 2023-02-27:
BK7231N, I had not seen that info. Read/write tested, OK!, Thanks.

closed 2023-03-03 by dinnaAPP