libretiny-eu/libretiny #92: IMPORTANT: New project name

kuba2k2, 2023-02-25:
The project name is about to be changed to something else, for a few reasons:
- to address any issues that may arise due to the possibility of "LibreTuya" name violating Tuya Inc.'s trademark rights. We don't want any legal trouble, nor to interfere with Tuya's marketing.
- the project is meant to support more than RTL8710BN/BK7231 - that means it will work with chips not necessarily developed/distributed by Tuya.

The rules are simple - it should be nice, have two words only: first word being `Libre`, second word starting with the letter T. Why? Because there's just *way too many* references and usages of "LT" in code, documentation, tools, etc. Also, *Libre* sounds cool.

We've had a discussion on our Discord server about name ideas. **You can vote under this post, by adding an appropriate reaction** (the emojis don't indicate any preference).

Emoji | Name | Comment
--|--|--
:+1: | LibreThings | It seems to relate to SmartThings, which may be confusing
:smile: | LibreTiny | "As in something that runs on small devices"
:tada: | LibreTurtle | "Turtles don't sue, and they're dope" idk, I kinda like this one
:heart: | LibreThink | 
:rocket: | LibreTech | 

If you have any good suggestions, feel free to post them in this thread, and I can add them to the list.

If you really need to, you can cast more than one vote. Thanks!

kuba2k2, 2023-02-25:
After careful consideration, I'm pretty much convinced about using LibreTiny as the name, despite the voting showing otherwise. I'd like to be satisfied about the project, so that might ultimately be my choice. I'm sorry for any inconvenience caused by this.

dbuezas, 2023-02-25:
LibreTiny is cool! 👍 
It is absolutely your choice, and a good idea to gather community opinions before deciding, regardless of your final pick

heffneil, 2023-02-25:
I love Tinkerer so LibreTinker for me!

agittins, 2023-02-25:
LibreTiny is a pretty solid choice, imo - it speaks to the smaller devices that these chips are most likely to be found in.

Another option might be LibreToyou (pronounced Libre To You) as a play on going from Tuya (to ya) to To You - however that might be a somewhat Australian-specific play on words :-)

Turtles are also dope, though.

digiblur, 2023-02-25:
It's on the ForYuh network ;)

hn, 2023-02-25:
LibreTino = Libre Tiny Arduino (compatible cores)

marcusrbrown, 2023-02-25:
**LibreToys**

_Libre_ can be a synonym for _free_ (i.e., FLOSS, OSS, etc.). LibreToys = "the free(ing of) toys."

Habbie, 2023-02-26:
LibrIOT / LibrioT? probably not better than some of the given options but wanted to share this thought anyway.

fantasmisiciliani, 2023-02-28:
Why not LibreTuma? 
https://en.m.wikipedia.org/wiki/Tuma_(cheese)
Yummy!😋 🙈

jbornema, 2023-03-01:
I bring in another one:
**LibreSmart**

descipher, 2023-03-15:
LibreType could be an option. Type cast away .. lol

kuba2k2, 2023-04-24:
Looking at the results of the poll, seems like LibreTiny is our winner :tada: 

Again, I apologize for my 2nd post in this thread, possibly causing any influence on your votes. I hope that the new name will meet everyone's expectations.

And huge thanks to @Chrille#2270 at Discord for originally suggesting this name!

closed 2023-04-24 by kuba2k2