libretiny-eu/libretiny #96: ESPHome install error on Ubuntu VM.

edautz, 2023-03-05:
After flashing a lot of LSC Connect tuya devices based on ESP stuff, I stumbled at you amazing project. 
I want reflash some new LSC Connect tuya devices based on BK7231T/N chips. 

Fore that I tried to get your modified ESPHome at work on a Ubuntu 20.04.5 LTS VM on a Windows 10 laptop.

I followed the steps on: https://docs.libretuya.ml/docs/projects/esphome/

Installed Ubuntu, Docker, Docker-compose, PlatformIO, Python2, Python3 and all the github code.
But I stuck with the container creating proces for your modified version of ESPHome:

cd libretuya-esphome
python docker/build.py --tag libretuya --arch amd64 --build-type docker build 

File "docker/build.py", line 58
    build_to: str

I tried also the CLI method:

python3 -m esphome
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/cloutcutter/libretuya-esphome/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/home/cloutcutter/libretuya-esphome/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/home/cloutcutter/libretuya-esphome/esphome/config.py", line 11, in <module>
    import voluptuous as vol
ModuleNotFoundError: No module named 'voluptuous'

and

python -m esphome
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 163, in _run_module_as_main
    mod_name, _Error)
  File "/usr/lib/python2.7/runpy.py", line 114, in _get_module_details
    return _get_module_details(pkg_main_name)
  File "/usr/lib/python2.7/runpy.py", line 119, in _get_module_details
    code = loader.get_code(mod_name)
  File "/usr/lib/python2.7/pkgutil.py", line 281, in get_code
    self.code = compile(source, self.filename, 'exec')
  File "/home/cloutcutter/libretuya-esphome/esphome/__main__.py", line 59
    safe_print(f"  [{i+1}] {desc}")

also with sudo but the problem persists.

kuba2k2, 2023-03-05:
Try 
python3 -m pip install . 
In the ESPHome directory.
Then run with python3 -m esphome

edautz, 2023-03-06:
Ok.

Tried that. I noticed the default python version 3.8 was not supported. Upgraded to 3.10.

After that:

To run python3 -m pip install . 

succesful I had manually first 

sudo apt-get purge python3-yaml.

Then

python3 -m pip install . installed a ton of software.

python3 -m esphome
usage: __main__.py [-h] [-v] [-q] [-s key value] command ...
__main__.py: error: the following arguments are required: command

**seems to be working.**

I want to get the ESPHome container:

python3 docker/build.py --tag libretuya --arch amd64 --build-type docker build

But it failed:

python3 docker/build.py --tag libretuya --arch amd64 --build-type docker build
$ docker buildx build --build-arg BASEIMGTYPE=docker --build-arg BUILD_VERSION=libretuya --cache-from type=registry,ref=ghcr.io/esphome/esphome-amd64:cache-dev --file docker/Dockerfile --platform linux/amd64 --target docker --tag esphome/esphome-amd64:libretuya --tag esphome/esphome-amd64:dev --tag ghcr.io/esphome/esphome-amd64:libretuya --tag ghcr.io/esphome/esphome-amd64:dev .
**unknown flag: --build-arg**
See 'docker --help'.

Usage:  docker [OPTIONS] COMMAND

A self-sufficient runtime for containers

Options:
      --config string      Location of client config files (default
                           "/root/.docker")
  -c, --context string     Name of the context to use to connect to
                           the daemon (overrides DOCKER_HOST env var
                           and default context set with "docker
                           context use")
  -D, --debug              Enable debug mode
  -H, --host list          Daemon socket(s) to connect to
  -l, --log-level string   Set the logging level
                           ("debug"|"info"|"warn"|"error"|"fatal")
                           (default "info")
      --tls                Use TLS; implied by --tlsverify
      --tlscacert string   Trust certs signed only by this CA
                           (default "/root/.docker/ca.pem")
      --tlscert string     Path to TLS certificate file (default
                           "/root/.docker/cert.pem")
      --tlskey string      Path to TLS key file (default
                           "/root/.docker/key.pem")
      --tlsverify          Use TLS and verify the remote
  -v, --version            Print version information and quit

Management Commands:
  builder     Manage builds
  config      Manage Docker configs
  container   Manage containers
  context     Manage contexts
  image       Manage images
  manifest    Manage Docker image manifests and manifest lists
  network     Manage networks
  node        Manage Swarm nodes
  plugin      Manage plugins
  secret      Manage Docker secrets
  service     Manage services
  stack       Manage Docker stacks
  swarm       Manage Swarm
  system      Manage Docker
  trust       Manage trust on Docker images
  volume      Manage volumes

Commands:
  attach      Attach local standard input, output, and error streams to a running container
  build       Build an image from a Dockerfile
  commit      Create a new image from a container's changes
  cp          Copy files/folders between a container and the local filesystem
  create      Create a new container
  diff        Inspect changes to files or directories on a container's filesystem
  events      Get real time events from the server
  exec        Run a command in a running container
  export      Export a container's filesystem as a tar archive
  history     Show the history of an image
  images      List images
  import      Import the contents from a tarball to create a filesystem image
  info        Display system-wide information
  inspect     Return low-level information on Docker objects
  kill        Kill one or more running containers
  load        Load an image from a tar archive or STDIN
  login       Log in to a Docker registry
  logout      Log out from a Docker registry
  logs        Fetch the logs of a container
  pause       Pause all processes within one or more containers
  port        List port mappings or a specific mapping for the container
  ps          List containers
  pull        Pull an image or a repository from a registry
  push        Push an image or a repository to a registry
  rename      Rename a container
  restart     Restart one or more containers
  rm          Remove one or more containers
  rmi         Remove one or more images
  run         Run a command in a new container
  save        Save one or more images to a tar archive (streamed to STDOUT by default)
  search      Search the Docker Hub for images
  start       Start one or more stopped containers
  stats       Display a live stream of container(s) resource usage statistics
  stop        Stop one or more running containers
  tag         Create a tag TARGET_IMAGE that refers to SOURCE_IMAGE
  top         Display the running processes of a container
  unpause     Unpause all processes within one or more containers
  update      Update configuration of one or more containers
  version     Show the Docker version information
  wait        Block until one or more containers stop, then print their exit codes

Run 'docker COMMAND --help' for more information on a command.

To get more help with docker, check out our guides at https://docs.docker.com/go/guides/

Command failed

with the error message: 

**unknown flag: --build-arg**

Any suggestions?

kuba2k2, 2023-03-06:
Yeah, you don't have Docker Buildx... which should be installed by default. Try updating Docker, or Google on how to install buildx. I don't remember the solution, but when I had these issues, googling helped.

edautz, 2023-03-06:
Busy evening:

Started over again with clean machine, took a lots of steps and learned a lot:

Upgraded python from 3.8 to 3.9
Installed docker with buildx
Installed platformIO
Installed all the git software

then specific:

sudo apt-get purge python3-yaml
cd libretuya-esphome
python3 -m pip install .
python3 docker/build.py --tag libretuya --arch amd64 --build-type docker build

Got error: #0 5.103 E: Version '7.74.0-1.3+deb11u5' for 'curl' was not found

Tried another curl version in the docker/Dockerfile but error persists.

Commented #curl=7.74.0-1.3+deb11u5 \

Tried again

python3 docker/build.py --tag libretuya --arch amd64 --build-type docker build

Success:

docker images
REPOSITORY                      TAG         IMAGE ID       CREATED         SIZE
esphome/esphome-amd64           dev         ab829a812744   3 minutes ago   286MB
esphome/esphome-amd64           libretuya   ab829a812744   3 minutes ago   286MB
ghcr.io/esphome/esphome-amd64   dev         ab829a812744   3 minutes ago   286MB
ghcr.io/esphome/esphome-amd64   libretuya   ab829a812744   3 minutes ago   286MB
hello-world                     latest      feb5d9fea6a5   17 months ago   13.3kB

Created docker compose .yaml file

docker compose -f esphome.yaml up -d

Container runs. Container reachable through http://localhost:6052/

Create a test.yaml file containing the usual ESPHome stuff but specific:

libretuya:
  board: cb2s
  framework:
    version: dev

within it.

Validate, install, manual download, modern format and a test.uf2 firmware file is compiled!!!!!

But the queston remains: **Does the commenting of #curl=7.74.0-1.3+deb11u5 \
causes problems somewhere in my setup??**

edautz, 2023-03-07:
@kuba2k2,

Can you please respond to my question?

**Does the commenting of #curl=7.74.0-1.3+deb11u5 \ in the Dockerfile causes problems somewhere in my setup??**

kuba2k2, 2023-03-07:
No. As long as it works, there should be no problems.

edautz, 2023-03-07:
Thanks for your fantastic product and swift support.

closed 2023-03-07 by edautz