libretiny-eu/libretiny #98: Help with compiling for device group

ferbulous, 2023-03-07:
Hi, I'm trying to get device group (dgr) working from [here](https://github.com/tony-fav/esphome-sync)
Any recommended fixes to compile it successfully in LT?

device.yml
```
esphome:
  name: aubess_wifi_mini
  name_add_mac_suffix: true
  includes:
    - ehdgr/support_device_groups_compat.h  # compatibility layer so support_device_groups_ehdgr.h need not be modified too much
    - ehdgr/support_device_groups_ehdgr.h   # support_device_groups.ino modified as minimally as possible
  on_boot:
    priority: -100
    then:
      - script.execute: dgr_start
  on_loop:
    then:
      - if:
          condition:
            switch.is_on: use_dgr
          then:
            script.execute: dgr_loop

libretuya:
  board: generic-bk7231n-qfn32-tuya
  framework:
    version: latest
  lt_config:
    LT_UART_DEFAULT_PORT: 1

external_components:
  - source: github://esphome/esphome@dev
    components: [ light ]

logger:
  hardware_uart: SERIAL1
  deassert_rts_dtr: true

api:
  password: ""
  reboot_timeout: 0s

ota:
  password: ""

web_server:
  port: 80

script:
  - id: dgr_start
    mode: single
    then:
      - lambda: |-
          EHDGR_Init("devicegrouptest");
  - id: dgr_loop
    mode: single
    then:
      - lambda: |-
          // Log EHDGR_State every 10s
          static unsigned long last_state_log_time = millis();
          if (millis() > last_state_log_time + 10000)
          {
            EHDGR_LogState();
            last_state_log_time = millis();
          }
          EHDGR_StartAndLoop();
          if (EHDGR_GetPowerState(1) != id(DGR_Relay_1).state) id(DGR_Relay_1).publish_state(EHDGR_GetPowerState(1));
  - id: dgr_end
    mode: single
    then:
[... 294 lines cut ...]
               ^
In file included from src/esphome/core/application.h:7:0,
                 from src/esphome\components\api\api_connection.h:7,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/hal.h:37:10: note: uint32_t esphome::millis()
 uint32_t millis();
          ^
aubess_wifi_mini.yml:59:38: error: call of overloaded 'millis()' is ambiguous
aubess_wifi_mini.yml:59:38: note: candidates are:
In file included from C:/Users/theba/.platformio/packages/framework-arduino-api/api/Interrupts.h:8:0,
                 from C:/Users/theba/.platformio/packages/framework-arduino-api/api/ArduinoAPI.h:29,
                 from C:/Users/theba/.platformio/platforms/libretuya/arduino/beken-72xx/cores/arduino/Arduino.h:10,
                 from <command-line>:0:
C:/Users/theba/.platformio/packages/framework-arduino-api/api/Common.h:103:15: note: long unsigned int millis()
 unsigned long millis(void);
               ^
In file included from src/esphome/core/application.h:7:0,
                 from src/esphome\components\api\api_connection.h:7,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/hal.h:37:10: note: uint32_t esphome::millis()
 uint32_t millis();
          ^
```

kuba2k2, 2023-03-07:
Are you sure the dgr file is up to date for latest versions of esphome? It looks like it's not including something.

ferbulous, 2023-03-07:
> Are you sure the dgr file is up to date for latest versions of esphome? It looks like it's not including something.

I'm not quite sure, I was able to compile this [config](https://github.com/tony-fav/esphome-sync/blob/main/dgr/d1mini-devgroup_MinimalRelay.yaml) in esphome

ferbulous, 2023-03-21:
Hi, updated to the latest version of LT and I'm getting a different error this time.

```
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\api_frame_helper.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\api_pb2_service.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\api_server.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\list_entities.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\proto.cpp.o
Compiling .pioenvs\aubess_wifi_mini\src\esphome\components\api\subscribe_state.cpp.o
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/api_frame_helper.cpp:5:0:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/api/proto.h:5:0,
                 from src/esphome/components/api/proto.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\proto.cpp.o] Error 1
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\api_pb2_service.cpp.o] Error 1
In file included from src/esphome/components/binary_sensor/binary_sensor.h:5:0,
                 from src/esphome/core/controller.h:5,
                 from src/esphome/core/component_iterator.h:4,
                 from src/esphome/components/api/list_entities.h:4,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/binary_sensor/binary_sensor.h:5:0,
                 from src/esphome/core/controller.h:5,
                 from src/esphome/components/api/api_server.h:4,
                 from src/esphome/components/api/api_server.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
src/esphome/core/helpers.h:553:41: error: invalid initialization of non-const reference of type 'esphome::Mutex&' from an rvalue of type '<brace-enclosed initializer list>'
   LockGuard(Mutex &mutex) : mutex_{mutex} { mutex_.lock(); }
                                         ^
In file included from src/esphome/components/binary_sensor/binary_sensor.h:5:0,
                 from src/esphome/core/controller.h:5,
                 from src/esphome/core/component_iterator.h:4,
                 from src/esphome/components/api/subscribe_state.h:4,
                 from src/esphome/components/api/subscribe_state.cpp:1:
src/esphome/core/helpers.h: In constructor 'esphome::LockGuard::LockGuard(esphome::Mutex&)':
[... cut ...]
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\api_connection.cpp.o] Error 1
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\api_pb2.cpp.o] Error 1
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\list_entities.cpp.o] Error 1
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\subscribe_state.cpp.o] Error 1
*** [.pioenvs\aubess_wifi_mini\src\esphome\components\api\api_server.cpp.o] Error 1
==================================================== [FAILED] Took 6.47 seconds ====================================================
```

kuba2k2, 2023-03-21:
You haven't updated LibreTuya, only ESPHome. These are separate things.

closed 2023-08-25 by ferbulous