libretiny-eu/libretiny #99: Compile error

edautz, 2023-03-13:


kuba2k2, 2023-03-14:
I hate this

edautz, 2023-03-14:
Sorry for the hassle.

Is there anything I can do?

kuba2k2, 2023-03-14:
Don't be sorry, it's not your fault. This issue is my greatest failure, and I hate trying to fix it constantly with no success. Encryption - doesn't build. No encryption - doesn't update OTA. Encryption again - doesn't build again, even if everything tells me it should...

RoganDawes, 2023-03-14:
Would it not make sense to simply use the slowest pure python solution, and skip trying to install  high-performance wheels? I mean, if it is the difference between 55 seconds or 60 seconds to build, I really don't think anyone will care. It's better than something that doesn't build at all.

kuba2k2, 2023-03-14:
Sure, 55 vs 60 seconds is not a huge difference. That's a difference on a fairly modern PC - it might be 55 vs 80 seconds on a Raspberry Pi or some other SBC. If the first build takes 55 seconds, the following builds might take 10 seconds. And 10 vs 35 seconds is already a huge difference.

I'll try to find the root of this issue and resolve it, for good this time. I have a Pi 3B, but for some reason it says it's aarch64. I have an aarch64 VM in QEMU as well, but I don't have any ARM 32 bit systems. Might have to make a VM for that too...

zviratkos, 2023-03-14:
Hello, 

won't that be same problem with not having install 32bit libc6-i386? Somewhere else was mentioned, that arm-linux-gnueabihf-gcc is only 32bit application? 
In my case "apt-get install libc6-i386" solved this error, but I'm not running dockerized version, just 64bit Debian VM.

   Jarda

kuba2k2, 2023-03-14:
No, this is a different problem. I believe it's fixed since yesterday, because I upgraded GCC to a version that has real 64 bit binaries.

RoganDawes, 2023-03-14:
> Sure, 55 vs 60 seconds is not a huge difference. That's a difference on a fairly modern PC - it might be 55 vs 80 seconds on a Raspberry Pi or some other SBC. If the first build takes 55 seconds, the following builds might take 10 seconds. And 10 vs 35 seconds is already a huge difference.

Sure, a 25 second difference is a large percentage. But still less than it would take to log a GitHub issue, and troubleshoot the problem. "First make it correct (i.e. actually work), then make it fast."

kuba2k2, 2023-03-14:
I've updated ltchiptool to (hopefully) not install PyCryptodomex on ARM. The related LT commit is eae64d34e265b68c1584a954cfb74b31bf858f02, so try updating, rebuild the docker image and compile again.

I noticed that in the Dockerfile there's `python3-cryptography` installed - so ltchiptool will now try to use that too. If all options fail, it will fall back to the slow PyAES.

edautz, 2023-03-14:
I rebuild the container from scratch, deleted the old config directory. Container runs fine, but  the compile failes with another bunch of errors. Look like there is no compiler.

INFO Reading configuration /config/test.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing test-cloud-cutter-lsc-connec1 (board: wb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/kuba2k2/libretuya.git
INFO Installing git+https://github.com/kuba2k2/libretuya.git
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-texhe6vd'...
Platform Manager: libretuya@0.12.6+sha.eae64d3 has been installed!
INFO libretuya@0.12.6+sha.eae64d3 has been installed!
Installing/updating ltchiptool
Collecting ltchiptool<4.0,>=^3.0.2
  Downloading ltchiptool-3.0.2-py3-none-any.whl (116 kB)
Collecting click<9.0.0,>=8.1.3
  Downloading click-8.1.3-py3-none-any.whl (96 kB)
Collecting prettytable<4.0.0,>=3.3.0
  Downloading prettytable-3.6.0-py3-none-any.whl (27 kB)
Collecting colorama<0.5.0,>=0.4.5
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting importlib-metadata<5.0.0,>=4.12.0
  Downloading importlib_metadata-4.13.0-py3-none-any.whl (23 kB)
Collecting bk7231tools<2.0.0,>=1.3.0
  Downloading bk7231tools-1.3.1-py3-none-any.whl (28 kB)
Collecting pyaes<2.0.0,>=1.6.1
  Downloading pyaes-1.6.1.tar.gz (28 kB)
Collecting xmodem<0.5.0,>=0.4.6
  Downloading xmodem-0.4.6.tar.gz (32 kB)
Collecting pyserial<4.0,>=3.5
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting zipp>=0.5
  Downloading zipp-3.15.0-py3-none-any.whl (6.8 kB)
Collecting wcwidth
  Downloading wcwidth-0.2.6-py2.py3-none-any.whl (29 kB)
Building wheels for collected packages: pyaes, xmodem
  Building wheel for pyaes (setup.py): started
  Building wheel for pyaes (setup.py): finished with status 'done'
  Created wheel for pyaes: filename=pyaes-1.6.1-py3-none-any.whl size=26346 sha256=d6db64f3757fb4834bf7b131331bb47b1a4df2da61f8d165ca43a5c3454c5614
  Stored in directory: /root/.cache/pip/wheels/33/13/e9/ef9894385084d8e4fab38f5f83afda49a3d593d8afc907c8fd
  Building wheel for xmodem (setup.py): started
  Building wheel for xmodem (setup.py): finished with status 'done'
  Created wheel for xmodem: filename=xmodem-0.4.6-py3-none-any.whl size=34564 sha256=c7fe453b51123f528e4cef18f25883fbdd3d4826ea1d931f6da6fad7c76aafcb
  Stored in directory: /root/.cache/pip/wheels/03/02/db/af85b84f5d4713e3fcbdf10bbd72306cb4516e1105eddd86d7
Successfully built pyaes xmodem
Installing collected packages: zipp, wcwidth, pyserial, xmodem, pyaes, prettytable, importlib-metadata, colorama, click, bk7231tools, ltchiptool
  Attempting uninstall: pyserial
    Found existing installation: pyserial 3.5
    Uninstalling pyserial-3.5:
      Successfully uninstalled pyserial-3.5
  Attempting uninstall: colorama
    Found existing installation: colorama 0.4.6
    Uninstalling colorama-0.4.6:
      Successfully uninstalled colorama-0.4.6
  Attempting uninstall: click
    Found existing installation: click 8.1.3
    Uninstalling click-8.1.3:
      Successfully uninstalled click-8.1.3
Successfully installed bk7231tools-1.3.1 click-8.1.3 colorama-0.4.6 importlib-metadata-4.13.0 ltchiptool-3.0.2 prettytable-3.6.0 pyaes-1.6.1 pyserial-3.5 wcwidth-0.2.6 xmodem-0.4.6 zipp-3.15.0
Tool Manager: Installing git+https://github.com/libretuya/framework-beken-bdk#v2021.06.07
INFO Installing git+https://github.com/libretuya/framework-beken-bdk#v2021.06.07
git version 2.30.2
Cloning into '/config/.esphome/platformio/cache/tmp/pkg-installing-wjygn1zo'...
Note: switching to '6491b8c001b737c790cba0bee16e548859d5e6fe'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
[... 149 lines cut ...]
Unpacking  [#####################---------------]   58%
Unpacking  [########################------------]   66%
Unpacking  [###########################---------]   75%
Unpacking  [##############################------]   83%
Unpacking  [#################################---]   91%
Unpacking  [####################################]  100%
Library Manager: DNSServer@1.1.0 has been installed!
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_connection.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: arm-none-eabi-g++: not found
Compiling .pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_pb2.cpp.o
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: arm-none-eabi-g++: not found
*** [.pioenvs/test-cloud-cutter-lsc-connec1/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================= [FAILED] Took 232.25 seconds =========================

edautz, 2023-03-14:
Tried this on a RPI3 and a RPI4. Both 64 bits machines with a 64 bits OS.

uname -a output RPI4:

Linux rpi4 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

catalin2402, 2023-03-14:
Try using this modifications from here in the Dockerfile. https://github.com/kuba2k2/libretuya-esphome/pull/5/files

kuba2k2, 2023-03-14:
Try `python docker/build.py --tag libretuya --arch arm64` (you have a 64-bit OS, so build the container for arm64, instead of armv7).

edautz, 2023-03-14:
Tried that in the first place, but as the Dockerfile show that isn't a valid option.

 python docker/build.py --tag libretuya --arch arm64 --build-type docker build
usage: build.py [-h] --tag TAG [--arch {amd64,armv7,aarch64}] --build-type
                {docker,ha-addon,lint} [--dry-run]
                {build,manifest} ...
build.py: error: argument --arch: invalid choice: 'arm64' (choose from 'amd64', 'armv7', 'aarch64')

catalin2402, 2023-03-14:
aarch64 == arm64

kuba2k2, 2023-03-14:
Oh, sorry, I got confused by seeing "arm64" [here](https://github.com/kuba2k2/libretuya-esphome/pull/5/files#diff-f34da55ca08f1a30591d8b0b3e885bcc678537b2a9a4aadea4f190806b374ddcR53)

edautz, 2023-03-14:
**Bad news.**

Docker build, compile and flash through Wifi succesfull, but devices is bricked.

**Not pingable, not in DHCP table, not in Arp table. No Wifi fallback AP. Not responding to it build-in switch. Nothing.**

Any way to revive the powerplug? Except for prying it open and do a serial flash?

Any idea what did go wrong in the process that bricked the device? I am afraid this method is rather useless until this is clear and solved. 

With my Ubuntu Amd64 VM with a libretuya ESPHome container version v2023.3.0-dev on a W10 i7 machine it worked.

Compile log:

INFO Reading configuration /config/test.yaml...
INFO Detected timezone 'Europe/Amsterdam'
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing test-cloud-cutter-lsc-connec2 (board: wb2s; framework: arduino; platform: https://github.com/kuba2k2/libretuya.git)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.0.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.0.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.0.0 has been installed!
INFO ESPAsyncWebServer-esphome@3.0.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/AsyncTCP-esphome
INFO Installing esphome/AsyncTCP-esphome
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.0.0 has been installed!
INFO AsyncTCP-esphome@2.0.0 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
HARDWARE: BK7231T 120MHz, 256KB RAM, 1.03MB Flash
 - framework-arduino-api @ 3.0.0-a4cbfc+sha.3a4cbfc 
 - framework-beken-bdk @ 0.0.0+v2021.06.07.sha.6491b8c 
 - library-flashdb@03500fa @ 3500.0.0-fa+sha.03500fa 
 - library-lwip@2.1.3-bdk @ 2.1.3-bdk+sha.57a7df1 
 - library-printf@6.0.0 @ 6.0.0+sha.8b831c1 
 - tool-ltchiptool @ 2.0.2+sha.7559033 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing DNSServer

Unpacking  [------------------------------------]    0%
Unpacking  [###---------------------------------]    8%
Unpacking  [######------------------------------]   16%
Unpacking  [#########---------------------------]   25%
Unpacking  [############------------------------]   33%
Unpacking  [###############---------------------]   41%
Unpacking  [##################------------------]   50%
Unpacking  [#####################---------------]   58%
Unpacking  [########################------------]   66%
Unpacking  [###########################---------]   75%
Unpacking  [##############################------]   83%
Unpacking  [#################################---]   91%
Unpacking  [####################################]  100%
Library Manager: DNSServer@1.1.0 has been installed!
Dependency Graph
|-- ESPAsyncWebServer-esphome @ 3.0.0
|   |-- AsyncTCP-esphome @ 2.0.0
|-- DNSServer @ 1.1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/test-cloud-cutter-lsc-connec2/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test-cloud-cutter-lsc-connec2/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/test-cloud-cutter-lsc-connec2/src/esphome/components/api/api_pb2.cpp.o
[... 447 lines cut ...]
Archiving .pioenvs/test-cloud-cutter-lsc-connec2/libbdk_ble_4_2.a
Linking .pioenvs/test-cloud-cutter-lsc-connec2/firmware.elf
|-- Image 1: firmware.elf
|   |-- bk7231t_app_0x011000.rbl
|   |   |-- firmware.bin
|   |-- bk7231t_app_0x011000.crc
|   |-- bk7231t_app_0x129F0A.rblh
|   |-- bk7231t_app.ota.rbl
|   |-- bk7231t_app.ota.ug.bin
Building UF2 OTA image
|-- firmware.uf2
RAM:   [====      ]  44.2% (used 115852 bytes from 262144 bytes)
Flash: [=====     ]  54.4% (used 589297 bytes from 1083136 bytes)
======================== [SUCCESS] Took 189.30 seconds ========================
INFO Successfully compiled program.
INFO Connecting to 192.168.2.222
INFO Uploading /config/.esphome/build/test-cloud-cutter-lsc-connec2/.pioenvs/test-cloud-cutter-lsc-connec2/firmware.uf2 (2052608 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.2.222 using esphome API
WARNING Can't connect to ESPHome API for 192.168.2.222: Timeout while connecting to ('192.168.2.222', 6053)
INFO Trying to reconnect to 192.168.2.222 in the background

edautz, 2023-03-14:
Used exactly the same yaml file as on my Ubuntu VM. Only changed the devicename to "test-cloud-cutter-lsc connec2” and the friendly name to "test cloud cutter lsc connec2”, to determine if the OTA upload is succesfull. You can find this file at the beginning of this issue.

I can save a firmware file for analyses. Does this help?

kuba2k2, 2023-03-15:
Yes, if you could send me the firmware.uf2 file that got uploaded to the device, I can test it to see if it works.

edautz, 2023-03-15:
[test-cloud-cutter-lsc-connec2.zip](https://github.com/kuba2k2/libretuya/files/10977183/test-cloud-cutter-lsc-connec2.zip)

I changed all the Wifi Sids and alle the passwords in TESTTEST

kuba2k2, 2023-03-15:
@edautz Thanks for the report, the issue has been fixed. Update everything and do a clean build.
Unfortunately, there's not much you can do to update the plug. The only option now is to disassemble it, and flash firmware using UART.

edautz, 2023-03-15:
@kuba2k2, Saw you fixed an infinite bootloop. This explaines why my devices doesn’t respond to anything. Thnxs. 

How shure is it to use the updated code?

I don’t want to brick another.

kuba2k2, 2023-03-15:
You'll have to disassemble the bricked devices anyway, so I'd try on these first. We have tested it with a few configs and it worked, but now I'm getting another similar issue on my board (but that might be completely unrelated).

edautz, 2023-03-15:
The bricked device will be solved when I got plenty time. It is difficult to pry it open at a decent way.

I buiild everything new from scratch.

Can you please test the new compiled firmware, before I flash it?

[test-cloud-cutter-lsc-connec3.zip](https://github.com/kuba2k2/libretuya/files/10984589/test-cloud-cutter-lsc-connec3.zip)

kuba2k2, 2023-03-15:
I can't tell if it boots or not. I can see the welcome message, and it doesn't reboot every 10 seconds, so chances are it actually works. You've probably removed the logger component (or set baud_rate to 0) so it doesn't produce any visible output.

edautz, 2023-03-15:
[test-cloud-cutter-lsc-connec4.zip](https://github.com/kuba2k2/libretuya/files/10984749/test-cloud-cutter-lsc-connec4.zip)

I made a version with logger baudrate 0 commented out.
Can you please test it, to make 100% shure with a decent logging?

Thanxs.

kuba2k2, 2023-03-15:
I can confirm it works:
[image]

edautz, 2023-03-16:
I flashed my devices succesfull.

Thanks for your wonderfull software and all the swift support 👍

closed 2023-03-16 by edautz