libretiny-eu/libretiny #100: libretuya-esphome support for wb2l_m1 board?

Elkropac, 2023-03-14:
Hi,
i have some bulbs with wb2l_m1 like this one https://www.elektroda.com/rtvforum/topic3880540.html .
I have succesfully flashed it with openbeken using tuya-cutter.
But I would like to use it with libretuya-esphome.
I tried to place
```
libretuya:
  board: wb2l_m1
  framework:
    version: latest
```
to config, but it does not compile.
Is there a way to compile it?

kuba2k2, 2023-03-14:
The bulb in the linked post doesn't have WB2L_M1, but from what I know WB2L_M1 is BK7231N, despite the "WB" prefix. 
It's not in the [board list](https://docs.libretuya.ml/docs/status/supported/#board-list), but CB2L is, so you can use that. You can also use [generic-bk7231n-qfn32-tuya](https://docs.libretuya.ml/boards/generic-bk7231n-qfn32-tuya/).

Elkropac, 2023-03-14:
Sorry i messed up the links then, I was flashing it several months ago.
In smartlife app it showed firmware version 1.3.21 , so i used it in tuya cutter and then uploaded openbk to it.

So is it safe to use the board you suggested?

kuba2k2, 2023-03-14:
Yes, 1.3.21 is BK7231N. You can use the boards I suggested.